springfall2008/batpred #1521: Octopus Integration - Smart Charging not working

Sycamore967, 2024-10-11:
**Describe the bug**
Predbat isn't detecting that I have my car plugged in or that the charging slot is before 23:30

**Expected behavior**
It used to work and I'm sure I've missed some sort of update or reconfig info I just don't know.

**Predbat version**

8.4.11

**Environment details**

- GE 3.6
- Standard HAOS
- A couple of weeks ago I noticed that the BottleCap Dave Octopus Integration wasn't showing any info.  When I checked my Octopus app my Hypervolt Device and all Intelligent settings had been deleted.  It may have something to do with Hypervolts updates to use their charger with Octopus.  Anyway I added everything back in to the Octopus App and most settings came back.  I've attached screenshots which I think point to my problem.  I'm hoping I can just switch Predbat to use another sensor, I'm just not sure what to update, or if there is something else I should have done already.

**Screenshots**
Octopus Integration sensors not available, Bottle Cap Dave senso showing next off peak start

**Log file**
I've attached my config and a log file, plus the screenshots of my sensors
[image]
[predbat.zip](https://github.com/user-attachments/files/17346542/predbat.zip)
[image]

Sycamore967, 2024-10-12:
Ok I've had another look and can understand the apps.yaml file better now I'm not rushing.  I've updated the following:

  octopus_intelligent_slot: 're:(binary_sensor.octopus_intelligent_planned_dispatch_slot)'
  octopus_ready_time: 're:(select.octopus_target_ready_by_time)'
  octopus_charge_limit: 're:(select.octopus_target_state_of_charge)'

I'd be grateful is you could let me know if this should work.  The history of the sensors certainly shows the intelligent dispatch slots I expected to trigger a charge cycle in Predbat and the correct charge level and ready times.  The proof will be my next car charge cycle that falls outside of the normal 23:30 start time.

springfall2008, 2024-10-13:
Looks correct to me, let me know if it works

silkit, 2024-10-20:
@Sycamore967 please see the issue I had as well with Hypervolt HP3 [here](https://github.com/springfall2008/batpred/issues/1548) Make sure your car charging planned sensor and response are correct as the entity provides a different response than expected.
Expected response = plugged in
Actual Response = on
Without that it won’t detect your car is plugged in.

gcoan, 2024-10-21:
> Make sure your car charging planned sensor and response are correct as the entity provides a different response than expected.

I have BTW added details about checking the car charging planned sensor response to the documentation steps.  It'll be in my next PR

chalfontchubby, 2026-08-29:
Thanks for the report. This issue has been open for over 6 months with no recent activity - we're volunteers working through a large backlog and need to filter what's still relevant. If this is still a problem for you, please let us know and we'll keep it open. Otherwise, we'll close it in a week (around 2026-09-05).

closed 2026-09-07 by chalfontchubby