springfall2008/batpred #1522: Solaredge configuration with multiple batteries

jrviz, 2024-10-12:
I have a Solaredge system with 3 batteries. I have just set up predbat and this is all very new at the moment. 
I can see in the configuration, which is based on the install template for the Solaredge inverter the entries for battery are:
  battery_rate_max:
    - sensor.solaredge_b1_max_charge_power
  battery_power:
    - sensor.solaredge_b1_dc_power
  soc_percent:
    - sensor.solaredge_b1_state_of_energy
  soc_max:
    - sensor.solaredge_b1_maximum_energy
    - 
This is just one battery. The Solaredge Modbus Multi integration enumerates multiple batteries as b1, b2, b3 identifiers in a set of entities for each battery.

To get predbat to use the correct information I assume that soc_max and soc_percent need to be a sum of each of the relevant entities but for battery_rate_max and battery_power can these simply be summed as I'm not sure how these values are used by predbat for calculations and control?  Can multiple batteries be used for the Solaredge configuration?

springfall2008, 2024-10-13:
I think you need to make template sensors to sum these up if its a single inverter with multiple batteries.

For battery power I assume again a template sensor summing them would work.

For battery max rate its just a fixed number so you can type in whatever your maximum charge/discharge rate is.

buttash, 2025-02-13:
I have installed two SolarEdge 10kWh batteries and am getting a discharge to grid with message:

`Inverter 0 adjust target soc for charge to 52% (10.09kWh/19.4kWh 19.4kWh) based on going from 80% -> 52% total add is -5.37kWh and this battery needs to add -5.37kWh to get to 10.09kWh`

Does the '10.09kWh' in the above come from ' battery_rate_max'?  I have that currently as 10000, whereas I've summed 'soc_max' to 19.4kWh.  'soc_percent' I have as the average soc of both batteries.

jrviz, 2025-02-13:
Isn't the 10.09 just the 52% of your 19.4 total?

I sum the battery capacities for the soc max but for  rate max I currently have 5000 which is the rate max each battery has but I have assumed the limit is set by the inverter and it will have a limit for the AC/DC conversion and for multiple batteries this is shared across them. I think that's what I see in practice on the overnight charges

buttash, 2025-02-13:
Thanks @jrviz for clearing the 10.09kWh figure.  I jumped to a conclusion that it was related to parameter and not a calculated value.  I looked the code but still it didn’t occur to me.

Ok - I will set battery_rate_max to 5000 and see how things go tonight.

TBH I’m confused by the charging/discharging behaviour - it discharges when the battery is not full and then at the end of the low rate period the batteries are not fully charged.  How this relates to battery_max_rate isn’t obvious but if the change fixes things then I’ll be happy!

jrviz, 2025-02-13:
I'm away at the moment and Predbat is charging my batteries at night and in the day the heating keeps a low background temperature and runs a security light schedule. The batteries then export excess in the evening. The discharge export slots move about a bit depending on the solar inputs and actual excess at the end of the day but I am ending up with reliable full charge by the end of the low rate period.

buttash, 2025-02-14:
Setting the value for battery_rate_max to 5000 seems to have worked.  I'll obviously check over the coming days, but it is looking good.  Thanks.

By way of a summary for anyone coming across this thread, I have set my values as follows:

1.  battery_rate_max - set to maximum of the 10kWh SolarEdge battery, 5000
2. charge_rate/discharge_rate - set to the SolarEdge battery spec, again 5000
3. soc_max - set to the combined battery power, 19.4kWh 
4. soc_percent - the average soc of my two batteries

jrviz, 2025-02-14:
Perhaps those of us with solar edge could update the documentation to add what we have got for working configurations? I'm happy to help with this

buttash, 2025-02-14:
A good idea - it ight help others to have a SolarEdge Modbus Multi integration example.

gcoan, 2025-04-01:
I have updated the documentation for multi-battery SolarEdge inverters, please check you are happy with the change, it'll be in the next PR I do 

https://github.com/gcoan/batpred/blob/main/docs/inverter-setup.md#solaredge-inverters

jrviz, 2025-04-01:
Great.  One slight build, where it says:
`
set battery_rate_max to the maximum value of the SolarEdge battery, e.g. 5000
set charge_rate and discharge_rate to the SolarEdge battery values, e.g. 5000
`
The rates are actually set by the Inverter as it is doing the DC/AC conversion. Where it says battery I think it should say inverter.

Thanks

gcoan, 2025-04-01:
@jrviz done

cyberkryten, 2025-04-01:
If you set those to physical values then you won't be able to use any of the Freeze options correctly as Predbat changes those to do the freeze operation.  

That might not be an issue (I'm currently undecided) due to the way the charge/discharge works but when those are set, but I think it makes sense for them to at least be setup correctly and available.

Here's the settings - note that the limits read from the inverter will likely change to some other value when the inverter does it's reset (mine goes to 11400).  If setup correctly like this, Predbat will detect that and revert it to the correct charge/discharge rate as set elsewhere in the apps.yaml config

 ```
 battery_rate_max:
    #- sensor.solaredge_b1_max_charge_power
    # For three batteries, this could be 15000 if DC charging
    # 31Mar25 NEW Sensor for total of three batteries Max Charge Power to test charge/discharge issues
    - sensor.semodbus_power_batteries_max_charge_power
  charge_rate:
    - number.solaredgemodbus_i1_storage_charge_limit
  discharge_rate:
    - number.solaredgemodbus_i1_storage_discharge_limit
```

The sensor I am using is this one.  There are times when the values read from the batteries drop to zero, no idea why but it is a SE issue, not the Modbus integration, so this adds the three totals together, but always ensures at least one charge rate of 5k is set.

```
# Calculate Max Battery Charge rate
# This is the sum of all three as the max charge rate can be higher than inverter capacity (8k) when charging from AC+Solar
# Always at least 5000W
# Used by Predbat
- sensor:
    - name: "SEModbus Power - Batteries Max Charge Power"
      unique_id: semodbus_power_batteries_max_charge_power
      unit_of_measurement: "W"
      device_class: "power"
      state_class: "measurement"
      state: >
        {% set myB1 = float(states('sensor.solaredgemodbus_b1_max_charge_power'),0) %}
        {% set myB2 = float(states('sensor.solaredgemodbus_b2_max_charge_power'),0) %}
        {% set myB3 = float(states('sensor.solaredgemodbus_b3_max_charge_power'),0) %}
        {% set myvalue = ((myB1 + myB2 + myB3)) | int %}
        {{ (myvalue if (myvalue) > 5000 else 5000) }}
```

jrviz, 2025-04-01:
That's some more detail. The charge and discharge rate settings above are those values in the template. The more accurate battery_rate_max relies on an aggregating sensor being set up.  Geoffrey has put some comments in the documentation pointing out the need to expand some of the other calculated sensors for multiple batteries. Perhaps it would be good to just have an example  three battery set of sensors in the documentation and then a comment that it can be scaled back for two and one batteries by taking out the b3 and b2 statements?

We're heading for an updated apps.yaml but to help others coming on board it would be good to have the sensors all in the document. This is all still fairly new to me, @cyberkryten would you be willing to share your three battery set of sensors for configuration.yaml (or referenced by)

Then there is the minor can of worms for approaches and automations to intervene after a unilateral inverter reset from SolarEdge !!!

cyberkryten, 2025-04-01:
Well, every time I start pulling together to upload my configs I find another issue with what I've got, perfect being the enemy of 'good enough' and all that!  I'll look at adding it all to a repository soon(ish).

There are some sensors which are very install specific - my Solar PV calculations are insanely complex and would confuse most newcomers and I use a Glow to read Import/Export figures from the Smart Meter directly.  I've posted both before.

However, in addition to what's above, I think these are the ones that relate to the batteries.  None are highly optimised, there seemed to be little benefit in making one-liner and I spent a lot of time pasting these into Developer Tools->Template so I could then add extra print statements for the B1/B2/B3 values individually which is much easier this way.

(though now I want to redo the first one to match the later naming convension of 'calc_' prefix):

```
  battery_power:                                                                                              
    - sensor.semodbus_power_batteries_dc_power     
  soc_percent:                                                                                               
    - sensor.calc_battery_all_state                                                                          
  soc_max:                                                                                                    
    - sensor.calc_battery_total_capacity                                                                      
  soc_kw:                                                                                             
    - sensor.calc_battery_current_capacity                                                            
```

Then these are the template sensors

```
# Used by Predbat
- sensor:
    - name: "Calc Battery All State"
      unique_id: calc_battery_all_state
      unit_of_measurement: "%"
      state: >
        {% set myB1 = float(states('sensor.solaredgemodbus_b1_state_of_energy'),0) %}
        {% set myB2 = float(states('sensor.solaredgemodbus_b2_state_of_energy'),0) %}
        {% set myB3 = float(states('sensor.solaredgemodbus_b3_state_of_energy'),0) %}
        {% set myvalue = ((myB1 + myB2 + myB3) / 3) | round(0) %}
        {{ myvalue }}

# Calculate Total Battery Power Values
# Used by Predbat
- sensor:
    - name: "SEModbus Power - Batteries DC Power"
      unique_id: semodbus_power_batteries_dc_power
      unit_of_measurement: "W"
      device_class: "power"
      state_class: "measurement"
      state: >
        {% set myB1 = float(states('sensor.solaredgemodbus_b1_dc_power'),0) %}
        {% set myB2 = float(states('sensor.solaredgemodbus_b2_dc_power'),0) %}
        {% set myB3 = float(states('sensor.solaredgemodbus_b3_dc_power'),0) %}
        {% set myvalue = ((myB1 + myB2 + myB3)) %}
        {{ myvalue }}

# Total Energy Stored in the Batteries
# Used by Predbat (indirectly)
- sensor:
    - name: "Calc Battery Total Capacity"
      unique_id: calc_battery_total_capacity
      unit_of_measurement: kWh
      state: >
        {% set myB1 = float(states('sensor.solaredgemodbus_b1_maximum_energy'),0) %}
        {% set myB2 = float(states('sensor.solaredgemodbus_b2_maximum_energy'),0) %}
        {% set myB3 = float(states('sensor.solaredgemodbus_b3_maximum_energy'),0) %}
        {% set myvalue = ((myB1 + myB2 + myB3)) %}
        {{ myvalue }}

# Current Energy Stored in the Batteries
# Used by Predbat
- sensor:
    - name: "Calc Battery Current Capacity"
      unique_id: calc_battery_current_capacity
      unit_of_measurement: kWh
      state: >
        {% set myCurrentCapacity = (float(states('sensor.calc_battery_all_state'),0) / 100) * float(states('sensor.calc_battery_total_capacity'),0) %}
        {{ myCurrentCapacity }}
```

Then I use this because I have a Backup Gateway:

```
  # Can be enabled you have a backup interface                                                    
  reserve:                                                                                        
    - number.solaredgemodbus_i1_backup_reserve   

```

cyberkryten, 2025-04-01:
It would be great if we could split out the apps.yaml to enable #include for subsections as I suspect the main predbat config and then including giv.yaml or solaredge.yaml would simplify things for those starting out

gcoan, 2025-04-01:
> It would be great if we could split out the apps.yaml to enable #include for subsections as I suspect the main predbat config and then including giv.yaml or solaredge.yaml would simplify things for those starting out

This is very much my intent to do something like this, either an apps.yaml that contains all the overarching stuff like cars, solcast, import/export rates, etc and then either #include or some other way of then incorporating the inverter-specific stuff into much smaller focussed files.

I had considered having just one master apps.yaml containing all the inverter stuff but with the growth of ever more inverters in predbat, this is unlikely to be practical, so hence my current thinking as above,  The current approach is a mess, changes made to one template don't get replicated across other templates and it is a labour to maintain.

Another job is to resequence apps.yaml so that it follows the same structure as the documentation, at the moment it is all over the place, and has grown organically.

On my list!

Back to the changes for multiple inverters, @jrviz @cyberkryten are you happy with the apps.yaml changes and template sensors posted above or more changes to make?

jrviz, 2025-04-01:
I think it's worth adding @cyberkryten battery_rate_max refinement using the added sensor.

Perhaps for the larger example set of sensors for three batteries there is some work still to do and that can be a later update or added to a discussion topic?

gcoan, 2025-04-02:
I've updated the solaredge template and inverter setup document to include battery_rate_max

I did notice that in @cyberkryten apps.yaml above, charge_rate and discharge_rate were set to number.solaredgemodbus_i1_storage_charge_limit not number.solaredge_i1_storage_charge_limit - is there a general renaming of the entities in apps.yaml that needs to be done?

> Perhaps for the larger example set of sensors for three batteries there is some work still to do and that can be a later update or added to a discussion topic?

You can carry on the discussion here if you want rather than starting a new issue

jrviz, 2025-04-03:
That's a good question. In my integration for SolarEdge Modbus Multi my entities are named <type>.solaredge_<device>_<entity>  but for @cyberkryten the integration name is solaredgemodbus. I didn't change anything from the default install so I assume that is the default.

cyberkryten, 2025-04-03:
I already had the SolarEdge integration, so when I added the SolarEdge Modbus Multi one I needed a different name and decided on that to make it clear which was which.

Maybe add a note that 'number.solaregde_' is the default naming for the integration and that if you renamed it then you will need to replace all instances with whatever you called it.

Alternatively, change all references to '<YOURINTEGRATION>_i1_storage_charge_limit' to force all users to edit it

gcoan, 2025-04-03:
> Maybe add a note that 'number.solaregde_' is the default naming for the integration and that if you renamed it then you will need to replace all instances with whatever you called it.

I've changed to using solaredge in the yaml file and templates rather than your custom name, and added a guidance note as you suggested

buttash, 2025-04-04:
Just by way of a comment, my modbus multi installation has a different naming convention to support multiple invertors.  For example:

```
sensor.solaredge_i1_b1_average_temperature
sensor.solaredge_i1_b1_max_charge_power
```
I've looked at the revisions to the templates and documents and they look good to me :)

cyberkryten, 2025-04-05:
```
      # Template sensor for Max Battery Charge rate
      # This is the sum of all three batteries charge rate as the max charge rate can be higher than inverter capacity (8k) when charging from AC+Solar
      # Always at least 5000W, the inverter limit
```

I'd reword this slightly.  8k is _my_ inverter limit, others may be different so maybe: (eg 8kW)

Alos, my reason for picking 5000W as the minimum to return is not to do with the inverter, it is the single battery charge/discharge limit (which appears to apply to both AC and DC charging).  I have periods when one or more of the batteries 'goes offline' for a while and stops reporting to Modbus but continue to work fine - so the 5kW means that at least one of them can always be charged.

gcoan, 2025-04-05:
> I'd reword this slightly. 8k is _my_ inverter limit, others may be different so maybe: (eg 8kW)
> 
> Alos, my reason for picking 5000W as the minimum to return
> ....

Done, have updated the inverter setup doc to reflect your comments

Any progress on the template updates for multiple battery support https://github.com/springfall2008/batpred/issues/1522#issuecomment-2770435383 ?

cyberkryten, 2025-04-07:
> Any progress on the template updates for multiple battery support [#1522 (comment)](https://github.com/springfall2008/batpred/issues/1522#issuecomment-2770435383) ?

I'm still using what I posted for the three battery setup

I can't easily post a lot of my other config for things like PV to House Load etc as I am not just using the SolarEdge sensors as I also have a Glow to read the Smart Meter plus Emporia Vue and Shelly CT clamps to get key information from the actual breaker circuits, which are more accurate than the SE ones.

gcoan, 2025-04-07:
> > Any progress on the template updates for multiple battery support #1522 (comment) ?
> I'm still using what I posted for the three battery setup

Great.  I thought you wanted to make some naming changes to it, so that's why I held of incorporating it into the template and documentation. 

If it's good to go, I'll merge it in

gcoan, 2025-04-07:
I've merged that all in now to the template and documentation, thanks @cyberkryten 

For consistency I named all the new template sensors calc_xxxx

gpsmith, 2025-08-27:
Just to confirm I took the new changes, adjusted it by removing the third battery (As I now have two) and adjusted the sensor as mine has the _i1_ part and it appears to be working. 

I did have to add template: and adjust the indentation to while adding it to configuration.yaml, which maybe obvious to some, but I needed to chatgpt to confirm what I was doing.

gcoan, 2025-11-05:
> Just to confirm I took the new changes, adjusted it by removing the third battery (As I now have two) and adjusted the sensor as mine has the _i1_ part and it appears to be working.

thanks, have updated the documentation to explain how to adjust for 2 or more than 3 batteries

> I did have to add template: and adjust the indentation to while adding it to configuration.yaml, which maybe obvious to some, but I needed to chatgpt to confirm what I was doing.

corrected indentation and explained about 'template:'

Am going to close this now as this is all incorporated in the documentation

closed 2025-11-05 by gcoan