springfall2008/batpred #1536: 404 error retrieving norddata

gcoan, 2024-10-14:
**Describe the bug**
Getting 404 error retrieving norddata future prices:

8774 2024-10-14 17:35:01.851376: Info: record_status Warn: Error downloading futurerate data from cloud
8773 2024-10-14 17:35:01.828201: Warn: Error downloading futurerate data from URL https://www.nordpoolgroup.com/api/marketdata/page/325?currency=GBP 

Not just me getting this error, a number of other people on the GivEnergy forum are getting the same error.

**Expected behavior**
No error messages!

Has the norddata page changed?

I found https://data.nordpoolgroup.com/auction/gb-half-hour/prices?currency=GBP

and tried this, but it didn't work.  Probably the format is different?

**Predbat version**

8.4.12

**Environment details**

HA 2024.10.2

slopemad, 2024-10-14:
I recall that this has happened before, but the data feed came back at that URL.

But also that I had found the new location for the data, in a different format which is probably slightly less convoluted to parse...  (the date parameter needs adjusting on a daily basis)...
https://dataportal-api.nordpoolgroup.com/api/DayAheadPrices/singleAreaHistory?date=2024-10-15&market=N2EX_DayAhead&deliveryArea=UK&currency=GBP

gcoan, 2024-10-15:
The fact that we are getting 404 errors makes me think the page has been permanently moved.  But let's see what happens.

I tried that alternative URL and couldn't get it to work, even hard coding the date in.  The problem was the last bit, the '&currency=GBP' which kept on being parsed as [some random character]cy=GBP

slopemad, 2024-10-15:
OK, so I already wrote a perl script to grab the new dataset which runs on a cron job at 09:01 UTC every day, this just grabs the data set, it doesn't do anything with it, but it's pretty trivially parseable.  There's a prices array in the JSON which has data like this:
"prices":[{"price":46.82,"deliveryStart":"2024-10-09T22:00:00Z","deliveryEnd":"2024-10-09T23:00:00Z"}]
The 'price' can then be adjusted in the same way the price is done in the old dataset.

```
#!/usr/bin/perl

use strict;
use warnings;
use LWP::UserAgent;
use JSON;
use DateTime;
use Time::HiRes qw(sleep);

my $datadir = "/home/dev.energy/data";

# Function to get tomorrow's date
sub get_tomorrow_date {
    my $tomorrow = DateTime->now(time_zone => 'local')->add(days => 1);
    return $tomorrow->ymd;
}

# Function to fetch JSON data from the API
sub fetch_json_data {
    my $date = shift;
    my $url = "https://dataportal-api.nordpoolgroup.com/api/DayAheadPrices?date=$date&market=N2EX_DayAhead&deliveryArea=UK&currency=GBP";
    my $ua = LWP::UserAgent->new;
    my $response = $ua->get($url);
    return $response;
}

# Main routine
sub main {
    my $date = get_tomorrow_date();
    my $retry_interval = 300;  # Retry interval in seconds (5 minutes)
    
    while (1) {
        my $response = fetch_json_data($date);
        
        if ($response->is_success && $response->code == 200) {
            my $json = $response->decoded_content;
            my $data = eval { decode_json($json) };
            if ($@) {
                print "Error decoding JSON: $@\n";
            } else {
                my $filename = "prices_$date.json";
                open(my $fh, '>', "$datadir/$filename") or die "Cannot open file '$filename' for writing: $!";
                print $fh $json;
                close($fh);
                print "JSON data saved to $filename\n";
                last;  # Exit the loop if JSON is successfully fetched
            }
        } else {
            print "HTTP request failed: " . $response->status_line . ". Retrying in $retry_interval seconds...\n";
            sleep($retry_interval);
        }
    }
}

main();
```

gbenn324234, 2024-10-16:
Looks like they deprecated the v1 endpoints.

https://www.nordpoolgroup.com/en/trading/Operational-Message-List/2024/03/deprecation-of-market-data-v1-endpoints-20240318095300/

springfall2008, 2024-10-17:
New release coming shortly, update to v8.5.1 and change futurerate_url to:

```yaml
futurerate_url: 'https://dataportal-api.nordpoolgroup.com/api/DayAheadPrices?date=DATE&market=N2EX_DayAhead&deliveryArea=UK&currency=GBP'
```

gcoan, 2024-10-17:
New release works perfectly, thanks Trefor for the quick fix

closed 2024-10-17 by gcoan