springfall2008/batpred #1537: Load kWh shows zero after updating to v8.5.0

kams19, 2024-10-15:
**Describe the bug**
The load value is displayed as 0.0 in the Plan. hence Predbat thinks it wont need to keep charge in the battery and start force discharging.

**Expected behavior**
Predbat should display correct load values as per estimation from past days like it did before upgrading to 8.5.0

**Predbat version**

8.5.0

**Environment details**

- Inverter and battery setup. - AIO 
- Standard HAOS installer or Docker - standard
- Anything else? No settings changed after upgrading 

**Screenshots**
[image]
[image]

**Log file**
Warn: Load mean is zero, unable to calculate divergence

kams19, 2024-10-15:
Rolling back to v8.4.10 solves the problem so all the versions released after that seem to have problem

[image]

springfall2008, 2024-10-15:
What is your load sensor set to?

kams19, 2024-10-15:
Where can I check that?

gcoan, 2024-10-15:
> Where can I check that?
> 

It's in your apps.yaml file, near the top

Also worth checking what your days_previous and days_previous_weight are set to

kams19, 2024-10-15:
Thanks. Here they are

load_filter_threshold 30
load_today
- sensor.givtcp_chnnnngnnn_load_energy_today_kwh = 0.1
import_today
- sensor.givtcp_chnnnngnnn_import_energy_today_kwh = 1.2

days_previous
- 7
- 14
- 21
days_previous_weight
- 1

Is this what you are looking for?

kams19, 2024-10-16:
> days_previous - 7 - 14 - 21
>
> Thanks. I see you have set days_previous to 7, 14 and 21 days. Have you
> increased the amount of history in Home Assistant you are retaining by
> changing purge_keep_days in your configuration.yaml? By default Home
> Assistant only keeps 10 days of history, e.g.
>
> recorder:
>   purge_keep_days: 14
>
> If you are not retaining enough history in Home Assistant then Predbat
> will have problems.
>
> Also suggest looking at your logfile and see what Predbat reports when
> loading the load history, e.g.
>
> 2024-10-16 06:20:00.367973: Predbat mode is set to Control charge & discharge
> 2024-10-16 06:20:00.748187: Found 12961 load_today datapoints going back 9 days
> 2024-10-16 06:20:01.147512: Car charging hold True threshold 6.0
> 2024-10-16 06:20:01.147576: Current data so far today: load 14.52 kWh import 29.1 kWh export 5.7 kWh pv 0.0 kWh
> ...
> 2024-10-16 06:20:01.793201: Historical data totals for days [2, 3, 4, 5, 6, 7, 8] are [16.33, 11.68, 24.46, 13.95, 19.4, 20.06, 20.4] - min 11.68
> 2024-10-16 06:20:01.793255: Model filter enabled - Discarding day 3 as it is the lowest of the 7 datapoints
> 2024-10-16 06:20:01.793380: Warn: Historical day 2 has 15 minutes of gap in the data, filled from 16.33 kWh to make new average 16.5 kWh (percent 99%)
> 2024-10-16 06:20:01.795939: Warn: Historical day 4 has 175 minutes of gap in the data, filled from 24.46 kWh to make new average 27.85 kWh (percent 88%)
> [... 3 lines cut ...]
> 2024-10-16 06:20:01.836622: Historical days now [2, 4, 5, 6, 7, 8] weight [1, 1, 1, 1, 1, 1]
> 2024-10-16 06:20:01.865541: Today's load divergence 112.77 % in-day adjustment 112.13 % damping 0.95x
> 2024-10-16 06:20:01.865597: Today's predicted so far 0.64 kWh with 0.0 kWh car/iBoost excluded and 4.26 kWh import ignored and 0 forecast extra.
> 2024-10-16 06:20:01.865612: Today's actual load so far 2.71 kWh with 4.32 kWh Car/iBoost excluded and 7.49 kWh import ignored.
> ...

No, I havent done that. However shouldnt that affect all versions of
Predbat and not the last two?

SwiftRR, 2024-10-16:
I have been affected by these historical day warnings for a long time.

Rightly or wrongly, I decide to increase my load_filter_threshold from the default of 30 to 60:

`  load_filter_threshold: 60`

I don't get the warnings now.
I think that Trefor hinted that the warnings could be in response to low load.

Rob

gcoan, 2024-10-16:
> I have been affected by these historical day warnings for a long time.

Rob, the issue isn't about warnings of gaps in the load history data, it's that @kams19 is getting no load history at all on the more recent versions of predbat.

I'm not having any such similar issues and nobody else is, which suggests it's an issue with the  configuration.

@kams19 you have days previous set to 7, 14, 21 but you've not increased the home assistant history retention so effectively you are only looking for day 7 history. Was there any issues with GivTCP a week ago?

Can you please share the similar lines from your logfile and try upgrading again to the latest predbat release to see what it logs about history. 

If you do want predbat to average your consumption from the same day for the last 3 weeks you will have to increase the setting in configuration.yaml, but same day à week ago should work fine

SwiftRR, 2024-10-16:
> Rob, the issue isn't about warnings of gaps in the load history data, it's that @kams19 is getting no load history at all on the more recent versions of predbat.

Deleted as off topic and could confuse.

Rob

kams19, 2024-10-18:
@gcoan Can it be an issue with config if it works with earlier versions? 
I have added the following lines to configuration.yaml now
recorder:
    purge_keep_days: 21

I will try to upgrade it again and see if works now.

closed 2025-01-01 by springfall2008