springfall2008/batpred #1538: PredHeat - Next Volume Temp

thewhale21, 2024-10-15:
**Is your feature request related to a problem? Please describe.**
Unable to manually set next volume temp. Can this please be added to Apps.yaml so those with a weather dependent curve can dynamically adjust.

**Describe the solution you'd like**
        next_volume_temp = sensor.volume_temp in apps.yaml and predheat.py updated to read this variable. 

**Describe alternatives you've considered**
We could manually edit the input number but this is over written each time predheat runs.

**Additional context**
Setting next flow temp to current flow temp makes it difficult for accurate planning.

springfall2008, 2024-10-15:
You could update it with an automation from the sensor, but I agree it would make sense if you have your own sensor to connect it directly

springfall2008, 2024-10-15:
Although I think I broke it with the integration to Predbat, will fix

springfall2008, 2024-10-15:
Fix on main for the next_volume_temp bug and also option to override with a sensor by setting 'volume_temp' in apps.yaml

thewhale21, 2024-10-16:
> Fix on main for the next_volume_temp bug and also option to override with a sensor by setting 'volume_temp' in apps.yaml

Have added the below into my apps.yaml. Seems the next temp is still being over written. 
    flow_temp: sensor.espaltherma_leaving_water_temp_before_buh_r1t
    flow_difference_target: 10
#    volume_temp: sensor.espaltherma_boiler_heating_target_temp
    volume_temp: 31

springfall2008, 2024-10-16:
Did you put volume_temp into the Predheat section in apps.yaml? Are you running v8.5.1?

thewhale21, 2024-10-16:
> Did you put volume_temp into the Predheat section in apps.yaml? Are you running v8.5.1?

Yup, Selected Main from predbat versions and it has self updated to 8.5.1, and volume_temp was added below flow_difference_target

springfall2008, 2024-11-02:
Can you share your apps.yaml settings for Predheat?

thewhale21, 2024-11-04:
Volume temp seems to have stabilized back to previous behavior. Might be a miss understanding on my part, But i'd expect the flow temp to track the current temp and then the volume temp to be the target leaving water temp so it knows how much "power" it needs to put into the water.

springfall2008, 2024-11-10:
>volume_temp: sensor.espaltherma_boiler_heating_target_temp

This seems wrong, the volume temperature should be the radiator water temperature ideally at a location close to your thermostat. If you don't have that sensor then its best not to set it.

thewhale21, 2024-11-10:
> > volume_temp: sensor.espaltherma_boiler_heating_target_temp
> 
> This seems wrong, the volume temperature should be the radiator water temperature ideally at a location close to your thermostat. If you don't have that sensor then its best not to set it.

Thanks, what should flow temp be set to in that case?

chalfontchubby, 2026-08-20:
Closing - the initial issue seems to have been closed - no responses in a while. Please open another issue if you are still seeing problems.

closed 2026-08-20 by chalfontchubby