springfall2008/batpred #1542: SolarEdge - Battery Full at X % (97%)

Noodleyman, 2024-10-16:
**Is your feature request related to a problem? Please describe.**
with SolarEdge systems, if a battery has 97, 98 99 or 100% state of charge then the system will consider the battery full. If asked to charge when at 98% for example, the system will not charge the remaining 2%. This can cause Predbat to get a little confused if it wants to charge the last 2% but it never achieves it. Instead, it will keep the system in charge mode which in short means the system will just be running off grid. Thus, it would be nice to make the "fully charge" state a configurable value.  "Consider battery full at X %" in this case 97%.

**Describe the solution you'd like**
Add a new configurable entity to define at what SoC the battery is considered full, and not a candidate for charging. in this case for SE systems, if the battery is at 97% or higher don't try and charge the last 3%.

**Describe alternatives you've considered**
Considered using an automation to set discharge mode to lower battery to 96% or lower, if at 97-100% when predbat.status is Charging, but this seems wasteful to cycle power in this way and is not yet implemented

**Additional context**
None

raldred, 2024-11-04:
I have a similar thing with my LuxPower inverter, it never reaches 100%, I made a template sensor to correct the battery SOC.

Use in place of the sensor you configure for `soc_percent` within your apps.yaml

Here's my example, you can create something similar for your system

`sensor.lux_battery` is the current battery SOC (%)
`sensor.lux_bms_limit_charge_live` is the BMS charge limit, which will be zero once the BMS has decided the battery is full and stops it charging (you may have to utilise another mechanism to determine it's full here)

So I basically have the condition, if BMS charge limit is 0 and the SOC is above 97 then it's considered 100%

```
{% set soc = states('sensor.lux_battery')|int %}
{% set charging_stopped = states('sensor.lux_bms_limit_charge_live')|float == 0 %}

{% if charging_stopped and soc > 97 %}
  100
{% else %}
  {{ soc }}
{% endif %}
```

PieBeard, 2024-12-03:
I find that when charging from a below 97% e.g. battery is at 80% and then charges it will charge to 100%, if the battery is used and 98% remains and then excess solar is available it doesn't charge the battery but exports it.  I think this is might be a form of battery protection so you are not constantly charge and discharging the top few percent.  When I drops below a certain figure it will then charge to 100% again.

cyberkryten, 2024-12-11:
Not all SE systems consider 96%+ full.  

Do you have one battery?  I have three and that makes this sort of functionality very complicated as it is possible that two batteries are at 100% and the third is at 90% which means my total SOC template calculation says 97%.  

It also makes any sort of charge rate calculation (slower charge when approaching 100%) no use.

Until recently (#1632), I've never had issues with charging all three to 100%

It is also worth looking at what your Maximum Energy, Available Energy and State of Health values are for the battery in the SEModbus Integration as that could be the cause.  

On mine, Max shows as 9.7kWh but the Available shows as 9.802kWh!  If there's a calculation somewhere of reaching 9.7kWh then that would show 100% before it is 'full' as the battery is reporting a higher figure as the actual maximum.

Noodleyman, 2025-02-27:
I wanted to follow up to say that I've gone with the templated solution, given I have a single battery and based upon my own analytics for the SOC threshold as charged.

I am using the following templated sensor:
```
  - platform: template
    sensors:
      solaredge_bespoke_battery_soc_for_predbat:
        friendly_name: "Solar Battery SOC for PredBat accounting for possible full at 97%"
        value_template: > 
          {% set soc = states('sensor.solar_b1_rounddown_soe')|int %}
          {% set charging_stopped = states('sensor.solaredge_battery_charge_rate')|float == 0 %}
          {% if charging_stopped and soc >= 96 %}
            100
          {% else %}
            {{ soc }}
          {% endif %}  
```

thank you @raldred  for sharing the idea. @cyberkryten , you also raise some very valid points. I think the underlying thing to really capture though is.. 

Is the battery done charging, even though it is in chargde mode and does not report 100%. If yes, then the sensor should report 100%. 

@gcoan , I opted to not use the suggested you made as I do use the reserves quite a lot in my setup as I have the off-grid backup module. I've built automations to set the reserve based upon weather alerts. it works really well. thus I didn't want to alter any config which references reserve as per the docs

The sensor can likely do that without evaluating the overall charge percentage. will keep it as is for now, it seems to work. Thank you all for your feedback.

chalfontchubby, 2026-08-29:
Thanks for the report. This issue has been open for over 6 months with no recent activity - we're volunteers working through a large backlog and need to filter what's still relevant. If this is still a problem for you, please let us know and we'll keep it open. Otherwise, we'll close it in a week (around 2026-09-05).

chalfontchubby, 2026-09-09:
Closing as flagged - no response to the note above, so we're assuming this is no longer live.

If you're still hitting this, please reopen or open a fresh issue with a current version number and a `predbat_debug_*.yaml` captured while it's happening; we can replay that directly against current code.

_Posted by Claude on Rik's behalf._

closed 2026-09-09 by chalfontchubby