springfall2008/batpred #1543: Warning: Error Downloading Futeratedata from cloud from yesterday

Redding43, 2024-10-16:
**Describe the bug**
Warning: Error Downloading Futeratedata from cloud

**Expected behavior**
No warning to show

This has started happening from yesterday after an update to the latest predbat. This may be coincidental and it could likely be the provider of the data is failing the call

**Predbat version**

v.8.5.0

**Environment details**

GivEnergy Battery, Octopus Agile Rates (Both Integrations)

**Screenshots**
[image]

**Log file**
2024-10-16 13:10:02.329335: Warn: You have not set pv_today in apps.yaml, you will have no previous pv data
2024-10-16 13:10:02.836963: Warn: Error downloading futurerate data from URL https://www.nordpoolgroup.com/api/marketdata/page/325?currency=GBP, code 404
2024-10-16 13:10:02.886812: Info: record_status Warn: Error downloading futurerate data from cloud
2024-10-16 13:10:03.014960: Warn: Error downloading futurerate data from URL https://www.nordpoolgroup.com/api/marketdata/page/325?currency=GBP, code 404
2024-10-16 13:10:03.058225: Info: record_status Warn: Error downloading futurerate data from cloud
2024-10-16 13:10:03.215850: Warn: Error downloading futurerate data from URL https://www.nordpoolgroup.com/api/marketdata/page/325?currency=GBP, code 404
2024-10-16 13:10:03.259973: Info: record_status Warn: Error downloading futurerate data from cloud
2024-10-16 13:10:03.260725: Warn: Error downloading futurerate data from URL https://www.nordpoolgroup.com/api/marketdata/page/325?currency=GBP
2024-10-16 13:10:03.306090: Info: record_status Warn: Error downloading futurerate data from cloud
2024-10-16 13:10:04.893173: Warn: Historical day 7 has 5 minutes of gap in the data, filled from 11.3 kWh to make new average 11.34 kWh (percent 100%)

Redding43, 2024-10-16:
Looks to be same issue as Issue #1536

springfall2008, 2024-10-17:
New release coming shortly, update to v8.5.1 and change futurerate_url to:

```yaml
futurerate_url: 'https://dataportal-api.nordpoolgroup.com/api/DayAheadPrices?date=DATE&market=N2EX_DayAhead&deliveryArea=UK&currency=GBP'
```

closed 2024-10-17 by gcoan