springfall2008/batpred #1548: Predbat not recognising "car_charging_planned" on Hypervolt H3P

silkit, 2024-10-17:
**Describe the bug**
I am on Intelligent Octopus Go. However, i have decided to remove my car from Octopus's automation and plan to use PredBat to control the charge at the low rate hours.
The reason why i have removed the car from the IOG integration is that Octopus are not giving me long enough charge slots to fully recharge from empty. I have raised this with them twice already but it has fallen on deaf ears.

Whilst i still have the octopus_intelligent_slot recorded in my YAML. I have followed the instructions and turned the switch.predbat_octopus_intelligent_charging switch to "off"

 A clear and concise description of what the bug is.
PredBat will not plan car charging as it does not pick up that the car is "plugged in"
**Expected behavior**
Expect the Predbat logs to reflect the car is "plugged in" as the other sensors confirm this.

**Predbat version**
v8.5.1

```
  car_charging_planned:
    - binary_sensor.hypervolt_car_plugged

  car_charging_planned_response:
    - 'Plugged in'

``` 
[image]

When looking at the logs, Predbat will not plan car charging as it is not plugged in.
```
2024-10-17 14:35:01.216756: Configuration: forecast_hours 48 num_cars 1 debug enable is False calculate_plan_every 10
2024-10-17 14:35:01.217650: Cars 1 charging from battery False planned [False], charging_now [False] smart [True], max_price [10.0], plan_time ['08:00:00'], battery size [18.819], limit [18.819], rate [3.5], exclusive [False]
2024-10-17 14:35:04.029265: Cars: SOC kWh: [0.18819] Charge limit [18.819] plan time ['08:00:00'] battery size [18.819]
2024-10-17 14:35:05.028509: Car 0 charging is not planned as it is not plugged in

``` 
Confirmed by nothing appearing in the plan:
[image]

silkit, 2024-10-17:
I also meant to add i have completely removed the charger from the Octopus intergration in HA as well as i wondered if this would still be causing an issue, but it does not seem to be the case:
[image]

mpartington, 2024-10-18:
Try putting that sensor into the states page in developer, it might be returning a slightly different response.  Otherwise the examples have " " rather than the ' ' in your config, so try "Plugged In" rather than 'Plugged in', see if that makes a difference

silkit, 2024-10-18:
Good thinking batman lol!
your 100% correct - it returned a completely different response. Something new I've learnt there then. I didn't realise that could happen
[image]
[image]

It looks like it is all working. PredBat has issued the plan straightaway as well ;-)
Thank you for your help and sharing your knowledge

closed 2024-10-18 by silkit