springfall2008/batpred #1549: Excess battery export during the evening peak.

bryanajwells, 2024-10-17:
**Is your feature request related to a problem? Please describe.**
Not problem related.

**Describe the solution you'd like**
I have OIG and Export, with predbat set to control the charge and discharge.
I have noticed that if the battery has plenty of charge at then and of the day, predbat will schedule a discharge :-)
However for the purposes of grid balancing it may be beneficial to schedule, at least part of, the discharge between 4 pm and 7pm (grid peek times).
Might need to tie this with a couple of other controls to define the time window.

**Describe alternatives you've considered**
I have considered changing the export rate in the apps.yaml for the 4-7 period to be slighly more than then normal export, say 16p, but this will mess with the savings.

**Additional context**

Nilogax, 2024-10-18:
As a counter I'd suggest that not exporting at peak will benefit you in the coming months. When Octopus start saving sessions again (to balance grid load when they most need it) they will compare your usage/export during the saving session with your normal usage/export over previous days, and the rate per KWh difference is generous.  Saving sessions usually occur during the peak period to shift load. 

Predbat will sign up to the saving session for you and set the appropriate high export rate, so your scheduled export will shift to take advantage of this. If you have your normal export at peak time (as is common on Flux) this will impact the benefit you get from saving sessions as that "normal" export will be deducted from whatever you export during the session to give the figure they use for your reward.

springfall2008, 2024-10-18:
If you enable the carbon sensor and add a small amount of p/Kg to your planning then Predbat will export when the grid carbon is the highest.

bryanajwells, 2024-10-18:
@Nilogax  - good point, dont want to compomise those sessions.
@springfall2008 - that would work, except today (and other days have been similar), grid carbon at 16:00-19:00 is arouns 190, but overnight its 340. So increasing the p/Kg would favour the overnight still.

closed 2024-12-15 by bryanajwells

springfall2008, 2024-12-16:
Saving sessions probably aren't worth much this year, but some people slightly lower their export rate during the times of saving sessions (5-7pm) to avoid exporting in this time