springfall2008/batpred #1550: Support for GivEnergy temperature restricted charge rate

dsteeley, 2024-10-18:
**Is your feature request related to a problem? Please describe.**
GivEnergy appear to have pushed firmware changes that restrict battery charge rates when temperature <20C. 
See this thread - https://community.givenergy.cloud/d/5096-battery-charge-power-27kw-instead-of-36kw/10. 

**Describe the solution you'd like**
Predbat should include the potential restricted charge rate in calculations. This may already be supported but the data on my system doesn't appear to include this. 

**Describe alternatives you've considered**
I do plan to add further insulation to my battery to mitigate this issue. A software workaround would be ideal in the meantime. 

**Additional context**
I'm unsure how we'd directly track the predicted temperature, potentially modelling exterior temperature, charge rate and other relevant factors to predict the batteries temperature and therefore whether throttling is enabled. 

I'd be very happy to contribute this enhancement if it's something you'd consider.

gcoan, 2024-10-18:
One solution could be to use the new Predbat manual API capability https://springfall2008.github.io/batpred/manual-api/ and write an automation that sets the inverter_limit_charge and inverter_limit_discharge based on the battery cell temperatures

dsteeley, 2024-10-21:
> One solution could be to use the new Predbat manual API capability https://springfall2008.github.io/batpred/manual-api/ and write an automation that sets the inverter_limit_charge and inverter_limit_discharge based on the battery cell temperatures

Thanks, I wasn't aware of that API. I'll take a look at this, my goal is to be predicting battery cell temperatures to inform Predbat the reduced charge rate threshold.

closed 2024-10-21 by dsteeley

gcoan, 2024-10-21:
> I'll take a look at this, my goal is to be predicting battery cell temperatures to inform Predbat the reduced charge rate threshold.

👍 if you get a solution that you are happy with, please do share and we can add it to the documentation