springfall2008/batpred #1551: Freeze Discharge broken on v8.4.11 onwards

mpartington, 2024-10-18:
**Describe the bug**
A few of us have noticed that the battery continued to charge when Predbat was in Freeze Discharge mode.  Digging into this further I can see that on my system Predbat is not changing the battery charge rate to zero, nor is it enabling the PauseCharge mode on the inverter.  If I roll back to versions 8.4.10 or earlier, this function works as expected.

This was the version where a toggle was added to switch off freeze discharge, so code change seems related

Additionally it seems to set a different status message of hold discharging, rather than freeze discharging when it is not working.

Can be replicated by selecting a manual force discharge freeze

**Expected behavior**
For the battery charge to be paused by setting the battery charge rate to zero and the inverter pause mode to PauseCharge

**Predbat version**

behaviour seen 8.4.11 onwards

**Environment details**

- AC3 2x 5.2kWh latest inverter firmware with pause Mode
- Standard HAOS installer 

**Screenshots**

[image]

**Log file**
Log file attached, however it covers testing across multiple versions to identify where the break happens.

mpartington, 2024-10-18:
[image]

gcoan, 2024-10-19:
Predbat 8.5.1

I'm seeing freeze discharging working fine, both battery pause is set and charge rate is set to zero correctly

[image]
[image]

mpartington, 2024-10-19:
You know what, it is for me now! I tried all sorts of downgrades, upgrades,
restarts and it was repeatable. Only thing new for me is a HA core update.

Will do some more testing later and probably close this off

gcoan, 2024-10-19:
🤷‍♂️

mpartington, 2024-10-19:
Ok, seems to have repeated it. I accidentally turned on discharge instead of freeze discharge. I cancelled that and tried freeze discharge and it's gone back to not setting pause or charge rate. The big vlusle.seems to be the status changes to hold discharging, rather than freeze discharging.

There must be a trigger sequence that upsets predbat.

[image]

Will.grab logs when home

mpartington, 2024-10-19:
[write_log_inv_1.log](https://github.com/user-attachments/files/17445765/write_log_inv_1.log)
[predbat.log](https://github.com/user-attachments/files/17445767/predbat.log)

mpartington, 2024-10-19:
**SOLVED**
And it is related to the introduction of the 'switch.predbat_set_discharge_freeze' entity.

If this is turned OFF, and you select manual force discharge freeze, then it sets Hold Discharging (but doesn't set pause or change charge rate to zero)

If it is turned ON and you select manual force discharge freeze, then it sets Hold Discharging and it **DOES** set pause and change charge rate to zero

Now as a user I would intuitively assume that if I select an override, then that will override any other setting (which it doesn't), so I would say this is still a bug. However it is something I could live with either way now I understand it.

Trefor, please close (or add to improvement queue) depending on your view

gcoan, 2024-10-19:
I have to admit I think I've seen freeze charge not working as it should do when I set a manual session but never looked into it.

Can you please check if there is a similar pattern with freeze charge @springfall2008 ?

springfall2008, 2024-10-19:
Well tracked down and pushed a fix on 'main' for testing.

I've started to add unit testing for the execute plan part of the code and introduce this fix along with some other fixes/simplifications. Now set_charge_freeze/set_discharge_freeze has no impact on the plan execution it will only change if the plan will include these actions.

mpartington, 2024-10-22:
And the plans:
[image]
[image]

nbullus, 2024-10-22:
I'm on 15p export and was definitely getting Freeze Discharges prior to testing the 'main' version. Worth Trefor taking another look to see why it's stopped Freeze Discharging at the 15p export rate.

mpartington, 2024-10-28:
Still.not seeing any Automatic freeze discharge on my plans, but the subject of this ticket is fixed.

closed 2024-10-28 by mpartington