springfall2008/batpred #1553: Hold Charge Behaviour Not As Documented

nbullus, 2024-10-19:
**Describe the bug**

The Hold Charge behaviour is not as documented and confusing:

1. Documentation states that Hold Charge slots should not be explicitly selected, yet Hold Charge slots are being explicitly planned in advance (see two examples in screenshot).
2. Documentation states that Hold Charge is set where the target SOC is the same as the current SOC, yet I see Hold Charge slots with a target SOC lower than the current SOC. How can the target SOC be achieved given the battery discharge will be paused under a Hold Charge?

**Expected behavior**

Hold Charge should behave as per the documentation or the documentation updated. In the attached screenshot, the two Hold Charge slots should have a target SOC that matches the current SOC, i.e. hold battery at current SOC level.

**Predbat version**

v8.5.1

**Environment details**

- GivEnergy AIO
- Standard HAOS installer
- Predbat Addon

**Screenshots**

[image]

**Log file**

Not applicable

springfall2008, 2024-10-19:
I think what it means is its a coincidence in that a charge level has need picked which is the same as the prediction SOC, but when you get to that time its unlikely to be the case

nbullus, 2024-10-20:
Apologies Trefor, but I'm still struggling to understand this. The way I'm reading the above plan is that the 5:00-5:30 HoldChrg slot will be 58% SOC at the start of the slot but will lower to 56% and held by the end at 5:30. But given this will put a PauseDischarge in place immediately, the SOC will still be 58% at 5:30 at not 56% as the plan indicates. It also indicates that the cost of this Hold Charge slot is 0p due to it using 2% from the battery , but if the battery is paused for the duration of the slot then it will be drawing from grid and there will be a cost associated.

I may be misunderstanding what the Hold Charge is meant to be doing. I can understand if it plans to let the SOC reduce from 58% to 56% first and then it puts the PauseDischarge in place to hold it at 56% for the remainder of the slot. However I'm not sure if it does this or whether it would set the PauseDischarge at the start of the slot and therefore hold the SOC at 58% for the duration and pull from the grid for the whole slot.

chalfontchubby, 2026-08-20:
Can this be closed?

nbullus, 2026-08-20:
Yes Rik.

closed 2026-08-20 by chalfontchubby