springfall2008/batpred #1556: Exception raised float() argument must be a string or a real number, not 'NoneType'

mikejreading, 2024-10-20:
Error: Exception raised float() argument must be a string or a real number, not 'NoneType'

`
4391 | 2024-10-20 13:05:03.706742: Error: float() argument must be a string or a real number, not 'NoneType'
-- | --
4390 | 2024-10-20 13:05:03.706653: Info: record_status Error: Exception raised float() argument must be a string or a real number, not 'NoneType'
4388 | TypeError: float() argument must be a string or a real number, not 'NoneType'
4375 | 2024-10-20 13:05:03.689749: Error: Traceback (most recent call last):
4374 | 2024-10-20 13:05:03.689512: Error: Exception raised float() argument must be a string or a real number, not 'NoneType'
4279 | 2024-10-20 13:05:01.735262: Warn: Inverter 0: Using inverter type SK - not all features are available
4244 | 2024-10-20 13:01:04.175602: Error: float() argument must be a string or a real number, not 'NoneType'
4243 | 2024-10-20 13:01:04.175515: Info: record_status Error: Exception raised float() argument must be a string or a real number, not 'NoneType'
4241 | TypeError: float() argument must be a string or a real number, not 'NoneType'
4228 | 2024-10-20 13:01:04.159383: Error: Traceback (most recent call last):
4227 | 2024-10-20 13:01:04.159042: Error: Exception raised float() argument must be a string or a real number, not 'NoneType'
4118 | 2024-10-20 13:00:01.932925: Warn: Inverter 0: Using inverter type SK - not all features are available
4083 | 2024-10-20 12:56:51.428122: Error: float() argument must be a string or a real number, not 'NoneType'
4082 | 2024-10-20 12:56:51.428030: Info: record_status Error: Exception raised float() argument must be a string or a real number, not 'NoneType'
4080 | TypeError: float() argument must be a string or a real number, not 'NoneType'
4067 | 2024-10-20 12:56:51.412139: Error: Traceback (most recent call last):
4066 | 2024-10-20 12:56:51.411891: Error: Exception raised float() argument must be a string or a real number, not 'NoneType'
4030 | 2024-10-20 12:56:46.310607: Warn: Inverter 0: Using inverter type SK - not all features are available

`

`2024-10-20 13:21:27.559139: Export threshold used for optimisation was 14.0p
2024-10-20 13:21:27.681890: High export rate windows [0: 10-20 13:00:00 - 10-20 13:30:00 @ 15.0, 1: 10-20 13:30:00 - 10-20 14:00:00 @ 15.0, 2: 10-20 14:00:00 - 10-20 14:30:00 @ 15.0, 3: 10-20 14:30:00 - 10-20 15:00:00 @ 15.0, 4: 10-20 15:00:00 - 10-20 15:30:00 @ 15.0, 5: 10-20 15:30:00 - 10-20 16:00:00 @ 15.0, 6: 10-20 16:00:00 - 10-20 16:30:00 @ 15.0, 7: 10-20 16:30:00 - 10-20 17:00:00 @ 15.0, 8: 10-20 17:00:00 - 10-20 17:30:00 @ 15.0, 9: 10-20 17:30:00 - 10-20 18:00:00 @ 15.0, 10: 10-20 18:00:00 - 10-20 18:30:00 @ 15.0, 11: 10-20 18:30:00 - 10-20 19:00:00 @ 15.0, 12: 10-20 19:00:00 - 10-20 19:30:00 @ 15.0, 13: 10-20 19:30:00 - 10-20 20:00:00 @ 15.0, 14: 10-20 20:00:00 - 10-20 20:30:00 @ 15.0, 15: 10-20 20:30:00 - 10-20 21:00:00 @ 15.0, 16: 10-20 21:00:00 - 10-20 21:30:00 @ 15.0, 17: 10-20 21:30:00 - 10-20 22:00:00 @ 15.0, 18: 10-20 22:00:00 - 10-20 22:30:00 @ 15.0, 19: 10-20 22:30:00 - 10-20 23:00:00 @ 15.0, 20: 10-20 23:00:00 - 10-20 23:30:00 @ 15.0, 21: 10-20 23:30:00 - 10-21 00:00:00 @ 15.0, 22: 10-21 00:00:00 - 10-21 00:30:00 @ 15.0, 23: 10-21 00:30:00 - 10-21 01:00:00 @ 15.0, 24: 10-21 01:00:00 - 10-21 01:30:00 @ 15.0, 25: 10-21 01:30:00 - 10-21 02:00:00 @ 15.0, 26: 10-21 02:00:00 - 10-21 02:30:00 @ 15.0, 27: 10-21 02:30:00 - 10-21 03:00:00 @ 15.0, 28: 10-21 03:00:00 - 10-21 03:30:00 @ 15.0, 29: 10-21 03:30:00 - 10-21 04:00:00 @ 15.0, 30: 10-21 04:00:00 - 10-21 04:30:00 @ 15.0, 31: 10-21 04:30:00 - 10-21 05:00:00 @ 15.0, 32: 10-21 05:00:00 - 10-21 05:30:00 @ 15.0, 33: 10-21 05:30:00 - 10-21 06:00:00 @ 15.0, 34: 10-21 06:00:00 - 10-21 06:30:00 @ 15.0, 35: 10-21 06:30:00 - 10-21 07:00:00 @ 15.0, 36: 10-21 07:00:00 - 10-21 07:30:00 @ 15.0, 37: 10-21 07:30:00 - 10-21 08:00:00 @ 15.0, 38: 10-21 08:00:00 - 10-21 08:30:00 @ 15.0, 39: 10-21 08:30:00 - 10-21 09:00:00 @ 15.0, 40: 10-21 09:00:00 - 10-21 09:30:00 @ 15.0, 41: 10-21 09:30:00 - 10-21 10:00:00 @ 15.0, 42: 10-21 10:00:00 - 10-21 10:30:00 @ 15.0, 43: 10-21 10:30:00 - 10-21 11:00:00 @ 15.0, 44: 10-21 11:00:00 - 10-21 11:30:00 @ 15.0, 45: 10-21 11:30:00 - 10-21 12:00:00 @ 15.0, 46: 10-21 12:00:00 - 10-21 12:30:00 @ 15.0, 47: 10-21 12:30:00 - 10-21 13:00:00 @ 15.0, 48: 10-21 13:00:00 - 10-21 13:30:00 @ 15.0, 49: 10-21 13:30:00 - 10-21 14:00:00 @ 15.0, 50: 10-21 14:00:00 - 10-21 14:30:00 @ 15.0, 51: 10-21 14:30:00 - 10-21 15:00:00 @ 15.0, 52: 10-21 15:00:00 - 10-21 15:30:00 @ 15.0, 53: 10-21 15:30:00 - 10-21 16:00:00 @ 15.0, 54: 10-21 16:00:00 - 10-21 16:30:00 @ 15.0, 55: 10-21 16:30:00 - 10-21 17:00:00 @ 15.0, 56: 10-21 17:00:00 - 10-21 17:30:00 @ 15.0, 57: 10-21 17:30:00 - 10-21 18:00:00 @ 15.0, 58: 10-21 18:00:00 - 10-21 18:30:00 @ 15.0, 59: 10-21 18:30:00 - 10-21 19:00:00 @ 15.0, 60: 10-21 19:00:00 - 10-21 19:30:00 @ 15.0, 61: 10-21 19:30:00 - 10-21 20:00:00 @ 15.0, 62: 10-21 20:00:00 - 10-21 20:30:00 @ 15.0, 63: 10-21 20:30:00 - 10-21 21:00:00 @ 15.0, 64: 10-21 21:00:00 - 10-21 21:30:00 @ 15.0, 65: 10-21 21:30:00 - 10-21 22:00:00 @ 15.0, 66: 10-21 22:00:00 - 10-21 22:30:00 @ 15.0, 67: 10-21 22:30:00 - 10-21 23:00:00 @ 15.0, 68: 10-21 23:00:00 - 10-21 23:30:00 @ 15.0, 69: 10-21 23:30:00 - 10-22 00:00:00 @ 15.0, 70: 10-22 00:00:00 - 10-22 00:30:00 @ 15.0, 71: 10-22 00:30:00 - 10-22 01:00:00 @ 15.0, 72: 10-22 01:00:00 - 10-22 01:30:00 @ 15.0, 73: 10-22 01:30:00 - 10-22 02:00:00 @ 15.0, 74: 10-22 02:00:00 - 10-22 02:30:00 @ 15.0, 75: 10-22 02:30:00 - 10-22 03:00:00 @ 15.0, 76: 10-22 03:00:00 - 10-22 03:30:00 @ 15.0, 77: 10-22 03:30:00 - 10-22 04:00:00 @ 15.0, 78: 10-22 04:00:00 - 10-22 04:30:00 @ 15.0, 79: 10-22 04:30:00 - 10-22 05:00:00 @ 15.0, 80: 10-22 05:00:00 - 10-22 05:30:00 @ 15.0, 81: 10-22 05:30:00 - 10-22 06:00:00 @ 15.0, 82: 10-22 06:00:00 - 10-22 06:30:00 @ 15.0, 83: 10-22 06:30:00 - 10-22 07:00:00 @ 15.0, 84: 10-22 07:00:00 - 10-22 07:30:00 @ 15.0, 85: 10-22 07:30:00 - 10-22 08:00:00 @ 15.0, 86: 10-22 08:00:00 - 10-22 08:30:00 @ 15.0, 87: 10-22 08:30:00 - 10-22 09:00:00 @ 15.0, 88: 10-22 09:00:00 - 10-22 09:30:00 @ 15.0, 89: 10-22 09:30:00 - 10-22 10:00:00 @ 15.0, 90: 10-22 10:00:00 - 10-22 10:30:00 @ 15.0, 91: 10-22 10:30:00 - 10-22 11:00:00 @ 15.0, 92: 10-22 11:00:00 - 10-22 11:30:00 @ 15.0, 93: 10-22 11:30:00 - 10-22 12:00:00 @ 15.0, 94: 10-22 12:00:00 - 10-22 12:30:00 @ 15.0, 95: 10-22 12:30:00 - 10-22 13:00:00 @ 15.0, 96: 10-22 13:00:00 - 10-22 13:30:00 @ 15.0]
2024-10-20 13:21:27.697678: Import threshold used for optimisation was 4.7p
2024-10-20 13:21:27.814698: Low import rate windows [0: 10-20 13:00:00 - 10-20 13:30:00 @ 1.98, 1: 10-20 13:30:00 - 10-20 14:00:00 @ 1.07, 2: 10-20 14:00:00 - 10-20 14:30:00 @ 1.99, 3: 10-20 14:30:00 - 10-20 15:00:00 @ 1.99, 4: 10-20 15:00:00 - 10-20 15:30:00 @ 2.2, 5: 10-20 15:30:00 - 10-20 16:00:00 @ 2.11, 6: 10-20 20:30:00 - 10-20 21:00:00 @ 4.14, 7: 10-20 21:30:00 - 10-20 22:00:00 @ 1.32, 8: 10-20 22:00:00 - 10-20 22:30:00 @ 4.41, 9: 10-20 22:30:00 - 10-20 23:00:00 @ -0.7, 10: 10-20 23:00:00 - 10-20 23:30:00 @ -0.7, 11: 10-20 23:30:00 - 10-21 00:00:00 @ -0.7, 12: 10-21 03:00:00 - 10-21 03:30:00 @ 4.41, 13: 10-21 03:30:00 - 10-21 04:00:00 @ 0.0, 14: 10-21 04:30:00 - 10-21 05:00:00 @ -0.64, 15: 10-21 05:00:00 - 10-21 05:30:00 @ 1.76, 16: 10-21 05:30:00 - 10-21 06:00:00 @ -0.8, 17: 10-21 06:00:00 - 10-21 06:30:00 @ 0.66, 18: 10-21 06:30:00 - 10-21 07:00:00 @ 0.78, 19: 10-21 07:00:00 - 10-21 07:30:00 @ 0.0, 20: 10-21 07:30:00 - 10-21 08:00:00 @ 1.54, 21: 10-21 08:00:00 - 10-21 08:30:00 @ 2.2, 22: 10-21 08:30:00 - 10-21 09:00:00 @ 4.41, 23: 10-21 09:00:00 - 10-21 09:30:00 @ 0.17, 24: 10-21 09:30:00 - 10-21 10:00:00 @ 2.2, 25: 10-21 10:00:00 - 10-21 10:30:00 @ 0.0, 26: 10-21 10:30:00 - 10-21 11:00:00 @ 1.41, 27: 10-21 11:00:00 - 10-21 11:30:00 @ 0.0, 28: 10-21 11:30:00 - 10-21 12:00:00 @ 1.9, 29: 10-21 12:00:00 - 10-21 12:30:00 @ 1.17, 30: 10-21 12:30:00 - 10-21 13:00:00 @ 1.71, 31: 10-21 13:00:00 - 10-21 13:30:00 @ 1.98, 32: 10-21 13:30:00 - 10-21 14:00:00 @ 1.07, 33: 10-21 14:00:00 - 10-21 14:30:00 @ 1.99, 34: 10-21 14:30:00 - 10-21 15:00:00 @ 1.99, 35: 10-21 15:00:00 - 10-21 15:30:00 @ 2.2, 36: 10-21 15:30:00 - 10-21 16:00:00 @ 2.11, 37: 10-21 20:30:00 - 10-21 21:00:00 @ 4.14, 38: 10-21 21:30:00 - 10-21 22:00:00 @ 1.32, 39: 10-21 22:00:00 - 10-21 22:30:00 @ 4.41, 40: 10-21 22:30:00 - 10-21 23:00:00 @ -0.7, 41: 10-21 23:00:00 - 10-21 23:30:00 @ -0.7, 42: 10-21 23:30:00 - 10-22 00:00:00 @ -0.7, 43: 10-22 03:00:00 - 10-22 03:30:00 @ 4.41, 4 [... cut ...]
    self.write_and_poll_value("charge_limit", self.base.get_arg("charge_limit", indirect=False, index=self.id), soc)
  File "/config/inverter.py", line 1243, in write_and_poll_value
    matched = abs(float(current_state) - new_value) <= fuzzy
TypeError: float() argument must be a string or a real number, not 'NoneType'

2024-10-20 13:21:27.864571: Info: record_status Error: Exception raised float() argument must be a string or a real number, not 'NoneType'
2024-10-20 13:21:27.864663: Error: float() argument must be a string or a real number, not 'NoneType'
`

springfall2008, 2024-10-20:
The error here is that 'charge_limit' in apps.yaml isn't pointing to something that is a number you are getting None instead. What did you set it to?

springfall2008, 2024-10-27:
Can you go to the web interface and select apps.yaml page and see what shows against charge_limit?

mikejreading, 2024-11-23:
This was my error - I hadn't set the inverter integration properly

closed 2024-11-23 by mikejreading