springfall2008/batpred #1558: Long Term Stats

icornish72, 2024-10-20:
We have generated statistics for 'predbat foxess h1 g2 0 charge limit' (sensor.predbat_foxess_h1_g2_0_charge_limit) in the past, but it no longer has a state class, therefore, we cannot track long term statistics for it anymore.

Statistics cannot be generated until this entity has a supported state class.
If the state class was previously provided by an integration, this might be a bug. Please report an issue.
If you previously set the state class yourself, please correct it. The different state classes and when to use which can be found in the developer documentation.
If the state class has permanently been removed, you may want to delete the long term statistics of it from your database.
Do you want to permanently delete the long term statistics of sensor.predbat_foxess_h1_g2_0_charge_limit from your database?

**Describe the bug**
LONG term stats do not retain a specific sensor value as it has undefined units

**Expected behavior**
No error on the Long Term Statistics data

V8.5.1

xxxx

**Environment details**

- Inverter and battery setup
Foxes H1 G2 + 4* ECS 4800 cubes 
- standard haos add in
- ha 2024.10.3

onetoomany, 2024-10-22:
ditto for "sensor.predbat_sf_0_charge_limit" - I guess the SF_0 is the inverter type and inverter_id

gcoan, 2024-10-22:
I don't think these will be changed.

I have a GivEnergy inverter (so native Predbat support) and apart from the sensor.predbat_pv_* solar forecast entities, none of my predbat sensors have long term statistics.

To be honest I'm not sure why you'd want long term stats on things like Predbat's setting of the charge limit.   Yes maybe there is a case to keep some LTS on your inverter data (SoC level, import/export kWh, etc) but those would come natively from your inverter not Predbat.

I suspect the inclusion of LTS for your charge limits was an aberration that has been corrected and so you can just delete all the stats that are now orphaned

onetoomany, 2024-10-24:
the problem is that this long term stat keep reappearing after I delete it. Is there a way of permanently killing them?

icornish72, 2024-10-24:
I have just reviewed my recorder config, and will feedback in the next 24-48 hours.

closed 2025-11-02 by gcoan