springfall2008/batpred #1559: Code error for "predbat_status" sensor

silkit, 2024-10-21:
**Describe the bug**
In the Predbat log. In order to calculate the discharge curve. It is looking for the entity "predbat_status"
However, the entity name that exists is "**predbat.status**"

**Predbat version**
v8.5.1

**Environment details**
- Fox H1 G2 & 5x EC411
- Standard HAOS installer on RPI4

**Screenshots**
Log file looking for "predbat_status"
```
2024-10-20 21:01:37.351983: Note: Cannot find battery charge curve (no final curve), one of the required settings for predbat_status, soc_kw, battery_power and charge_rate do not have history, check apps.yaml
``` 

Entity name as it exists in HA:
[image]
[image]

**Log file**
[PredBat Log.txt](https://github.com/user-attachments/files/17457443/PredBat.Log.txt)

gcoan, 2024-10-21:
No problem, I have just fixed the code, all of the different logging messages from the curve calculations said "charge curve" regardless of whether it was a charge or discharge curve being calculated !

Also corrected predbat_status.

The fix will be in the next PR I issue, thanks for raising it

gcoan, 2024-10-21:
If you want to try the fixed code, copy inverter.py from my fork https://github.com/gcoan/batpred/tree/main/apps/predbat over the top of the existing inverter.py and restart Predbat, it should produce the right messages now

silkit, 2024-10-21:
Oh dear....to many "curves" lol.
I can confirm it's all fixed. I copied the code over and the changes work as expected.
It displays the correct named entity and the correct error around discharge curve and doesn't go onto say that it cannot find the charge curve.
Thanks for that fix :-)

silkit, 2024-10-21:
Erm. Slight issue. Just noticed a new error shown up in the logs since replacing the inverter.py file.
Whilst Predbat had restarted itself due to the change of the configuration. I noticed it came up with an new error about not being able to find the charge_stop service. But when i looked at the GUI it was not coloured red to show an error.
[image]
I then when into my add-ons and force restarted Predbat. At this point it comes up with this error and predbat won't load.
```
-----------------------------------------------------------
Bootstrap Predbat
Startup
 Add-on: Predbat
 Home Battery Prediction and Control
-----------------------------------------------------------
 Add-on version: 1.2.1
 You are running the latest version of this add-on.
 System: Home Assistant OS 13.2  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2024.10.3
Predbat files are installed correctly for version v8.5.1
 Home Assistant Supervisor: 2024.10.2
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
Traceback (most recent call last):
  File "/config/hass.py", line 5, in <module>
    import predbat
  File "/config/predbat.py", line 76, in <module>
    from inverter import Inverter
ImportError: cannot import name 'Inverter' from 'inverter' (/config/inverter.py)
s6-rc: info: service base-addon-banner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service base-addon-timezone: starting
s6-rc: info: service base-addon-log-level: starting
s6-rc: info: service fix-attrs successfully started
[11:29:22] INFO: Configuring timezone (Europe/London)...
Shutdown, sleeping 20 seconds before restarting
s6-rc: info: service base-addon-log-level successfully started
s6-rc: info: service base-addon-timezone successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[11:29:44] INFO: Predbat init script running
Running Predbat inside Add-on
Your API key is: b6449471b3a67b601b3a23534a50c7c50d52db91dfa100157fd81095611809ca680f8f0c2e609deb8effb5886facfd71398bc9dc15c2f81d
Bootstrap Predbat
Startup
Predbat files are installed correctly for version v8.5.1
Traceback (most recent call last):
  File "/config/hass.py", line 5, in <module>
    import predbat
  File "/config/predbat.py", line 76, in <module>
    from inverter import Inverter
ImportError: cannot import name 'Inverter' from 'inverter' (/config/inverter.py)
Shutdown, sleeping 20 seconds before restarting
``` 
I have copied back the original Inverter.py from the main fork and Predbat restarts fine. Tried again your revised .py and a force restart of Predbat and it won't start again. Original file copied back and Predbat starts fine again.

gcoan, 2024-10-21:
I think the problem is that my code fork is of Predbat 8.4.11 and you're running 8.5.1.  Almost certainly there's something been changed by Trefor between the two versions that means running just inverter.py from 8.4.11 alongside the rest of Predbat 8.5.1 doesn't work.  I know there has been rewriting and more unit testing added to the freeze discharge logic so this might be it.

If you want to try copying the entire of my code fork across and restart the add-on, then it should work OK although you'll be downgraded to 8.4.11.

Once I PR this in to the core codebase it should be OK as most of my changes are to the documentation, only a couple of minor code changes (of which this is one)

silkit, 2024-10-21:
Ok yep i understand then. That makes sense.
I'm not too sure how to copy the whole fork across (i'm ok with doing single copy & pastes but other than that, i'm not experienced enough as to how to change the forks in the settings)
I'll go back to leaving it with the error for the time being noting if doesn't affect anything and that the test reflects the changes work and wait until the main fork gets updated with the details.
Thanks again

gcoan, 2024-10-21:
Yeah you don't want to change the fork in the settings, bad things might happen!  You could manually copy each of the six python files across, and in fact it might just be predbat.py you need, but as you say, we've proven the messages are now correct in my fix and we know that it's just a log message issue.

I'll PR it before the end of the week, just want to finish some other documentation changes off first

closed 2025-01-01 by springfall2008