springfall2008/batpred #1562: Improve the accuracy of the nord data agile prediction

gcoan, 2024-10-22:
**Is your feature request related to a problem? Please describe.**
I've noticed that whilst the nord data next day market prices give a broad indication of the shape of the agile prices, there's always a variation.

I was thinking that it was due to regional pricing, my agile rates are always 0.7p more than @swiftrr in Yorkshire and the nord data feed and predbat calculator doesn't reflect that, so thinking that what is needed is a regional adder option.

But looking at the prices on a number of days, it's not as simple as that.

October 12th predicted 
[image] 

12th actual 
[image]

13th predicted 
[image]

13th actual 
[image]

20th predicted
[image]

20th actual 
[image]

21st predicted 
[image]

21st actual 
[image]

**Describe the solution you'd like**
Reviewing the predictions against actuals I believe:

- there is a regional adder that isn't currently included that if there was, could improve prediction accuracy 
- negative price predictions are overstated. Eg -12.98p predicted for  3:30 Monday 21st, actually was -0.92p. I think there's some damping that Octopus apply 
- it probably needs an indepth comparison of actual to prediction to work out the adjustment factors 

Maybe the agile prices prediction API could help improve the prediction accuracy https://prices.fly.dev

SwiftRR, 2024-10-22:
To add to comment from @gcoan, I live in Yorkshire region and I believe that @gcoan is in Eastern England region.

The table below compares Agile rates for these 2 regions between 19:00 and 22:30 today (23-10-2024).
[image]
Source: https://files.energy-stats.uk/csv_output (showing historical Agile rate 2018-2024)

You can see the differences between the rates across regions are greater when the Agile rates are higher.

I would presume that the Nordpool prediction is for the whole GB.

For rates close to the fixed export rate of 15p, this may result in Predbat making a different prediction dependent of the region.

Rob

gcoan, 2024-10-23:
Thanks Rob

Another example of why improving the accuracy of the nord data prediction is important:

Tonight's norddata prediction is for rates to drop overnight tonight and it to be cheaper to charge the battery up than to export the solar generation today.  So Predbat is scheduling my battery to be on freeze discharge for most of the day, only letting it half fill up.

[image]
[image]

But if that forecast is a penny or two out, the time at which the cheaper rates come in is slightly different, then it would have been better to keep some of the generated solar rather than exporting it all.

So for me today I will put some force idle's in to give a bit more buffer space and retain a bit more solar. Less efficient but a bit safer

SwiftRR, 2024-10-23:
For comparison, this is my plan from 11:50 am

[image]

I will be interesting to see how the Nordpool predictions stand up after 4 pm when the Octopus rates come through ........ and to see Predbat's predictions then.

Rob

SwiftRR, 2024-10-23:
And a direct comparison. We have different Nordpool predictions with different Predbat predictions. So are the Nordpools Regional?

[image]

Rob

gcoan, 2024-10-23:
> We have different Nordpool predictions with different Predbat predictions. So are the Nordpools Regional?
> 

Ah I think I remember that the nordpool data is used to create a series of delta's on the 'roll forward' prices which of course do incorporate regional variation.  So the way predbat works already may give some of the regional pricing variation I thought was needed, but I think there is still opportunity for accuracy improvement in the predictions.

SwiftRR, 2024-10-23:
And here's the end of my analysis. Nordpool predictions against Octopus actuals:

[image]

Although the two sets of data are in the same ballpark, I am surprised at some of the differences. Not quite 'all over the place', but look at 23:00, 00:30 and 02:30.

Still better though than just carrying forwards rates from previous 24 hours.

Rob

gcoan, 2024-10-23:
I think I've posted enough before and after screenshots already, but for completeness here is my actual Octopus rates for tonight

[image]

Comparing the same 3 times Rob did:

23:00 predicted 14.11, actual 18.08
00:30 predicted 15.46, actual 11.47
02:30 predicted 13.33, actual 9.48

A lot of variation, both up and down

SwiftRR, 2024-10-23:
Going right back to the first post of @gcoan for this issue:
> **Is your feature request related to a problem? Please describe.**
> I've noticed that whilst the nord data next day market prices give a broad indication of the shape of the agile prices, there's always a variation.

From our analysis, it seems that Nordpool predictions indicate the direction of travel of future rates. However, the error range is such that some actual rates are higher and some lower than the Nordpool predictions. These differences can be significant.
The Nordpool predictions do seem to take into account regional variations in rates (Ofgem).

Can Predbat take these rather random variations into account, or is Predbat doing as much as it can by considering the direction and extent of travel of the rates?

Rob

gcoan, 2024-10-23:
Agreed Rob,

> Can Predbat take these rather random variations into account, or is Predbat doing as much as it can by considering the direction and extent of travel of the rates?

But I'd go a step further and ask an alternative question:

Should Predbat be basing its charge and discharge activity for today (e.g. freeze discharge and export all solar, force discharge the battery) based on the expectation that the norddata derived rates are 100% correct when we now clearly see that they are not.

Would it be better for Predbat to apply some form of 'uncertainty factor' to the norddata future rates to recognise the inherent deviations in projected to actual rates that are being seen and the plan be more conservative as a result?
Maybe this won't be the optimum plan all the time but better to let the battery charge unnecessarily and then have to force discharge it than let the solar be exported and have to be imported later on at a higher rate than norddata predicted,  or hold onto the charge and not export it, maybe need to export it later on when the future agile rates are known?

SwiftRR, 2024-10-23:
With both Nordpool and Octopus values, we have bracketed values based on losses.
For me, the bracketed cost is typically about 1 p higher.
From our analysis today (and experience from recent days), the Nordpool values can be significantly 'out' (±4p today).

There could be a case for showing the bracketed Nordpool cost as a ± value
But this would likely confuse many users and I could just imagine a deluge of queries. 

More questions than answers!

Rob

gcoan, 2024-10-27:
Hi @springfall2008 I see you have introduced improvements with the nord data handling in 8.5.2 #1567

Looking at the change history it looks like some of the documentation changes you made for this new feature got lost in energy-rates.md:

[image]

springfall2008, 2024-10-27:
I think this has been fixed with today's release, please check?

gcoan, 2024-10-27:
> I think this has been fixed with today's release, please check?

Thanks Trefor for the enhancements, I am away from tomorrow for a few weeks so I won't have the chance to upgrade until I get back

springfall2008, 2024-11-02:
Please open a new ticket in case of issues

closed 2024-11-02 by springfall2008

gcoan, 2024-11-02:
> Please open a new ticket in case of issues

I have installed 8.5.3 and checked the nord data and actual Agile prices for today and they seem to be a lot closer than they were before. Quite a few prices were predicted to within a decimal place of the actual price, and where there were variations they were at most a couple of pence.  The pre-4pm charge plan and the post-4pm charge plan were quite similar.

Will need to separately look at negative agile prices as none coming through at the moment, but this looks a big improvement, thanks