springfall2008/batpred #1566: Predbat time labels in the plan don't reflect DST change

gcoan, 2024-10-26:
**Describe the bug**
Low priority as it only happens twice a year.

The Octopus Agile rates correctly reflect the double 1:00 and 1:30 time slots as the clocks go back from BST to UTC:
[image]
[image]

The Predbat plan however labels the rates at the wrong times, the second 1:00 slot is labelled as 2:00, the 2:00 rate as 3:00 etc

[image]

**Expected behavior**
Correct handling of DST changes with correct rates reflected 

I suspect it'll sort itself out after 2:00

**Predbat version**
8.5.1

**Environment details**
HAOS 2024.10.3

springfall2008, 2024-10-27:
Yes right, the plan is showing times as minutes from midnight but ignores the time change

springfall2008, 2026-11-14:
Automated first-pass triage (a maintainer will review before any action is taken).

**Bug:** the plan's row times are built as `midnight_utc + timedelta(minutes=…)` and printed without re-resolving the timezone, so on a clock-change night every slot from the point the clocks change onward is labelled an hour ahead — the prices themselves are right, only the labels are wrong.

**You're on v8.5.1; the latest release is v9.7.11.** Updating won't clear this one — I checked the defect against current main and it is still there — but there have been a lot of fixes since, so it's still worth updating.

What I checked, against current main (`v9.7.11`, 59018b15):

- The two halves run on different clocks, which is why the prices are right and the labels are not. Rate minute-keys are true elapsed minutes from local midnight: `str2time()` keeps each slot's own UTC offset and `minute_data()` subtracts the aware timestamps, so the doubled 01:00/01:30 (BST then GMT) and everything after it sit at their real positions — minute 60/90 = 01:00/01:30 BST, minute 120 = the second 01:00, 01:00 GMT. That matches the Octopus sensor exactly, as in your screenshots.
- The renderer then adds those minute offsets to `self.midnight_utc` (`publish_html_plan` in `output.py`), which on the night of the change still carries the pre-change offset (00:00 BST) and is never re-resolved. I reproduced the offset arithmetic directly with pytz: minute 120 prints `02:00 BST+0100` where the real local time is 01:00 GMT, and minute 180 prints `03:00` for the 02:00 rate. That is your screenshot exactly, and it clears at the next run at or after the change, as you suspected.
- The same stamp build is used by the web plan table (it publishes the same wrong wall-clock/offset pair as `row.time`, and the UI renders that pair), the rate / cost / load / charge-limit / export-limit chart series, and the low/high rate and charge/export window start/end sensor attributes, so those share the shift. Some of the write path is protected and some is not: the control ledger converts to an absolute epoch (`.timestamp()`) and is fine, but `adjust_force_export()` / `adjust_charge_window()` render the same fixed-offset datetimes as `HH:MM` strings for the inverter, which is a real control bug on a change night and is tracked separately in #5749.
- Nothing has fixed it since: the History view already does the right thing — it re-labels each rebuilt row with `local_tz.normalize(view_midnight + timedelta(minutes=slot_minute))`, added with the billing work (PR #5926) — and PR #6084's clock-change helpers (`start_of_local_day()`, `clock_change_minutes()`, `local_day_end_minute()`) are in the same area, but the live plan/report path has none of it.
- No test run: no existing module covers the label path. `test_clock_change_day` covers the "today so far" reads from #5760, not the row labels.

Fix shape: re-resolve each built stamp in the configured zone before formatting it — `self.local_tz.normalize(dt)`, guarded with `hasattr(self.local_tz, "normalize")` as the History path does, or `dt.astimezone(self.local_tz)`. That leaves the instant untouched and only corrects the printed wall clock, so everything that compares instants (the web UI's `new Date(row.time)`, debug-snapshot matching, billing) is unaffected. Expect the repeated hour to show two rows labelled 01:00 and 01:30; distinguishing them by offset or abbreviation is a display choice, not needed for correctness. This is labels only — the 23/25-hour day axis (a later run re-anchoring `midnight_utc` so the day's first hour drops off) is a separate design question, tracked in #6083.

Related: #5708 (same symptom on v9.4.1, closed as a duplicate of this issue), #6083 (the rest of the clock-change family — savings baseline, day-ago load history, chart starts), #5749 (the write side — charge/export windows sent to the inverter as an hour-out `HH:MM`), #3688/#3694 (the spring side), and PR #5729 (open; the same "resolve the date, then attach the zone" policy, but for manual override times — it does not touch the plan labels).

I've left your `bug` and `priority_low` labels as they were; this issue is already queued for a fix under `BOT_PR`.