springfall2008/batpred #1570: Repeated messages for auto calibration of Nordpool predictions in v8.5.2 log files

SwiftRR, 2024-10-27:
**Describe the bug**
 I upgraded to v8.5.2 at 12:25pm and I get this message every 5 mins when the predbat plan updates:
```
2024-10-27 12:25:21.642205: Info: Calibrating Nordpool data... adjust_import True adjust_export False
```
The predbat.log shows several lines each time that seem to be identical.
```
2024-10-27 14:45:03.556550: Fetching futurerate data from https://dataportal-api.nordpoolgroup.com/api/DayAheadPrices?date=DATE&market=N2EX_DayAhead&deliveryArea=UK&currency=GBP
2024-10-27 14:45:03.557062: Return cached futurerate data for https://dataportal-api.nordpoolgroup.com/api/DayAheadPrices?date=2024-10-27&market=N2EX_DayAhead&deliveryArea=UK&currency=GBP age 139.7 minutes
2024-10-27 14:45:03.558194: Return cached futurerate data for https://dataportal-api.nordpoolgroup.com/api/DayAheadPrices?date=2024-10-28&market=N2EX_DayAhead&deliveryArea=UK&currency=GBP age 139.7 minutes
2024-10-27 14:45:03.571260: Loaded 97 datapoints of futurerate analysis
2024-10-27 14:45:03.609465: Info: Calibrating Nordpool data... adjust_import True adjust_export False
2024-10-27 14:45:03.933420: Calibration for import best diff 2975.772000000019 multiply 1.92 add_peak 11.5 add_all 0.0 
2024-10-27 14:45:03.945736: Calibration for export best diff 9999999 multiply 0.95 add_peak 1.0 add_all 5.0 
2024-10-27 14:45:03.959367: Predicted future rates: ['10-27 00:00:00 => 19.17 / 14.74', '10-27 01:00:00 => 19.41 / 14.86', '10-27 02:00:00 => 16.89 / 13.61', '10-27 03:00:00 => 14.52 / 12.43', '10-27 04:00:00 => 15.4 / 12.87', '10-27 05:00:00 => 18.18 / 14.25', '10-27 06:00:00 => 23.26 / 16.76', '10-27 07:00:00 => 22.36 / 16.31', '10-27 08:00:00 => 19.58 / 14.94', '10-27 09:00:00 => 20.38 / 15.33', '10-27 10:00:00 => 18.69 / 14.5', '10-27 11:00:00 => 16.37 / 13.35', '10-27 12:00:00 => 15.42 / 12.88', '10-27 13:00:00 => 16.87 / 13.6', '10-27 14:00:00 => 17.38 / 13.85', '10-27 15:00:00 => 18.24 / 14.28', '10-27 16:00:00 => 35.78 / 18.03', '10-27 17:00:00 => 38.44 / 19.35', '10-27 18:00:00 => 34.55 / 17.42', '10-27 19:00:00 => 18.65 / 14.48', '10-27 20:00:00 => 17.4 / 13.86', '10-27 21:00:00 => 16.87 / 13.6', '10-27 22:00:00 => 15.26 / 12.8', '10-27 23:00:00 => 14.62 / 12.48', '10-28 00:00:00 => 14.52 / 12.43', '10-28 01:00:00 => 13.39 / 11.87', '10-28 02:00:00 => 12.06 / 11.22', '10-28 03:00:00 => 11.09 / 10.74', '10-28 04:00:00 => 12.14 / 11.25', '10-28 05:00:00 => 15.12 / 12.73', '10-28 06:00:00 => 18.43 / 14.37', '10-28 07:00:00 => 19.66 / 14.98', '10-28 08:00:00 => 18.93 / 14.62', '10-28 09:00:00 => 18.87 / 14.59', '10-28 10:00:00 => 19.72 / 15.01', '10-28 11:00:00 => 19.01 / 14.66', '10-28 12:00:00 => 19.13 / 14.72', '10-28 13:00:00 => 20.28 / 15.28']
2024-10-27 14:45:03.996840: Adding rate rates_import_override: {'date': '2024-10-20', 'start': '13:00:00', 'end': '14:00:00', 'rate': 0, 'load_scaling': 1.0} => 10-27 13:00:00 to 10-27 14:00:00 @ 0.0 date 2024-10-20 00:00:00 increment False
2024-10-27 14:45:04.882344: Rate min forward looking: now 10.91 at end of forecast 10.91

```
I guess this is the 'auto calibration' but do you think it is working properly?

The predbat.log is here: 
[predbat.log](https://github.com/user-attachments/files/17534750/predbat.log)

I have also copied the Nordpool predictions just before 4 pm and the Octopus rates just after 4 pm so you can compare and access the accuracy of the predictions.

[image]

[image]

**Expected behaviour**
The Nordpool message in logs not being there!

**Predbat version**
v8.5.2

**Environment details**
- Inverter and battery setup: GE AC3 connected to two 8.2 kWh batteries
- Standard HAOS installer: HAOS 13.1 with Core 2024.10.2 running on raspberry pi 4 
- Anything else? Nothing

Rob

springfall2008, 2024-10-27:
I think its all working as expected, but yes I can probably turn off the log message in the next version. I put it there for debug in case people had issues.

gcoan, 2024-10-27:
Since export is false (not on agile export), there's no point in doing the calculation and logging the calibration for export rates.

And presumably vice versa if import is false

SwiftRR, 2024-10-27:
Yes, please turn off the debug as it is clogging up the logs.

Other than that, v8.5.3 (and for me v8.5.2) is working just fine with pretty good agreement between Nordpool predictions and actual rates for Agile compared with previous versions.

Note the comments from @goan about export. The default in apps.yaml: futurerate_adjust_export is False. So any check of export is unnecessary bandwidth. 

Rob

springfall2008, 2024-11-02:
Fixed in the latest release

closed 2024-11-02 by springfall2008