springfall2008/batpred #1573: Debugging to get preheat working

mogons57, 2024-10-29:
**Is your feature request related to a problem? Please describe.**
I'm struggling to get preheat working. Followed instructions in the docs (several times) but nothing happening.

**Describe the solution you'd like**
Some way of working out what I've got wrong with the setup so it starts working.

**Describe alternatives you've considered**
I've tried everything I can think of to get the preheat setup in apps.yaml correct [i.e. problems with sensors] but still no luck. When I do a search for predheat entities the only one that appears is "switch.predbat_predheat_enable". Done a complete HA reboot but nothing improves.

**Additional context**
Running v8.5.3 and the enable preheat switch is on. Using Predbat 1.2.1 add on. The only mention of preheat in the logs is the line that says Watching ['./prediction.py', './predheat.py', ....  In the web log file there is no mention of preheat at all.

springfall2008, 2024-11-02:
Can you attach your apps.yaml?

springfall2008, 2024-11-05:
The Predheat line needs to be intended within Predbat, I wonder if you have put this at the same level as Predbat instead?

springfall2008, 2024-11-05:
e.g.

```yaml
predbat:
   ...
   predheat:
      ...
```

springfall2008, 2024-11-11:
> Oops that was easy - all working now, thanks. I just checked the web page configuration guide and it does not show the predheat: indented. Maybe that should be made a bit clearer.

Yeh, I did notice that and made a fix in the latest release to the docs

closed 2024-11-11 by springfall2008