springfall2008/batpred #1574: No Export available after upgrading to Outgoing octopus 12M fixed tariff

Dipstikk, 2024-10-30:
Predbat cannot set up an export time.

**Expected behaviour**
A clear and concise description of what you expected to happen.
I would expect an export session to be available in the plan

**Predbat version**

8.5.3

**Environment details**

- Inverter and battery setup
- AC3.0 2x8.2 batteries
- Standard HAOS installer or Docker
- Standard
- Anything else?
**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.
[image]

**Log file**Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please 
[Copy of Predbat Log File.docx](https://github.com/user-attachments/files/17569558/Copy.of.Predbat.Log.File.docx)
enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

Dipstikk, 2024-10-30:
Hi Mike, I do not have a template line in my apps.yaml file, it’s one of the first things i removed when I first set it up, note, Predbat has been working absolutely perfectly for over 12 months.

Please check attached yaml to see if you can find the template line, i also notice that the predbat log file I collected last night is dated 02/09/2024, not current, is this correct?
# ------------------------------------------------------------------
# This is an example configuration, please modify it
# ------------------------------------------------------------------
---
pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # Timezone to work in
  timezone: Europe/London

  # Set to auto-match with a GivEnergy serial number, but you can override the serial or the sensor names
  # if it doesn't work or if you have more than one inverter you will need to list both
  geserial: 're:sensor.givtcp_(.+)_soc_kwh'
  #geserial2: 're:sensor.givtcp2_(.+)_soc_kwh'
  
  # ha_url: 'http://192.168.0.65:8123'

  #
  # Sensors, more than one can be specified and they will be summed up automatically
  #
  # For two inverters the load today would normally be the master load sensor only (to cover the entire house)
  # If you have three phase and one inverter per phase then you would need three load sensors
  #
  # For pv_today if you have multiple solar inverter inputs then you should include one entry for each inverter
  #
  load_today:
    - sensor.givtcp_{geserial}_load_energy_today_kwh
  import_today:
    - sensor.givtcp_{geserial}_import_energy_today_kwh
  export_today:
    - sensor.givtcp_{geserial}_export_energy_today_kwh
  pv_today:
    - sensor.givtcp_{geserial}_pv_energy_today_kwh

  # Load forecast can be used to add to the historical load data (heat-pump)
  # To link to Predheat
  # Data must be in the format of 'last_updated' timestamp and 'energy' for incrementing kWh
  #load_forecast:
  #  - predheat.heat_energy$external
  #
  # If you enable ge_cloud_data then the load/import and export data will be fetches from the GE cloud instead of from GivTCP sensors
  # this is usually less efficient and of course prone to internet downtime, but could be useful if you lost your GivTCP data
  # Set the serial to the inverter serial to pull the data from and the key to your API key
  # When this is set load_today, import_today and export_today are not used
  #
  # ge_cloud_data: False
  # ge_cloud_serial: '{geserial}'
  # ge_cloud_key: 'xxxx'
  #
  # Controls/status - must by 1 per inverter
  #
  num_inverters: 1
  #
  # Run balance inverters every N seconds (0=disabled) - only for multi-inverter
  # balance_inverters_seconds: 60
  #
  # When set use the REST API rather than HA entity for control, should be more reliable/faster to control
  # Set one per inverter
  # If using Docker then change homeassistant.local to the Docker IP address
  givtcp_rest:
   - 'http://192.168.0.65:6345'
   - 'http://homeassistant.local:6346'

[... 245 lines cut ...]
  export_triggers:
    - name: 'large'
      minutes: 60
      energy: 1.0
    - name: 'small'
      minutes: 15
      energy: 0.25

  # If you have a sensor that gives the energy consumed by your solar diverter then add it here
  # this will make the predictions more accurate. It should be an incrementing sensor, it can reset at midnight or not
  # It's assumed to be in Kwh but scaling can be applied if need be
  #iboost_energy_today: 'sensor.xxxxx'
  #iboost_energy_scaling: 1.0

  # Nordpool market energy rates
  #futurerate_url: 'https://www.nordpoolgroup.com/api/marketdata/page/325?currency=GBP'
  #futurerate_adjust_import: False
  #futurerate_adjust_export: True
  #futurerate_peak_start: "16:00:00"
  #futurerate_peak_end: "19:00:00"
  #futurerate_peak_premium_import: 14
  #futurerate_peak_premium_export: 6.5
  
  # Carbon Intensity date from National Grid
  carbon_intensity: 're:(sensor.carbon_intensity_uk)'

mpartington, 2024-10-30:
Just realised the log you shared is 2024-08-28, so you need to upload a more recent version. Have you changed from the app Daemon to add on version as the location changed
[image]

Dipstikk, 2024-10-30:
Hi Mike,
I haven't moved to the add on version so its as I built it up, I have just powered off my Pi and am waiting for it to restart to see if that clears things a bit, I thought I might restore a backup that was prior to the Octopus change.
Update, just got into the Pi, HA and Predbat and the log file is exactly the same so I am going to do a restore and see what happens, I have had grief with GivTCP 3.x.x. throwing predbat figures and graphs all over the place.

mpartington, 2024-10-30:
It might be worth trying the add on. One less point of failure. Think there are some instructions in the documentation. You'd just need to copy the apps.yaml to the new location and stop App Daemon

Dipstikk, 2024-10-30:
Oh well resetting back to a month ago hasn’t helped, I remember now when the add on came out I gave it a go but it failed, a few days later a new version was released as there was an issue with the original files and I have never got around to installing it, I’ll give it a go later on.
I’ll let you know how it goes.

SwiftRR, 2024-10-30:
I think you need to go back to square one.

Follow the instructions in the instructions step-by-step:
https://springfall2008.github.io/batpred/install/#

I recommend going too the Predbat standalone add-on, leaving appdeamon well behind (as @mpartington has suggested):
https://springfall2008.github.io/batpred/install/#predbat-add-on-install

 Hope that helps

Rob

mpartington, 2024-10-31:
>  I remember now when the add on came out I gave it a go but it failed

You might be grabbing the logs from the 'add on' location then, which would explain why they are so old. They are stored in a different folder location for appdaemon (I couldn't tell you everywhere, but I think it's down a level).  This would include the location of your apps.yaml file.

Dipstikk, 2024-10-31:
That is exactly correct. Just to be clear i can just copy and paste my exiting apps.yaml file into the new directory, i’m questioning this as i have had a read of the newer apps.yaml and there is quite a bit more in it.

Dipstikk, 2024-11-01:
Ok, update, I have installed the predbat add on and it’s working, I have found that I had a couple of cross overs in Solcast, which I have sorted and a couple of error messages regarding the comments on Octopus ready time and octopus charge limit, I have commented both out for the time being to get rid of the errors, however I still do not have export in the plan, I have noticed this comment in a box at the bottom of the apps.yaml file viewed through predbat, see attached screen grab.
[image]

Any suggestions on how it can be enabled? Or do I need to contact GE?

mpartington, 2024-11-01:
What you have shared looks good. I'm not sure if some of the octopus sensors had a name change recently, which may be the cause of the errors. However the important ones seem to be picked up correctly.  

Scheduled discharge enable is just turned on by Predbat as needed, so off is ok. However on mine,.I don't have the subsequent 'on's . Is something wrong with apps.yaml here I wonder?

Feels like you are getting closer. Does the plan show enough battery left to export?

[image]

SwiftRR, 2024-11-01:
This is better and looks like you are getting there.

As with @mpartington, I also do not have the three 'on' lines. This is mine:
[image]

Please upload the complete apps.yaml file and we can then see exactly what you have.
Within file editor, download the file and then attach the file to your reply.

Rob

gcoan, 2024-11-01:
> What you have shared looks good. I'm not sure if some of the octopus sensors had a name change recently, which may be the cause of the errors

There were a couple of the Octopus Intelligent Go sensors renamed in the latest Octopus integration (v13.0.0).  The new sensor names are in the documentation and the latest apps.yaml template

Dipstikk, 2024-11-02:
Ok, another possibly rookie error, it has been suggested I download and send the apps.yaml file, when I select download on the file, well the 3 dots, I get the error message, 401 not authorised, what the heck, surely I am admin of my own Pi?
In the mean time I’ll check the file names in the Octopus app although I am seeing the cheap rate night time figures.
Again thanks for all you comments and help.

mpartington, 2024-11-02:
I often get that as well and don't really understand why. I've installed the samba add on, which means I can browse the files on a laptop on the same network. Makes it easy to just copy them across to GitHub.

Upload a new log file and screenshot of your plan as well (,showing up to the next charge period). We may be able to spot something.

[image]

SwiftRR, 2024-11-02:
On your 401 error, I don't know! I am still graduating from my HA rabbit level but there are others more knowledgable than me who may be able to help here.

This is what I see when I select the vertical 3 dots:
[image]

If this fails, you can post the whole apps.yaml, as you did before in your first post.

Make sure that you add 3 back ticks above and below : ```
Otherwise you may get some lines in bold text.

Or you could try selecting all your apps.yaml code, copying into a text editor, saving that and uploading the file here.

Do check out the suggestion above from @gcoan.

Rob

SwiftRR, 2024-11-02:
I have looked right back to the start and one question from @mpartington:

> Its strange its doing anything at all as there are no calculations in the logs, you don't have 2 instances running at the same time do you (the add on and AppDaemon)?

You have copied a screen from the webUI so you must have the standalone Predbat installed. 
But do you still have the Appdaemon version installed?
**I am wondering whether you are accessing the Appdaemon apps.yaml file in File Editor.**
The original Word doc that you uploaded references to standalone Predbat.

Check the documentation again:
https://springfall2008.github.io/batpred/install/#editing-configuration-files-in-home-assistant

When you open File Editor, you will see the screen below. Select <- Home Assistant:
[image]

Now Select addon_configs
[image]

If you are Standalone, you should see this:
[image]

And then:
[image]
This is the apps.yaml file that Predbat standalone will be using.

If you are on appdaemon, you reach the Appdaemon-Predbat apps.yaml file via another route.
https://springfall2008.github.io/batpred/install/#predbat-add-on-install

Once you are on standalone Predbat, you can uninstall the Appdaemon version. When I did this, I noticed that the config files were still there (from File Editor). I then deleted the Appdaemon-Predbat files.

Before deleting things, I would take a backup!

Hope there is something here to help. Sorry if I have meme,ised things that you have already done but we need to get to the root of the problem.

Rob

Dipstikk, 2024-11-02:
Hi Rob, yes I get add-on_configs then 6adb4f0d_predbat, which I am accessing the apps.yamel file from, but under the addon_configs I get a0d7b954_appdaemon, I take it I can delete this file set?

SwiftRR, 2024-11-02:
Yes, you can delete this do please backup first. I would hate to have given you false advice that can't be undone!
First uninstall your old Appdaemon-Predbat add-on which will have been in here:
[image]

I have also uninstalled Appdaemon but you may be using this for other applications.

Then at least you will have a clean slate.

Then get a copy of your current apps-yaml file to us somehow!

Rob

gcoan, 2024-11-02:
> Hi Rob, yes I get add-on_configs then 6adb4f0d_predbat, which I am accessing the apps.yamel file from, but under the addon_configs I get a0d7b954_appdaemon, I take it I can delete this file set?

As long as the appdaemon add-on is not started, there is no issue with leaving all the appdaemon files there.  I would suggest leaving them in place so (a) you can refer to them if you need to and (b) it gives an easy way to revert back from the predbat addon.  Once you are absolutely settled with the predbat addon running fine, delete the files and delete the appdaemon addon

On the 401 error you got, it may be a permissions thing with how you were trying to download it.  HA partitions addons so that you can't by default access other addon folders in /addon_configs

If you use either file editor, studio server or SSH then you should be OK as it sets the global permissions needed

SwiftRR, 2024-11-02:
Follow advice from @gcoan.
He has been my mentor since I embarked on Predbat. 

Rob

Dipstikk, 2024-11-02:
[Predbat log.txt](https://github.com/user-attachments/files/17607513/Predbat.log.txt)
[Apps yaml.txt.txt](https://github.com/user-attachments/files/17607514/Apps.yaml.txt.txt)

Here you go, if yo can find an error please let me know

springfall2008, 2024-11-02:
You might want to edit the apps.yaml to remove your solcast API key

Dipstikk, 2024-11-02:
Thanks Trefor.

Dipstikk, 2024-11-02:
[Predbat log.txt](https://github.com/user-attachments/files/17607947/Predbat.log.txt)
[Apps yaml.txt.txt](https://github.com/user-attachments/files/17607948/Apps.yaml.txt.txt)

Here you are, if you can find an issue with them please let me know.
Thank you, Dave.

Dipstikk, 2024-11-02:
Just to let you all know my export through Predbat has burst into life, thank you all.

closed 2024-11-02 by Dipstikk

SwiftRR, 2024-11-02:
Well done.

Do you know what the issue was?

Rob

Dipstikk, 2024-11-03:
Hi Rob,
The only thing I could put it down to was I hadn’t upgraded predbat to the new add on, since doing that viewing the apps yaml file showed me sections that were highlighted in red meaning they were either wrong or were not needed so I commented out the two lines Octopus_ready_time and Octopus_charge_limit and set a few things in the Config that were showing red to base settings then left it, a couple of hours later I received a txt saying Predbat status was set to idle and from that point forward the battery graph showed sets of discharge oh and a couple of lines have appeared in the apps.yaml as per the attached screen dump.
[image]

#gotthereintheend

Now to watch Trefor’s video on tweeking for performance.