springfall2008/batpred #1575: service "stop charging" called at least every 5 minutes although no change in plan is happening. possible to call only on change of plan?

eddysteurs, 2024-10-31:
**Is your feature request related to a problem? Please describe.**
I got predbat working yesterday with calling services to soliscloud (a HA script with python script "solis control"
but the "stop charging" service is called at least every 5 minutes (sometime sooner) but there is no change in the predbat plan
So it is reconfiguring the solis inverter without need.

**Describe the solution you'd like**
Only call one of the 4 services when there is an actual plan change

**Describe alternatives you've considered**
I made 4 different HA scripts stop - start - charge - discharge
I keep now a state (helper) in HA and do test in script if actual service call is identical to this previous helper state, in that case I only make a message in HA but do not call the python script to update the solis inverter.
This is in my opinion logic that should come from predbat.
Or am i missing something in the predbat configuration?

**Additional context**
Add any other context or screenshots about the feature request here.
Trigger callback for input_number set_value
2024-10-31 13:40:08.264363: Info: record_status Idle
2024-10-31 13:45:03.898762: Warn: Inverter 0: Using inverter type MINE - not all features are available
2024-10-31 13:45:05.324354: Info: Web socket service script/set_solis_inverter_config_predbat_stop_charge socket for url http://supervisor/core/api/websocket
2024-10-31 13:45:05.403994: Info: record_status Idle
2024-10-31 13:50:05.390989: Warn: Inverter 0: Using inverter type MINE - not all features are available
2024-10-31 13:50:06.944103: Info: Web socket service script/set_solis_inverter_config_predbat_stop_charge socket for url http://supervisor/core/api/websocket
2024-10-31 13:51:00.834797: Warn: Inverter 0: Using inverter type MINE - not all features are available
2024-10-31 13:51:00.869265: Info: Web socket service script/set_solis_inverter_config_predbat_stop_charge socket for url http://supervisor/core/api/websocket
2024-10-31 13:51:00.969776: Info: record_status Idle
2024-10-31 13:55:04.752418: Warn: Inverter 0: Using inverter type MINE - not all features are available
2024-10-31 13:55:07.105759: Info: Web socket service script/set_solis_inverter_config_predbat_stop_charge socket for url http://supervisor/core/api/websocket
2024-10-31 13:55:07.209508: Info: record_status Idle
2024-10-31 14:00:04.638913: Warn: Inverter 0: Using inverter type MINE - not all features are available
2024-10-31 14:00:06.145597: Info: Web socket service script/set_solis_inverter_config_predbat_stop_charge socket for url http://supervisor/core/api/websocket
2024-10-31 14:00:06.245402: Info: record_status Idle
2024-10-31 14:05:04.414100: Warn: Inverter 0: Using inverter type MINE - not all features are available
2024-10-31 14:05:05.973973: Info: Web socket service script/set_solis_inverter_config_predbat_stop_charge socket for url http://supervisor/core/api/websocket
2024-10-31 14:05:58.265933: Warn: Inverter 0: Using inverter type MINE - not all features are available
2024-10-31 14:05:58.299219: Info: Web socket service script/set_solis_inverter_config_predbat_stop_charge socket for url http://supervisor/core/api/websocket
2024-10-31 14:05:58.376177: Info: record_status Idle
2024-10-31 14:10:04.193197: Warn: Inverter 0: Using inverter type MINE - not all features are available
2024-10-31 14:10:05.745162: Info: Web socket service script/set_solis_inverter_config_predbat_stop_charge socket for url http://supervisor/core/api/websocket
2024-10-31 14:10:05.815804: Info: record_status Idle
2024-10-31 14:15:04.428770: Warn: Inverter 0: Using inverter type MINE - not all features are available
2024-10-31 14:15:05.925929: Info: Web socket service script/set_solis_inverter_config_predbat_stop_charge socket for url http://supervisor/core/api/websocket
2024-10-31 14:15:06.006115: Info: record_status Idle
2024-10-31 14:20:05.138117: Warn: Inverter 0: Using inverter type MINE - not all features are available
2024-10-31 14:20:06.629613: Info: Web socket service script/set_solis_inverter_config_predbat_stop_charge socket for url http://supervisor/core/api/websocket
2024-10-31 14:21:00.164518: Warn: Inverter 0: Using inverter type MINE - not all features are available
2024-10-31 14:21:00.200851: Info: Web socket service script/set_solis_inverter_config_predbat_stop_charge socket for url http://supervisor/core/api/websocket
2024-10-31 14:21:00.284302: Info: record_status Idle
2024-10-31 14:25:03.638905: Warn: Inverter 0: Using inverter type MINE - not all features are available
2024-10-31 14:25:05.113213: Info: Web socket service script/set_solis_inverter_config_predbat_stop_charge socket for url http://supervisor/core/api/websocket
2024-10-31 14:25:05.182341: Info: record_status Idle
2024-10-31 14:30:03.986161: Warn: Inverter 0: Using inverter type MINE - not all features are available
2024-10-31 14:30:05.434073: Info: Web socket service script/set_solis_inverter_config_predbat_stop_charge socket for url http://supervisor/core/api/websocket
2024-10-31 14:30:05.509650: Info: record_status Idle
2024-10-31 14:35:06.802037: Warn: Inverter 0: Using inverter type MINE - not all features are available
2024-10-31 14:35:08.400459: Info: Web socket service script/set_solis_inverter_config_predbat_stop_charge socket for url http://supervisor/core/api/websocket
2024-10-31 14:36:00.670063: Warn: Inverter 0: Using inverter type MINE - not all features are available
2024-10-31 14:36:00.703925: Info: Web socket service script/set_solis_inverter_config_predbat_stop_charge socket for url http://supervisor/core/api/websocket
2024-10-31 14:36:00.784735: Info: record_status Idle
2024-10-31 14:40:04.220180: Warn: Inverter 0: Using inverter type MINE - not all features are available
2024-10-31 14:40:05.651788: Info: Web socket service script/set_solis_inverter_config_predbat_stop_charge socket for url

springfall2008, 2024-11-01:
Can you please try my fix on 'main' and see if it works for you?

eddysteurs, 2024-11-01:
Copied the predbat.py and inverter.py to my install and started predbat
Its now running about 50 minutes and it only called the service at the start,  nothing anymore since then.
So it seems to work.
I also just added a manual charge period and the service call start charge came through, so I think everything looks great

Thanks

            Eddy Steurs
            エディスタルス
            Asterlaan 16, 2950 Kapellen
             Phone +32 473 630 377
             Email: ***@***.******@***.***>

springfall2008, 2024-11-02:
Fixed in 8.5.4

closed 2024-11-02 by springfall2008