springfall2008/batpred #1578: FR - GivEnergy battery maintenance mode to better assist with SOC recalculations

thewookiewon, 2024-11-01:
**Is your feature request related to a problem? Please describe.**
GivEnergy batteries are known for the occasional SOC issue if batteries do not frequently get below 50% and stay idle for 30 minutes or longer under than threshold, I'd like to have an option toggle on/off that will force Predbat to on a weekly basis dump the entire battery just before a charge slot.

**Describe the solution you'd like**
 I'd like to have an option toggle on/off that will force Predbat to on a weekly basis dump the entire battery just before a charge slot.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Manually execute my above idea which is someitmes hard to remember with a growing family :).

**Additional context**
None

gcoan, 2024-11-01:
You could achieve what you are asking for by an automation calling the select.predbat_manual controls - setting a series of forced discharges and then a forced idle.

Would need a bit of logic to determine when to kick off the series of forced discharges depending on the SoC level, but this would then be entirely under your control rather than another switch and control to maintain in the core predbat product.

Happy to share your automation in the predbat documentation once done

thewookiewon, 2024-11-01:
A lot of that is beyond my technical level 😩.

gcoan, 2024-11-02:
Opportunity to learn then !

Seriously though, there's a lot of good tutorials and videos on Home Assistant and it's not actually that difficult to learn.  I'd recommend the Smart Home Junkie channel on YouTube, he does excellent videos on HA and has a recent series on Jinja and YAML that explain how you can write loops and conditional logic.

Have a look at the predbat documents and in particular understand how the manual predbat controls work, these enable you to instruct Predbat to force a particular task in a half hour slot.

Here's a simple automation that forces Predbat to Idle mode at 3pm and 3:30pm every day (it used to keep discharging just before the peak and this was to stop that behaviour)

You'll need to set the times for the select.predbat_manual_force_discharge control but study this and it should give you the idea of how to force the discharge and then the idle.

Then its a matter of triggering it when you want to, or just run it manually ad-hoc

```
alias: "Predbat: stop 3pm discharge"
description: ""
mode: single
triggers:
  - at: "23:00:00"
    trigger: time
actions:
  - data:
      option: "15:00:00"
    target:
      entity_id:
        - select.predbat_manual_idle
    action: select.select_option
  - delay:
      minutes: 1
  - data:
      option: "15:30:00"
    target:
      entity_id: select.predbat_manual_idle
    action: select.select_option
```

thewookiewon, 2024-11-02:
I'm not in the mind frame to learn at, hand surgery so still recouping. I still think it's a good FR for the project and coming from experience of having a faulty battery it makes sense.

Maybe I might automate it in the future but certainly not in my current state.

mpartington, 2024-11-04:
I raised a couple of similar FRs ( #686 and #1428 ). I think to more flexibly automate using the overrides, it really needs the PauseBoth option adding. Otherwise if you wanted to pause at say 20%, and the SOC adjusts to a higher or lower value, then it could either continue to discharge or charge (depending on pause mode chosen) and you have to start again.

To get around this, i started to manually force slot discharges so it hits empty inside the cheap rate, then force at least another 30 mins slot for the idle. However its a bit of a brute force approach and relies on having a long cheap rate period to get back to 100% (which I don't).  

More recently (now I get 15p export) I have just let Predbat discharge to near empty every night (it automatically schedules this anyway). I'm on 3017 firmware and I've had no SOC issues, so doesn't seem to need the OCVs anymore.

Others (non Predbat users) have come up with an automation to do this for IOG, so it could be possible to use that approach and put Predbat into read only.

chalfontchubby, 2026-08-29:
With a bit of tweaking to predbat_manual_soc - or similar - to set a maximum soc, the automations could be simplified a lot and predbat can plan the discharge down to sat 40% at midnight rather than a forced discharge. I'll try that.

chalfontchubby, 2026-08-29:
Opened #4841 - adds `select.predbat_manual_soc_max`, a discharge-direction ceiling sibling to `select.predbat_manual_soc` (which is documented as a minimum, not a maximum, so it can't force a discharge down to a target on its own). This should let your calibration automation collapse to just setting a target %/time on the new selector instead of the full export-duration calculation, and the optimiser should make somewhat better use of the discharged energy than a fixed-duration forced export since it's a real cost signal in the normal plan search rather than a blunt block.

closed 2026-09-12 by springfall2008