springfall2008/batpred #1586: Nordpool prices

Doorshifter, 2024-11-03:
First of all I want to thank you for sharing this really nice piece of work. I use it with [https://github.com/hif2k1/battery_sim](url) to size my future battery.

Nordpool prices are either today's or today's + tomorrow's prices if tomorrow_valid: true in the sensor's attributes.
I use these lines to fill the pred_bat sensors in the apps.yaml (we get same rates for in - and export till 2027 in NL):
  metric_octopus_import: 'sensor.nordpool_kwh_nl_eur_3_10_021'
  metric_octopus_export: 'sensor.nordpool_kwh_nl_eur_3_10_021'

I would expect there is a maximum of two days in these sensors:
  predbat.rates
  predbat.rates_export

however I can see yesterday, today, tomorrow, the day after tomorrow and more.
[image]
and similar to this, my apex chart.
[image]
Not sure if this influences the actual prediction.

**Predbat version**

8.5.6

**Environment details**

- I use the SE inverter, which I actually have but with the BMS part disabled. I could not get pred_bat error free when defining 'my own inverter' as described in the doc
- I use the Add-on
- I noticed that there are intermediate timestamps as well, even though Nordpool doesn't have that.
min: 25.4
max: 48.3
average: 29.23
threshold: 46.9
results: 
2024-11-02T00:00:00+0100: 27.9
2024-11-02T00:30:00+0100: 27.9
2024-11-02T01:00:00+0100: 27.3
2024-11-02T01:30:00+0100: 27.3
2024-11-02T02:00:00+0100: 26.4
2024-11-02T02:30:00+0100: 26.4
2024-11-02T03:00:00+0100: 25.9
2024-11-02T03:30:00+0100: 25.9
2024-11-02T04:00:00+0100: 26.1
2024-11-02T04:30:00+0100: 26.1
2024-11-02T05:00:00+0100: 26.2
2024-11-02T05:30:00+0100: 26.2
2024-11-02T06:00:00+0100: 27.1
2024-11-02T06:30:00+0100: 27.1
2024-11-02T07:00:00+0100: 27.4
2024-11-02T07:30:00+0100: 27.4
2024-11-02T08:00:00+0100: 26.2
2024-11-02T08:30:00+0100: 26.2
2024-11-02T09:00:00+0100: 26.9
2024-11-02T09:30:00+0100: 26.9
2024-11-02T10:00:00+0100: 23.9
2024-11-02T10:30:00+0100: 23.9
2024-11-02T11:00:00+0100: 22.8
2024-11-02T11:30:00+0100: 22.8
2024-11-02T12:00:00+0100: 22
2024-11-02T12:30:00+0100: 22
2024-11-02T13:00:00+0100: 22.7
2024-11-02T13:30:00+0100: 22.7
2024-11-02T14:00:00+0100: 25.4
2024-11-02T14:30:00+0100: 25.4
2024-11-02T15:00:00+0100: 27
2024-11-02T15:30:00+0100: 27
2024-11-02T16:00:00+0100: 29.1
2024-11-02T16:30:00+0100: 29.1
2024-11-02T17:00:00+0100: 30.7
2024-11-02T17:30:00+0100: 30.7
2024-11-02T18:00:00+0100: 30.8
2024-11-02T18:30:00+0100: 30.8
2024-11-02T19:00:00+0100: 30.2
2024-11-02T19:30:00+0100: 30.2
2024-11-02T20:00:00+0100: 29.1
2024-11-02T20:30:00+0100: 29.1
2024-11-02T21:00:00+0100: 28
2024-11-02T21:30:00+0100: 28
2024-11-02T22:00:00+0100: 27.8
2024-11-02T22:30:00+0100: 27.8
2024-11-02T23:00:00+0100: 27.1
2024-11-02T23:30:00+0100: 27.1
2024-11-03T00:00:00+0100: 27.9
2024-11-03T00:30:00+0100: 27.9
2024-11-03T01:00:00+0100: 27.3
2024-11-03T01:30:00+0100: 27.3
2024-11-03T02:00:00+0100: 26.4
2024-11-03T02:30:00+0100: 26.4
2024-11-03T03:00:00+0100: 25.9
2024-11-03T03:30:00+0100: 25.9
2024-11-03T04:00:00+0100: 26.1
2024-11-03T04:30:00+0100: 26.1
2024-11-03T05:00:00+0100: 26.2
2024-11-03T05:30:00+0100: 26.2
2024-11-03T06:00:00+0100: 27.1
2024-11-03T06:30:00+0100: 27.1
2024-11-03T07:00:00+0100: 27.4
2024-11-03T07:30:00+0100: 27.4
2024-11-03T08:00:00+0100: 26.2
2024-11-03T08:30:00+0100: 26.2
2024-11-03T09:00:00+0100: 26.9
2024-11-03T09:30:00+0100: 26.9
2024-11-03T10:00:00+0100: 23.9
2024-11-03T10:30:00+0100: 23.9
2024-11-03T11:00:00+0100: 22.8
2024-11-03T11:30:00+0100: 22.8
2024-11-03T12:00:00+0100: 22
2024-11-03T12:30:00+0100: 22
2024-11-03T13:00:00+0100: 22.7
2024-11-03T13:30:00+0100: 22.7
2024-11-03T14:00:00+0100: 25.4
2024-11-03T14:30:00+0100: 25.4
2024-11-03T15:00:00+0100: 27
2024-11-03T15:30:00+0100: 27
2024-11-03T16:00:00+0100: 29.1
2024-11-03T16:30:00+0100: 29.1
2024-11-03T17:00:00+0100: 30.7
2024-11-03T17:30:00+0100: 30.7
2024-11-03T18:00:00+0100: 30.8
2024-11-03T18:30:00+0100: 30.8
2024-11-03T19:00:00+0100: 30.2
2024-11-03T19:30:00+0100: 30.2
2024-11-03T20:00:00+0100: 29.1
2024-11-03T20:30:00+0100: 29.1
2024-11-03T21:00:00+0100: 28
2024-11-03T21:30:00+0100: 28
2024-11-03T22:00:00+0100: 27.8
2024-11-03T22:30:00+0100: 27.8
2024-11-03T23:00:00+0100: 27.1
2024-11-03T23:30:00+0100: 27.1
2024-11-04T00:00:00+0100: 26.8
2024-11-04T00:30:00+0100: 26.8
2024-11-04T01:00:00+0100: 26.5
2024-11-04T02:30:00+0100: 26.5
2024-11-04T03:00:00+0100: 26.4
2024-11-04T03:30:00+0100: 26.4
2024-11-04T04:00:00+0100: 26.6
2024-11-04T04:30:00+0100: 26.6
2024-11-04T05:00:00+0100: 27.4
2024-11-04T05:30:00+0100: 27.4
2024-11-04T06:00:00+0100: 30.4
2024-11-04T06:30:00+0100: 30.4
2024-11-04T07:00:00+0100: 32.6
2024-11-04T07:30:00+0100: 32.6
2024-11-04T08:00:00+0100: 32.2
2024-11-04T08:30:00+0100: 32.2
2024-11-04T09:00:00+0100: 30.2
2024-11-04T09:30:00+0100: 30.2
2024-11-04T10:00:00+0100: 27.7
2024-11-04T10:30:00+0100: 27.7
2024-11-04T11:00:00+0100: 25.8
2024-11-04T11:30:00+0100: 25.8
2024-11-04T12:00:00+0100: 25.7
2024-11-04T12:30:00+0100: 25.7
2024-11-04T13:00:00+0100: 25.8
2024-11-04T13:30:00+0100: 25.8
2024-11-04T14:00:00+0100: 27.1
2024-11-04T14:30:00+0100: 27.1
2024-11-04T15:00:00+0100: 30.8
2024-11-04T15:30:00+0100: 30.8
2024-11-04T16:00:00+0100: 36.7
2024-11-04T16:30:00+0100: 36.7
2024-11-04T17:00:00+0100: 48.3
2024-11-04T17:30:00+0100: 48.3
2024-11-04T18:00:00+0100: 43.7
2024-11-04T18:30:00+0100: 43.7
2024-11-04T19:00:00+0100: 35.2
2024-11-04T19:30:00+0100: 35.2
2024-11-04T20:00:00+0100: 30.5
2024-11-04T20:30:00+0100: 30.5
2024-11-04T21:00:00+0100: 29
2024-11-04T21:30:00+0100: 29
2024-11-04T22:00:00+0100: 28.4
2024-11-04T22:30:00+0100: 28.4
2024-11-04T23:00:00+0100: 26.9
2024-11-04T23:30:00+0100: 26.9
2024-11-05T00:00:00+0100: 26.8
2024-11-05T00:30:00+0100: 26.8
2024-11-05T01:00:00+0100: 26.5
2024-11-05T02:30:00+0100: 26.5
2024-11-05T03:00:00+0100: 26.4
2024-11-05T03:30:00+0100: 26.4
2024-11-05T04:00:00+0100: 26.6
2024-11-05T04:30:00+0100: 26.6
2024-11-05T05:00:00+0100: 27.4
2024-11-05T05:30:00+0100: 27.4
2024-11-05T06:00:00+0100: 30.4
2024-11-05T06:30:00+0100: 30.4
2024-11-05T07:00:00+0100: 32.6
2024-11-05T07:30:00+0100: 32.6
2024-11-05T08:00:00+0100: 32.2
2024-11-05T08:30:00+0100: 32.2
2024-11-05T09:00:00+0100: 30.2
2024-11-05T09:30:00+0100: 30.2
2024-11-05T10:00:00+0100: 27.7
2024-11-05T10:30:00+0100: 27.7
2024-11-05T11:00:00+0100: 25.8
2024-11-05T11:30:00+0100: 25.8
2024-11-05T12:00:00+0100: 25.7
2024-11-05T12:30:00+0100: 25.7
2024-11-05T13:00:00+0100: 25.8
2024-11-05T13:30:00+0100: 25.8
2024-11-05T14:00:00+0100: 27.1
2024-11-05T14:30:00+0100: 27.1
2024-11-05T15:00:00+0100: 30.8
2024-11-05T15:30:00+0100: 30.8
2024-11-05T16:00:00+0100: 36.7
2024-11-05T16:30:00+0100: 36.7
2024-11-05T17:00:00+0100: 48.3
2024-11-05T17:30:00+0100: 48.3
2024-11-05T18:00:00+0100: 43.7
2024-11-05T18:30:00+0100: 43.7
2024-11-05T19:00:00+0100: 35.2
2024-11-05T19:30:00+0100: 35.2
2024-11-05T20:00:00+0100: 30.5
2024-11-05T20:30:00+0100: 30.5
2024-11-05T21:00:00+0100: 29
2024-11-05T21:30:00+0100: 29
2024-11-05T22:00:00+0100: 28.4
2024-11-05T22:30:00+0100: 28.4
2024-11-05T23:00:00+0100: 26.9
2024-11-05T23:30:00+0100: 26.9
2024-11-06T00:00:00+0100: 26.8
2024-11-06T00:30:00+0100: 26.8
2024-11-06T01:00:00+0100: 26.5
2024-11-06T02:30:00+0100: 26.5
2024-11-06T03:00:00+0100: 26.4
2024-11-06T03:30:00+0100: 26.4
2024-11-06T04:00:00+0100: 26.6
2024-11-06T04:30:00+0100: 26.6
2024-11-06T05:00:00+0100: 27.4
2024-11-06T05:30:00+0100: 27.4
2024-11-06T06:00:00+0100: 30.4
2024-11-06T06:30:00+0100: 30.4
2024-11-06T07:00:00+0100: 32.6
2024-11-06T07:30:00+0100: 32.6
2024-11-06T08:00:00+0100: 32.2
2024-11-06T08:30:00+0100: 32.2
2024-11-06T09:00:00+0100: 30.2
2024-11-06T09:30:00+0100: 30.2
2024-11-06T10:00:00+0100: 27.7
2024-11-06T10:30:00+0100: 27.7
2024-11-06T11:00:00+0100: 25.8
2024-11-06T11:30:00+0100: 25.8
2024-11-06T12:00:00+0100: 25.7
2024-11-06T12:30:00+0100: 25.7
2024-11-06T13:00:00+0100: 25.8
2024-11-06T13:30:00+0100: 25.8

friendly_name: Export rates
state_class: measurement
unit_of_measurement: c
icon: mdi:currency-usd

Doorshifter, 2024-11-03:
I am not sure, but this might be related.
[image]

springfall2008, 2024-11-04:
The energy rate data is processed by Predbat first, any missing rate information is made up by duplicating the previous days data in order to make the planning more accurate.

Energy rates are in 30 minute segments for planning so again Predbat fills in the gap.

The average p/kWh spike was a bug as it counted your standing charge, I've made a new release and fix that already.

Doorshifter, 2024-11-06:
Indeed the spikes seem to be caused by something different.

I do get the duplications now. Duplications are much better than the at the time of the prediction still not available rates.
I also noticed the '?'s in the plan now, next to those duplicated prices.

closed 2024-11-06 by Doorshifter