springfall2008/batpred #1588: Not all sensors honor the apps.yaml setting for valuta

Doorshifter, 2024-11-03:
Not all valuta related sensors honor the apps.yaml setting:

I use the settings below
  # Currency, symbol for main currency second symbol for 1/100s e.g. $ c or £ p or e c
  currency_symbols:
    - '€'
    - 'c'
however when checking them in the dashboard I see that some are hardcoded.
[image]
[image]
I would expect that this would be in line with the apps.yaml setting.
Also some of the amounts are getting quite large, it might be better to show them in $, €, £, ... iso c, p, ...
**Predbat version**

8.5.6

**Environment details**

- SE Inverter and battery setup: [https://github.com/hif2k1/battery_sim](url)
- I use the Add-on

gcoan, 2024-11-03:
The currency symbol in apps.yaml is currently only used on the html plan, the sensors don't use the values.

Its on my to-do list to address this as its inconsistent

Doorshifter, 2024-11-06:
Looks good now, doesn't it 😄

closed 2024-11-06 by Doorshifter