springfall2008/batpred #1596: Unexpected p/kWh spikes

Doorshifter, 2024-11-04:
Unexplainable spikes in rates graph. 
[image]

**Predbat version**

8.5.9

**Environment details**

- SEbInverter and battery bat_sim
- Standard HAOS installer 
- Anything else? Nope

springfall2008, 2024-11-10:
The reason is this broken is the calculation does quite make sense

What it was doing is (total cost) / (import energy - export energy)

But if you have close to 0 energy used (e.g. a battery) you get close to infinity.

I've changed it to use house Load now, this is on 'main' for testing

closed 2024-11-10 by springfall2008

Doorshifter, 2024-11-11:
I’ve got the latest HA and pred_bat installed. But unfortunately I got this today. 
[image]

Doorshifter, 2024-11-11:
Also , the p/kWh feels like hard coded p’s iso my c’s in the apps.yaml see other solved issue