springfall2008/batpred #1597: Service calls - Bug since avoiding duplicate calls

nickgee31, 2024-11-06:
**Describe the bug**
Since the update to avoid duplicate service calls, I have discovered a bug which is leaving the inverter in idle mode when switching between discharge and charge. For example, at 5AM this morning when switching from discharge to charge it calls the charge service first and then a few seconds later the stop charge service:

05:00:09.899734: Inverter 0 Calling service charge_start_service domain charge
05:00:22.799816: Inverter 0 Calling service charge_stop_service domain discharge

This has not been noticed as an issue before, however since v8.5.4 where duplicate service calls are no longer sent to the inverter it leaves the next half hour in effect idle state and not doing what was intended as it is not sending the start charge again. I think to avoid this in future the stop command should be sent before the start.

**Expected behaviour**
I think it would be better for the stop to be sent before the start when it is switching between modes.

**Predbat version**

V8.5.9

**Environment details**

- FoxESS Inverter

**Log file**
[apps.log](https://github.com/user-attachments/files/17647764/apps.log)
[predbat.log](https://github.com/user-attachments/files/17647765/predbat.log)

springfall2008, 2024-11-10:
Interesting, so one is stop discharge and the other is start charge...

nickgee31, 2024-11-10:
I think as this inverter has only 3 modes. Charge, discharge and self use. So in effect has no stop charge or stop discharge. Self use stops either

springfall2008, 2024-11-10:
Right got it, I'm getting a fix ready now

springfall2008, 2024-11-10:
Fix on 'main' for testing

nickgee31, 2024-11-10:
Just tried a force discharge and it selected discharge and then one second later back to self use. Out at the minute so can’t get the full log

-11-10 13:10:50.697155: Inverter 0 set force discharge to True
2024-11-10 13:10:50.697325: Inverter 0 Calling service charge_stop_service domain charge with data {'device_id': '', 'target_soc': 69, 'power': 4000}
2024-11-10 13:10:50.839105: Inverter 0 Calling service discharge_start_service domain discharge with data {'device_id': '', 'target_soc': 69, 'power': 4000}
2024-11-10 13:10:51.306690: Inverter 0 Calling service charge_stop_service domain charge with data {'device_id': ''}

springfall2008, 2024-11-10:
Another fix on 'main' for testing, I forgot something.

nickgee31, 2024-11-10:
Looks like that has all worked well now. Did a force discharge, force charge and then back to idle and all good. Thank you!

closed 2024-11-10 by springfall2008