springfall2008/batpred #1599: Solarman integration. Sofar wifi dongle lsw3

kukumagi, 2024-11-07:
**Is your feature request related to a problem? Please describe.**
I found a integration for Solarman lsw3 wifi dongle that works with Sofar inverter and I am looking for ways to make it work with predbat.
https://github.com/davidrapan/ha-solarman

**Describe the solution you'd like**
Solution would be to use Solarman integration with predbat for Sofar inverter and other supported inverters.

**Describe alternatives you've considered**
Alternative is to use Sofar2Mqtt witch requires additional hardware.

**Additional context**
Same Feature request in Solarman repository 
https://github.com/davidrapan/ha-solarman/issues/256

My current apps.yaml configuration with Solarman integration
`# ------------------------------------------------------------------
# This is an example configuration, please modify it
# ------------------------------------------------------------------
---
pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # Timezone to work in
  timezone: Europe/Tallinn

  # Currency, symbol for main currency second symbol for 1/100s e.g. $ c or £ p or e c
  currency_symbols:
   - '€'
   - 's'

  # Number of threads to use in plan calculation
  # Can be auto for automatic, 0 for off or values 1-N for a fixed number
  threads: auto

  # Sets the maximum period of zero load before the gap is filled, default 30 minutes
  # To disable set it to 1440
  load_filter_threshold: 30

  #
  # Sensors, more than one can be specified and they will be summed up automatically
  #
  # For two inverters the load today would normally be the master load sensor only (to cover the entire house)
  # If you have three phase and one inverter per phase then you would need three load sensors
  #
  # For pv_today if you have multiple solar inverter inputs then you should include one entry for each inverter
  #
  load_today:
    - sensor.inverter_today_load_consumption
  import_today:
    - sensor.inverter_today_energy_import
  export_today:
    - sensor.inverter_today_energy_export
  pv_today:
    - sensor.inverter_today_production

  #
  # Controls/status - must by 1 per inverter
  #
  num_inverters: 1
  inverter_type: "SF"
  #
  # Run balance inverters every N seconds (0=disabled) - only for multi-inverter
  balance_inverters_seconds: 60
  #

  battery_power:
[... 296 lines cut ...]
  # Gas rates for comparison
  #metric_octopus_gas: 're:(sensor.(octopus_energy_|)gas_[0-9a-z]+_[0-9a-z]+_current_rate)'

  # Nordpool market energy rates
  #futurerate_url: 'https://dataportal-api.nordpoolgroup.com/api/DayAheadPrices?date=DATE&market=N2EX_DayAhead&deliveryArea=UK&currency=GBP'
  #futurerate_adjust_import: True
  #futurerate_adjust_export: False
  #futurerate_peak_start: "16:00:00"
  #futurerate_peak_end: "19:00:00"
  #futurerate_peak_premium_import: 14
  #futurerate_peak_premium_export: 6.5

  # Watch list, a list of sensors to watch for changes and then update the plan if they change
  # This is useful for things like the Octopus Intelligent Slot sensor so that the plan update as soon as you plugin in
  # Only uncomment the items you actually have set up above in apps.yaml, of course you can add your own as well
  # Note those using +[] are lists that are appended to this list, whereas {} items are single items only
  #watch_list:
  #  - '{octopus_intelligent_slot}'
  #  - '{octopus_ready_time}'
  #  - '{octopus_charge_limit}'
  #  - '{octopus_saving_session}'
  #  - '+[car_charging_planned]'
  #  - '+[car_charging_soc]'
  #  - '{car_charging_now}'
`

springfall2008, 2024-11-10:
I'm not sure I know what the question is here?

kukumagi, 2024-11-10:
> I'm not sure I know what the question is here?

Can I use predbat battery control with this: https://github.com/davidrapan/ha-solarman
Instead of using Sofar2mqtt?

gcoan, 2024-11-11:
> Can I use predbat battery control with this: https://github.com/davidrapan/ha-solarman
> Instead of using Sofar2mqtt?

You probably can but it will require some configuration in apps.yaml

If you use the existing Sofar configuration with inverter_type of SF then Predbat knows how to talk to that inverter using the sofar2mqtt integration.

Presumably the ha-solarman integration uses different control names to start and stop discharge, read the SoC values, etc.

You will need to define a custom inverter type and tell Predbat what the controls are that it needs to use.
See https://springfall2008.github.io/batpred/inverter-setup/#i-want-to-add-an-unsupported-inverter-to-predbat

kukumagi, 2025-05-01:
Closing issue.

closed 2025-05-01 by kukumagi