springfall2008/batpred #1601: IOG Hold for Car

matttheotter, 2024-11-08:
**Describe the bug**
This may very well be a config error on my part.

Configured Predbat for IOG with my Hypervolt Home 3 Pro.

I have set "car_charging_now" in apps.yaml to see when the car is charging as IOG doesn't report it, this is set to a binary sensor for ease.

Predbat seems to incorrectly be seeing the entire 8h period for car charging regardless of Car SoC, IOG Target and Charging Status.

Predbat is reporting "hold for car" and hasn't run any discharge cycles after the car completed its charge.

**Expected behaviour**
Predbat should be able to see what the cars current SoC is and what IOGs target to create an update to the plan that shows the charging sessions that will actually be required to achieve the target charge.

**Predbat version**

8.5.9

**Environment details**

- GivEnergy HY5.0 Gen3, 9.5 Batt Gen 2, Hypervolt Home 3 Pro
- Proxmox

[image]

[image]

matttheotter, 2024-11-08:
Right realised, one of my mistakes, I was assuming IOG knew I wanted to get to 80% however its actually reporting how much charge I need to add to the currect level so last night I asked for 80% of the cars battery as charge slots.

Swapped Charge Limit back to my HA Sensor so I can tell Predbat what the SoC limit will be, looks like this is going to be my error.

Going to do another test later to see what happens.

gcoan, 2024-11-08:
I think you are misunderstanding how IOG led charging works vs Predbat led charging

With IOG charging, Octopus is in control of the charging and will charge at any period it wants to to meet the ready time and SoC target.  Predbat has to just follow what IOG does. If you don't have a sensor that indicates when the car is actually charging then Predbat can't make any assumptions about the charging and will have to hold the battery SoC as otherwise it risks discharging the battery into the car.

With Predbat led charging, Predbat is entirely in control of the charging and so can plan to charge the car and the battery at the best rates.

https://springfall2008.github.io/batpred/car-charging/

matttheotter, 2024-11-08:
@gcoan I did configure **car_charging_now** and added the response as specified as a workaround as IOG doesn't read if the Hypervolt is charging the car, so Predbat should have known it wasn't charging if that is what the guide is saying?

   ```
# In some cases car planning is difficult (e.g. Ohme with Intelligent doesn't report slots)
    # The car charging now can be set to a sensor to indicate the car is charging and to plan
    # for it to charge during this 30 minute slot
    #car_charging_now:
    - switch.hypervolt_charging

  # Positive responses for car_charging_now
  car_charging_now_response:
    - "on"
```

matttheotter, 2024-11-08:
LOL now I have copied out my code I can see the error, I have added the sensor but not enabled the option.

closed 2024-11-08 by matttheotter

gcoan, 2024-11-08:
> LOL now I have copied out my code I can see the error, I have added the sensor but not enabled the option.

Yeah I see it now.

It's very easy to accidentally introduce YAML errors.  The Predbat web console makes it easier to spot them, but still all too easy to do

matttheotter, 2024-11-08:
> > LOL now I have copied out my code I can see the error, I have added the sensor but not enabled the option.
> 
> Yeah I see it now.
> 
> It's very easy to accidentally introduce YAML errors.  The Predbat web console makes it easier to spot them, but still all too easy to do

Honestly, the web console didn't make it super clear as Predbat thought the sensor was for the planned charge response.

I only twigged when I couldn't get car_charging_now added to the watch.

Now on to Octopus to figure out why they are doing weird schedules with Hypervolt prior to the charging slots.