springfall2008/batpred #1604: Predbat crash in Docker container - second time

PianSom, 2024-11-09:
**Describe the bug**
Random Predbat crash. Similar crash occurred 5 Oct
Works perfectly most of the time

**Expected behaviour**
Stability!

**Predbat version**
8.5.9

**Environment details**
 - AIO
 - Docker setup for both HA and Predbat

**Screenshots**
Non suitable

**Log file**

23599 | 2024-11-09 06:41:58.284926: Warn: Failed to decode response from http://192.168.99.200:8123/addons/self/info
-- | --
23590 | 2024-11-08 20:35:13.816292: Warn: Web Socket closed, will try to reconnect in 5 seconds
23588 | aiohttp.client_exceptions.S erverDisconnectedError: Server disconnected
23572 | 2024-11-08 20:35:13.816237: Error: Traceback (most recent call last):
23571 | 2024-11-08 20:35:13.815496: Error: Web Socket exception in startup: Server disconnected
23569 | 2024-11-08 20:35:08.809361: Warn: Web Socket closed, will try to reconnect in 5 seconds
23567 | aiohttp.client_exceptions.C lientConnectorError: Cannot connect to host 192.168.99.200:8123 ssl:default [Connect call failed ('192.168.99.200', 8123)]
23566 | raise client_error(req. connection_key, exc) from exc
23538 | ConnectionRefusedError: [Er rno 111] Connect call failed ('192.168.99.200', 8123)
23537 | raise OSError(err, f'Co nnect call failed {address}')
23522 | 2024-11-08 20:35:08.809263: Error: Traceback (most recent call last):
23521 | 2024-11-08 20:35:08.799166: Error: Web Socket exception in startup: Cannot connect to host 192.168.99.200:8123 ssl:default [Connect call failed ('192.168.99.200', 8123)]
23519 | 2024-11-08 20:35:03.796860: Warn: Web Socket closed, will try to reconnect in 5 seconds
23517 | aiohttp.client_exceptions.S erverDisconnectedError: Server disconnected
23501 | 2024-11-08 20:35:03.796643: Error: Traceback (most recent call last):
23500 | 2024-11-08 20:35:03.793471: Error: Web Socket exception in startup: Server disconnected
23498 | 2024-11-08 20:35:00.703990: Error: ('Connection aborted.', ConnectionResetError(104, 'Connection reset by peer'))
23497 | 2024-11-08 20:34:58.782686: Warn: Web Socket closed, will try to reconnect in 5 seconds
23112 | 2024-11-08 20:23:39.582741: Error:
23111 | 2024-11-08 20:23:39.582692: Info: record_status Error: Exception raised
23109 | ValueError
23108 | raise ValueError
23102 | 2024-11-08 20:23:39.577256: Error: Traceback (most recent call last):
23101 | 2024-11-08 20:23:39.567137: Error: Exception raised
23100 | 2024-11-08 20:23:39.565706: Info: record_status Error: metric_octopus_import not set correctly or no energy rates can be read
23099 | 2024-11-08 20:23:39.560234: Error: metric_octopus_import is not set correctly or no energy rates can be read
23098 | 2024-11-08 20:23:39.560217: Warn: No Octopus data in sensor event.octopus_energy_electricity_22m0155214_1023519225465_current_day_rates attribute 'all_rates' / 'rates' / 'raw_today'
23097 | 2024-11-08 20:23:39.560194: Warn: No Octopus data in sensor sensor.octopus_energy_electricity_22m0155214_1023519225465_previous_rate attribute 'all_rates'
23083 | 2024-11-08 20:23:21.940530: Warn: Web Socket closed, will try to reconnect in 5 seconds
23081 | aiohttp.client_exceptions.S erverDisconnectedError: Server disconnected
23065 | 2024-11-08 20:23:21.940446: Error: Traceback (most recent call last):
23064 | 2024-11-08 20:23:21.939446: Error: Web Socket exception in startup: Server disconnected
23062 | 2024-11-08 20:23:16.932886: Warn: Web Socket closed, will try to reconnect in 5 seconds
23060 | aiohttp.client_exceptions.S erverDisconnectedError: Server disconnected
23044 | 2024-11-08 20:23:16.932816: Error: Traceback (most recent call last):
23043 | 2024-11-08 20:23:16.931738: Error: Web Socket exception in startup: Server disconnected
23041 | 2024-11-08 20:23:11.927361: Warn: Web Socket closed, will try to reconnect in 5 seconds
23039 | aiohttp.client_exceptions.S erverDisconnectedError: Server disconnected
23023 | 2024-11-08 20:23:11.927129: Error: Traceback (most recent call last):
23022 | 2024-11-08 20:23:11.919391: Error: Web Socket exception in startup: Server disconnected
23020 | 2024-11-08 20:23:06.872172: Warn: Web Socket closed, will try to reconnect in 5 seconds

springfall2008, 2024-11-10:
Its saying it can not connect to host, maybe a network issue or the HA server is down or changed IP?

PianSom, 2024-11-10:
Network logs are clear, HA (and host) showed no errors, IP is fixed and has been for 12m+.

Similar crash report here - https://community.givenergy.cloud/d/3696-first-night-live-on-predbat/2430

mervin39, 2024-11-11:
Same happens to me if Home Assistant is down for a while (say during an upgrade). A manual restart of the predbat docker container fixes it. Logs below - I updated HA just before 09:00, and noticed it was down about 11:05 and did a `docker restart predbat`

`<html><body>

24901 | 2024-11-11 11:15:02.311093:  Warn: Historical day 14 has 10 minutes of gap in the data, filled from 14.16 kWh to make new average 14.26 kWh (percent 99%)
-- | --
24900 | 2024-11-11 11:15:02.309530:  Warn: Historical day 7 has 5 minutes of gap in the data, filled from 15.13 kWh to make new average 15.18 kWh (percent 100%)
24771 | 2024-11-11 11:10:01.710208:  Warn: Historical day 14 has 5 minutes of gap in the data, filled from 14.23 kWh to make new average 14.27 kWh (percent 100%)
24770 | 2024-11-11 11:10:01.708745:  Warn: Historical day 7 has 5 minutes of gap in the data, filled from 15.08 kWh to make new average 15.13 kWh (percent 100%)
24449 | 2024-11-11 11:07:35.689717:  Warn: Historical day 7 has 5 minutes of gap in the data, filled from 15.07 kWh to make new average 15.13 kWh (percent 100%)
24404 | 2024-11-11 11:07:26.632240:  Warn: Regular expression argument: carbon_intensity unable to match re:(sensor.carbon_intensity_uk), now will disable
24403 | 2024-11-11 11:07:26.631380:   Warn: Regular argument car_charging_planned expression  re:(sensor.wallbox_portal_status_description\|sensor.myenergi_zappi_[0-9a-z]+_plug_status)  failed to match - disabling this item
24392 | 2024-11-11 11:07:26.587242:  Warn: Failed to decode response  from http://192.168.0.3:8123/addons/self/info
24383 | 2024-11-11 09:00:16.475062:  Warn: Web Socket closed, will try to reconnect in 5 seconds
24381 | aiohttp.client_exceptions.C lientConnectorError: Cannot connect to host 192.168.0.3:8123 ssl:default [Connect call failed ('192.168.0.3', 8123)]
24380 | raise client_error(req. connection_key, exc) from exc
24352 | ConnectionRefusedError: [Er rno 111] Connect call failed ('192.168.0.3', 8123)
24351 | raise OSError(err, f'Co nnect call failed {address}')
24336 | 2024-11-11 09:00:16.474949:  Error: Traceback (most recent call last):
24335 | 2024-11-11 09:00:16.473299:   Error: Web Socket exception in startup: Cannot connect to host  192.168.0.3:8123 ssl:default [Connect call failed ('192.168.0.3', 8123)]
24333 | 2024-11-11 09:00:11.466784:  Warn: Web Socket closed, will try to reconnect in 5 seconds
24331 | aiohttp.client_exceptions.C lientConnectorError: Cannot connect to host 192.168.0.3:8123 ssl:default [Connect call failed ('192.168.0.3', 8123)]
24330 | raise client_error(req. connection_key, exc) from exc
24302 | ConnectionRefusedError: [Er rno 111] Connect call failed ('192.168.0.3', 8123)
24301 | raise OSError(err, f'Co nnect call failed {address}')
24286 | 2024-11-11 09:00:11.466556:  Error: Traceback (most recent call last):
24285 | 2024-11-11 09:00:11.462522:   Error: Web Socket exception in startup: Cannot connect to host  192.168.0.3:8123 ssl:default [Connect call failed ('192.168.0.3', 8123)]
24283 | 2024-11-11 09:00:06.454089:  Warn: Web Socket closed, will try to reconnect in 5 seconds
24281 | aiohttp.client_exceptions.C lientConnectorError: Cannot connect to host 192.168.0.3:8123 ssl:default [Connect call failed ('192.168.0.3', 8123)]
24280 | raise client_error(req. connection_key, exc) from exc
24252 | ConnectionRefusedError: [Er rno 111] Connect call failed ('192.168.0.3', 8123)
24251 | raise OSError(err, f'Co nnect call failed {address}')
24236 | 2024-11-11 09:00:06.453967:  Error: Traceback (most recent call last):
24235 | 2024-11-11 09:00:06.452322:   Error: Web Socket exception in startup: Cannot connect to host  192.168.0.3:8123 ssl:default [Connect call failed ('192.168.0.3', 8123)]
24233 | 2024-11-11 09:00:01.445289:  Warn: Web Socket closed, will try to reconnect in 5 seconds
24231 | aiohttp.client_exceptions.C lientConnectorError: Cannot connect to host 192.168.0.3:8123 ssl:default [Connect call failed ('192.168.0.3', 8123)]
24230 | raise client_error(req. connection_key, exc) from exc
24202 | ConnectionRefusedError: [Er rno 111] Connect call failed ('192.168.0.3', 8123)
24201 | raise OSError(err, f'Co nnect call failed {address}')
24186 | 2024-11-11 09:00:01.445058:  Error: Traceback (most recent call last):
24185 | 2024-11-11 09:00:01.440976:   Error: Web Socket exception in startup: Cannot connect to host  192.168.0.3:8123 ssl:default [Connect call failed ('192.168.0.3', 8123)]
24183 | 2024-11-11 09:00:00.682456:   Error: HTTPConnectionPool(host='192.168.0.3', port=8123): Max retries  exceeded with url: /api/states (Caused by NewConnectionError(': Failed to establish a new connection: [Errno 111] Connection refused'))
24182 | 2024-11-11 08:59:56.432689:  Warn: Web Socket closed, will try to reconnect in 5 seconds
24180 | aiohttp.client_exceptions.C lientConnectorError: Cannot connect to host 192.168.0.3:8123 ssl:default [Connect call failed ('192.168.0.3', 8123)]
24179 | raise client_error(req. connection_key, exc) from exc
24151 | ConnectionRefusedError: [Er rno 111] Connect call failed ('192.168.0.3', 8123)
24150 | raise OSError(err, f'Co nnect call failed {address}')
24135 | 2024-11-11 08:59:56.432579:  Error: Traceback (most recent call last):
24134 | 2024-11-11 08:59:56.397144:   Error: Web Socket exception in startup: Cannot connect to host  192.168.0.3:8123 ssl:default [Connect call failed ('192.168.0.3', 8123)]
24132 | 2024-11-11 08:59:51.375937:  Warn: Web Socket closed, will try to reconnect in 5 seconds
24041 | 2024-11-11 08:55:02.082005:  Warn: Historical day 7 has 5 minutes of gap in the data, filled from 15.02 kWh to make new average 15.07 kWh (percent 100%)
23724 | 2024-11-11 08:50:02.013651:  Warn: Historical day 7 has 5 minutes of gap in the data, filled from 15.01 kWh to make new average 15.07 kWh (percent 100%)

</body>
</html>`

closed 2025-01-01 by springfall2008