springfall2008/batpred #1607: Currently predbat does not work via GivTCP on GivEnergy 3 phase Inverters

LeeFarm11, 2024-11-09:
Brought here from https://github.com/britkat1980/giv_tcp/issues/286

**Is your feature request related to a problem? Please describe.**
GivEnergy 3 phase Inverters appear to be very different in controls to previous Inverters. The controls which predbat uses to control the Inverters are not available on the 3 phase. There are some controls which predbat uses which might be fixable within GivTCP, but there are some controls which predbat relies on which (I think) are simply not available on the Inverter.

**Summary problem -** 
3 phase Inverter does not have any modes. You cannot 'set' Eco or Timed_Export etc.
The GivTCP control of mode does not work, because it does not exist.
If I try setting to any of the other settings, it just reverts back to Eco (paused) next time the inverter is read.
But predbat relies on it.
[image]

There are only 3 main controls on the 3 phase Inverter (other than % and Times)
Force AC Charge Enable
Force Charge Enable
Force Discharge Enable

To charge from the grid, the 2 charge switches are set to On.
To discharge to the grid, the Discharge switch is set to On.

The relevant time slots and %s then become active.

I have not found any useful functionality to only using Force AC Charge Enable or Force Charge Enable. They do have an impact but I have not found any useful functionality. In my world so far, they may as well be a single switch.

One of the controls which predbat relies on in apps.yaml is 
```
  scheduled_charge_enable:
    - switch.givtcp_{geserial}_enable_charge_schedule
```
But switch.givtcp_{geserial}_enable_charge_schedule does not do anything (yet).

With help from @gcoan I have created some helpers/ automations which I think are now working for me.

**First improvement - scheduled_charge_enable**

Setting up a helper and automation for scheduled_charge_enable was the first big step forward -

I created a helper toggle switch. I called it helper_predbat_enable_charge_schedule.

I then created an automation -
```
alias: predbat_force_charge_automation
description: ""
trigger:
  - platform: state
    entity_id:
      - input_boolean.helper_predbat_enable_charge_schedule
    to: null
condition: []
action:
  - choose:
      - conditions:
          - condition: state
            entity_id: input_boolean.helper_predbat_enable_charge_schedule
            state: "on"
        sequence:
          - action: switch.turn_on
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_ac_charge_enable
          - action: switch.turn_on
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_charge_enable
      - conditions:
          - condition: state
            entity_id: input_boolean.helper_predbat_enable_charge_schedule
            state: "off"
        sequence:
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_ac_charge_enable
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_charge_enable
mode: single
```
I have commented out REST in apps.yaml -
```
  # When set use the REST API rather than HA entity for control, should be more reliable/faster to control
  # Set one per inverter
  # If using Docker then change homeassistant.local to the Docker IP address
  #givtcp_rest:
  #  - 'http://192.168.1.204:6345'
```
and I set this is apps.yaml
```
  scheduled_charge_enable:
  #  - switch.givtcp_{geserial}_enable_charge_schedule
    - input_boolean.helper_predbat_enable_charge_schedule
```

This works well, except that predbat turns off scheduled_charge_enable during Hold Charging or Freeze Charging.

This results in strange behaviour. I have cheap rate 13:00 - 16:00. This happens -
[image]

**Further testing with inverter_mode**

Because predbat is setting inverter mode via this -
```
  inverter_mode:
    - select.givtcp_{geserial}_mode
```
I did set up another helper and automation to set the 3 Inverter controls via the predbat inverter_mode. Unfortunately the predbat inverter mode stays in Eco unless Discharging, so it does not help too much.

**Further development using predbat.status**

I have now added another automation which takes the predbat.status and updates the 3 switches depending on predbat.status. predbat populates the Timeslots and percentages in advance of the Charge or Discharge periods, so potentially, this might delay the start of those periods by a couple of minutes, but they will end on time because the end slot times are populated.
```
alias: PredbatStatusChangeAutomation
description: Modify Inverter Settings depending on predbat.status
trigger:
  - platform: state
    entity_id:
      - predbat.status
    to: null
condition: []
action:
  - choose:
      - conditions:
          - condition: or
            conditions:
              - condition: state
                entity_id: predbat.status
                state: Charging
              - condition: state
                entity_id: predbat.status
                state: Hold charging
              - condition: state
                entity_id: predbat.status
                state: Freeze charging
              - condition: state
                entity_id: predbat.status
                state: No charge
        sequence:
          - action: switch.turn_on
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_ac_charge_enable
          - action: switch.turn_on
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_charge_enable
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_discharge_enable
      - conditions:
          - condition: or
            conditions:
              - condition: state
                entity_id: predbat.status
                state: Idle
              - condition: state
                entity_id: predbat.status
                state: Error
              - condition: state
                entity_id: predbat.status
                state: Calibration
        sequence:
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_ac_charge_enable
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_charge_enable
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_discharge_enable
      - conditions:
          - condition: or
            conditions:
              - condition: state
                entity_id: predbat.status
                state: Discharging
              - condition: state
                entity_id: predbat.status
                state: Freeze discharging
        sequence:
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_ac_charge_enable
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_charge_enable
          - action: switch.turn_on
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_discharge_enable
mode: single
```

Only been using this for a couple of hours and looks positive.

**But I think I do now see another issue already.**

In the GivTCP thread I said this -
I note that the Charge Slot SOC target was not changed (it was previously set at and remains at 4%), but the (global) Charge Up To % was changed by predbat/GivTCP (in this case to 98%). I assume that individual slots SOC targets are therefore not used by predbat, and therefore I have now set Slot 1 SOC to 100%. (I have also set Discharge Slot 1 SOC to 4% making the same assumption).

@gcoan replied with this -
On your question r/e charge slots, Predbat only ever uses slot 1 for charging and discharging and sets the global charge-to limit. So you need to make sure that the other slots are all cleared as otherwise the inverter will start obey those other charge/discharge instructions and Predbat won't know about them. 

What I now see is predbat aiming for 90% or 93% or whatever. It sets the global Charge Up To %. But overshoots. I think that global Charge Up to % is not being respected by the Inverter when Force Charge/ Force AC Charge is being used. Will try some more tests when time permits.

Ryuisnod, 2024-11-09:
I'm midway through trying to set this up myself so interesting to see it works in a limited fashion. I hope to get my battery to charge up during "octopus intelligent go" smart sessions and stop my Zappi from using the battery to charge the EV during this time! (expect having the battery on charge mode will just sort this tbh) hopefully, your helpers here will help on that road as I'm a relative novice. I'm following your threads on here, givenergy and givtcp as it seems we are in the same boat

springfall2008, 2024-11-10:
Hmm, it maybe better to add a new AC3 mode to Predbat to control in this way.

Is it just the three switches replacing the Battery mode that is needed, is everything else the same?

LeeFarm11, 2024-11-10:
> Hmm, it maybe better to add a new AC3 mode to Predbat to control in this way.
> 
> Is it just the three switches replacing the Battery mode that is needed, is everything else the same?

Not sure if everything else ids the same yet. @britkat1980 might be able to give you better behind-the-scenes information. I was told by GivEnergy just after my install when I complained about how lacking the available control is that they work very closely with him..... (passing the buck?)

I can send details of all the available settings in the Portal, but as you can see I'm not 100% certain what they all really do. I have posted those details before somewhere in GitHub. I can find them if they would help.

There are too many predbat modes. Haha.

Many "Warn:....." statuses, Read-Only etc. Too many statuses to be explicit. I have modified my automation as below -
If Charging, do charging stuff and turn discharging stuff off, otherwise if Discharging do discharging stuff and turn charging stuff off, otherwise turn all off.

```
alias: PredbatStatusChangeAutomation
description: Modify Inverter Settings depending on predbat.status
trigger:
  - platform: state
    entity_id:
      - predbat.status
    to: null
condition: []
action:
  - if:
      - condition: or
        conditions:
          - condition: state
            entity_id: predbat.status
            state: Charging
          - condition: state
            entity_id: predbat.status
            state: Hold charging
          - condition: state
            entity_id: predbat.status
            state: Freeze charging
          - condition: state
            entity_id: predbat.status
            state: No charge
    then:
      - action: switch.turn_on
        metadata: {}
        data: {}
        target:
          entity_id: switch.givtcp_td2343g049_force_ac_charge_enable
      - action: switch.turn_on
        metadata: {}
        data: {}
        target:
          entity_id: switch.givtcp_td2343g049_force_charge_enable
      - action: switch.turn_off
        metadata: {}
        data: {}
        target:
          entity_id: switch.givtcp_td2343g049_force_discharge_enable
    else:
      - if:
          - condition: or
            conditions:
              - condition: state
                entity_id: predbat.status
                state: Discharging
              - condition: state
                entity_id: predbat.status
                state: Freeze discharging
        then:
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_ac_charge_enable
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_charge_enable
          - action: switch.turn_on
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_discharge_enable
        else:
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_ac_charge_enable
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_charge_enable
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_discharge_enable
mode: single

```

LeeFarm11, 2024-11-13:
My automation is not picking up when predbat.status moves to or from "Warn: Inverter 0 Write to .......".

In theory the automation is triggering any time there is a change in status, but I wonder if these Warn statuses are not really a status and therefore the trigger is missed?

At the highlighted time when Idle changed to Warn, and also when Warn changed to Idle the automation should have triggered, but it seems it did not. 

[image]

gcoan, 2024-11-13:
Yes they are real state changes to predbat.status, that's why you can see them in the sensor history

The warn: statii are usually transient, the next predbat run usually predbat works OK

LeeFarm11, 2024-11-14:
If they are real Statuses, why does the automation (above) not fire when it changes to Warn:... and when it changes from Warn: ...

Anyway, I have now changed the automation to fire each time the attribute 'last updated' changes. This fires every time predbat recalculates even if there is no change of state. Hopefully this will work round the issue.

**The real problem is how to get predbat to work natively with the Inverter settings.**

**Freeze Discharge seems an odd one. Freeze Discharge does not appear to ever set any Discharge Timeslot times, so not sure how it is going to do any discharging. Maybe this is a fundamental difference between AC3 and previous Inverters?**

Here is my newly modified automation in case @Ryuisnod or anyone else is following.

```
alias: PredbatStatusChangeAutomation2
description: Modify Inverter Settings depending on predbat.status
triggers:
  - trigger: state
    entity_id:
      - predbat.status
    attribute: last_updated
conditions: []
actions:
  - if:
      - condition: or
        conditions:
          - condition: state
            entity_id: predbat.status
            state: Charging
          - condition: state
            entity_id: predbat.status
            state: Hold charging
          - condition: state
            entity_id: predbat.status
            state: Freeze charging
          - condition: state
            entity_id: predbat.status
            state: No charge
    then:
      - action: switch.turn_on
        metadata: {}
        data: {}
        target:
          entity_id: switch.givtcp_td2343g049_force_ac_charge_enable
      - action: switch.turn_off
        metadata: {}
        data: {}
        target:
          entity_id: switch.givtcp_td2343g049_force_discharge_enable
      - action: switch.turn_on
        metadata: {}
        data: {}
        target:
          entity_id: switch.givtcp_td2343g049_force_charge_enable
    else:
      - if:
          - condition: or
            conditions:
              - condition: state
                entity_id: predbat.status
                state: Discharging
              - condition: state
                entity_id: predbat.status
                state: Freeze discharging
        then:
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_ac_charge_enable
          - action: switch.turn_on
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_discharge_enable
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_charge_enable
        else:
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_ac_charge_enable
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_charge_enable
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_discharge_enable
mode: single
```

gcoan, 2024-11-14:
> Freeze Discharge seems an odd one. Freeze Discharge does not appear to ever set any Discharge Timeslot times, so not sure how it is going to do any discharging

I think you are misunderstanding the Freeze Discharge status.

Freeze Discharge is NOT a discharge, it is 'freeze the current SoC level and stop it increasing (by solar charging), but allow the battery to discharge if house demand requires it'

So effectively it's a capped version of the default Eco mode - Eco mode but don't let the SoC increase from solar generation.

It therefore doesn't need any start or end times.

LeeFarm11, 2024-11-15:
That script is not good when in Read-Only mode. The status continues to be updated each 5 minutes, so while in Read-Only mode (so that user can control the Inverter manually) the automation continues to turn off all the switches each 5 minutes. Now testing another version.

gcoan, 2024-11-15:
> That script is not good when in Read-Only mode. The status continues to be updated each 5 minutes, so while in Read-Only mode (so that user can control the Inverter manually) the automation continues to turn off all the switches each 5 minutes. Now testing another version.

You just need to add a condition to the automation to check that Predbat read only mode is not set. If it is, the condition will stop the automation

LeeFarm11, 2024-11-15:
> You just need to add a condition to the automation to check that Predbat read only mode is not set. If it is, the condition will stop the automation

Yes. I am now testing a new version which does something like that. When I am happy with it I will post it here.

LeeFarm11, 2024-11-15:
I seem to have a problem with the Load prediction. Not sure if this relates to 3 phase. 
apps.yaml is set to use last 2 days of data (due the issue with corrupt database a few days ago).
[image]

Between 20:30 and 21:00 yesterday and day before we were using around 9kW.
[image]
[image]

But predbat is predicting that today we will run at about 3.5kW during that time.
[image]

Maybe only picking up one phase? Or average of 3 phases? Or?
Here is the log for 16:25 calculation

```
2024-11-15 16:25:00.905381: --------------- PredBat - update at 2024-11-15 16:25:00+00:00 with clock skew 0 minutes, minutes now 985
2024-11-15 16:25:00.906791: Saved current settings to /addon_configs/6adb4f0d_predbat/predbat_config.json
2024-11-15 16:25:00.944121: Using cached GITHub data for https://api.github.com/repos/springfall2008/batpred/releases age 80.0 minutes
2024-11-15 16:25:00.944391: Predbat /config/predbat.py version v8.6.0 currently running, latest version is v8.6.0 latest beta v8.6.0
2024-11-15 16:25:00.944510: Autoupdate: Currently up to date
2024-11-15 16:25:01.027646: Configuration: forecast_hours 48 num_cars 0 debug enable is False calculate_plan_every 10
2024-11-15 16:25:01.028321: Predbat mode is set to Control charge & discharge
2024-11-15 16:25:01.028578: NOTE: Read-only mode is enabled, the inverter controls will not be used!!
2024-11-15 16:25:01.983827: Found 4321 load_today datapoints going back 3 days
2024-11-15 16:25:02.807967: Car charging hold True threshold 6.0
2024-11-15 16:25:02.808080: Current data so far today: load 46.5 kWh import 56.1 kWh export 2.9 kWh pv 10.4 kWh
2024-11-15 16:25:02.809377: Adding rate rates_import: {'start': '00:05:00', 'end': '04:59:59', 'rate': 11.84} => 11-15 00:05:00 to 11-15 04:59:00 @ 11.84 date None increment False
2024-11-15 16:25:02.810373: Adding rate rates_import: {'start': '05:00:00', 'end': '13:04:59', 'rate': 26.55} => 11-15 05:00:00 to 11-15 13:04:00 @ 26.55 date None increment False
2024-11-15 16:25:02.812006: Adding rate rates_import: {'start': '13:05:00', 'end': '15:59:59', 'rate': 11.84} => 11-15 13:05:00 to 11-15 15:59:00 @ 11.84 date None increment False
2024-11-15 16:25:02.812747: Adding rate rates_import: {'start': '16:00:00', 'end': '20:04:59', 'rate': 26.55} => 11-15 16:00:00 to 11-15 20:04:00 @ 26.55 date None increment False
2024-11-15 16:25:02.813784: Adding rate rates_import: {'start': '20:05:00', 'end': '21:59:59', 'rate': 11.84} => 11-15 20:05:00 to 11-15 21:59:00 @ 11.84 date None increment False
2024-11-15 16:25:02.814348: Adding rate rates_import: {'start': '22:00:00', 'end': '00:04:59', 'rate': 26.55} => 11-15 22:00:00 to 11-15 00:04:00 @ 26.55 date None increment False
2024-11-15 16:25:02.940012: Adding rate rates_export: {'rate': 12} => 11-15 00:00:00 to 11-15 00:00:00 @ 12.0 date None increment False
2024-11-15 16:25:02.944394: Standing charge is set to 53.66 p
2024-11-15 16:25:03.006616: Adding rate rates_import_override: {'start': '13:05:00', 'end': '13:59:59', 'rate': 20} => 11-15 13:05:00 to 11-15 13:59:00 @ 20.0 date None increment False
2024-11-15 16:25:05.538676: Rate min forward looking: now 0 at end of forecast 0.0
2024-11-15 16:25:05.539147: Import rates min 0 max 26.55 average 20.8
2024-11-15 16:25:05.608324: Export rates min 12.0 max 12.0 average 12.0
2024-11-15 16:25:05.608549: Rate thresholds (for charge/discharge) are import 26.05p (0.0) export 12.0p (0.0)
2024-11-15 16:25:05.629115: High export rate found rates in range 12.0 to 12.0
2024-11-15 16:25:05.640695: Low Import rate found rates in range 11.84 to 20.0
2024-11-15 16:25:05.640836: Battery level now 15.77 -1hr 9.4 midnight 0.7 battery value change hour 71.18 day 168.4 rate_forward 11.17
2024-11-15 16:25:05.641818: Hour energy 8.9 import 8.9 export 0.0 car 0 load 3.35 cost 106.24 import 106.24 export 0.0 car 0.0 carbon 0.0 kG
2024-11-15 16:25:05.899735: Today's energy import 56.1 kWh export 2.9 kWh total 53.2 kWh cost 783.1 p import 764.24 p export -34.8 p carbon 0.0 kg
2024-11-15 16:25:05.899851: Using Solcast integration from inside HA for solar forecast
2024-11-15 16:25:05.900049: PV Data for pv_forecast_today total 15.76 kWh
2024-11-15 16:25:05.900248: PV Data for pv_forecast_tomorrow total 15.42 kWh
2024-11-15 16:25:05.900374: PV Data for pv_forecast_d3 total 32.56 kWh
2024-11-15 16:25:05.900534: PV Data for pv_forecast_d4 total 6.4 kWh
2024-11-15 16:25:05.923017: PV Forecast for today is 15.76 (9.12 10% 25.59 90%) kWh and left today is 0.0 (0.0 10% 0.0 90%) kWh
2024-11-15 16:25:05.985062: PV Forecast for day tomorrow is 15.42 (6.08 10% 42.42 90%) kWh
2024-11-15 16:25:06.015836: PV Forecast for day d2 is 32.56 (5.45 10% 50.7 90%) kWh
2024-11-15 16:25:06.057894: PV Forecast for day d3 is 6.4 (0.58 10% 45.23 90%) kWh
2024-11-15 16:25:06.219170: Historical data totals for days [1, 2] are [38.46, 42.16] - min 38.46
2024-11-15 16:25:06.220301: Historical days now [1, 2] weight [1, 0.75]
2024-11-15 16:25:06.320210: Today's load divergence 90.07 % in-day adjustment 90.57 % damping 0.95x
2024-11-15 16:25:06.320309: Today's predicted so far 20.34 kWh with 18.65 kWh car/iBoost excluded and 11.3 kWh import ignored and 0 forecast extra.
2024-11-15 16:25:06.320362: Today's actual load so far 17.39 kWh with 19.58 kWh Car/iBoost excluded and 9.43 kWh import ignored.
2024-11-15 16:25:06.480853: Inverter 0: Type GE GivEnergy
2024-11-15 16:25:06.481196: Inverter 0 does not have timed pause support enabled
[... cut ...] 16 20:05:00 - 11-16 21:59:00 @ 11.84, 5: 11-17 00:05:00 - 11-17 04:59:00 @ 11.84, 6: 11-17 13:05:00 - 11-17 13:59:00 @ 20.0, 7: 11-17 13:59:00 - 11-17 15:59:00 @ 11.84]
2024-11-15 16:25:14.150018: Will not recompute the plan, it is 5.0 minutes old and max age is 10 minutes
2024-11-15 16:25:14.292920: Completed run status Read-Only
2024-11-15 16:25:14.332900: Info: record_status Read-Only
2024-11-15 16:25:14.374833: Saved current settings to /addon_configs/6adb4f0d_predbat/predbat_config.json
```

LeeFarm11, 2024-11-15:
I don't have Car or iBoost. But I do have EVC but it is not used. But log says 20kWh of car/iBoost excluded. Looks like maybe something related and not related to 3 phase. But I have no idea why.

```
2024-11-15 16:25:06.320309: Today's predicted so far 20.34 kWh with 18.65 kWh car/iBoost excluded and 11.3 kWh import ignored and 0 forecast extra.
2024-11-15 16:25:06.320362: Today's actual load so far 17.39 kWh with 19.58 kWh Car/iBoost excluded and 9.43 kWh import ignored.
```

I did disable GivEVC within GivTCP settings a couple of days ago. But that should not be relevant?

[image]

LeeFarm11, 2024-11-16:
I think I must have accidentally turned on 'Car charging hold' at some point. I turned it off and the load predictions now look more sensible. But since I have never used the car charger except once back in April, I don't understand how it could have such a large impact.

gcoan, 2024-11-16:
It's generally not advised to set days_previous to 1, see https://springfall2008.github.io/batpred/apps-yaml/#understanding-how-days_previous-works so this will have an effect on your prior load history.  It's better to look further back (as far as you can without your load history).  Do turn switch.predbat_load_filter_modal on to exclude the part-day that you will get in your broken load history.

Have a look at the car charging config https://springfall2008.github.io/batpred/apps-yaml/#car-charging-filtering
Predbat was removing whatever you have car_charging_energy set to, so might be worth checking the history on that sensor and whether it is correct?

LeeFarm11, 2024-11-16:
> Have a look at the car charging config https://springfall2008.github.io/batpred/apps-yaml/#car-charging-filtering Predbat was removing whatever you have car_charging_energy set to, so might be worth checking the history on that sensor and whether it is correct?

Thanks.
car_charging_energy is set as default in apps.yaml. Not commented out -
```
car_charging_energy: 're:(sensor.myenergi_zappi_[0-9a-z]+_charge_added_session|sensor.wallbox_portal_added_energy)'
```

I do not have zappi or Wallbox. I only have a GivEnergy EVC so therefore car_charging_energy presumably finds nothing. 
input_number.predbat_car_charging_threshold is set at 6kW and input_number.predbat_car_charging_rate is set at 7.4kW
Not sure which of these takes precedent but it seems one or both were being used while Car charging hold was On.

The documentation suggests that these alternate figures will only be used if car_charging_energy is commented out. But it is not commented out. So it seems they are also used if car_charging_energy is not valid.

gcoan, 2026-08-21:
although this is an old issue, I suggest we need to leave it open to develop a working Predbat configuration for 3 phase GivEnergy inverters 

as noted on #2060 the 3 phase inverter doesn't have a concept of 'mode' (Eco, Timed Export etc) so we need a custom inverter definition

@LeeFarm11 @trevordowling  where did you get to with this?

LeeFarm11, 2026-08-21:
I gave up, hopefully temporarily, but it's been a while. I've been controlling everything manually for the last year or so. I was aiming to set up a Raspberry pi and reinstall HA and then try again, but just been too busy. Now my batteries are playing up - calibration does not work for me. I have just yesterday asked about it on facebook and Paul Landregan suggested raising a ticket with predbat.com support. At this point, I'm just fed up with the whole thing, but if my batteries come back to life, maybe I'll get some motivation back. 

Axle has been the only positive recently. They have recently managed to control my Inverter for the export events, normally without intervention from my side.

Ryuisnod, 2026-08-21:
> @LeeFarm11 <https://github.com/LeeFarm11> where did you get to with this?
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

I've been using givenergy cloud with a few custom entities to help control
things. Works well, if you want more information just give me a shout

trevordowling, 2026-08-21:
Just an update from my side...

Since the GE collapse,  I did a few things before they folded.

Collected what I believe to be a complete register map of the three phase
inverter.

Worked on code to read and write settings not normally accessible via the
GE interface with help from Claude. (I wanted to run my system in
unbalanced mode, which had been a suggestion from GE R&D engineer).

At the moment I've parked running Predbat,  as I have a simple tariff
structure that I'm just running with a fixed schedule.

Until such time that gas units cost more than my export payments, I'm
unlikely to need to change things.

However, that doesn't mean I don't want to see the three phase support in
PB, as I love the software and what it can do. I can see a time that I'll
need to be up and running again. I'm also likely to be exploring new
tariffs as British Gas tariffs are becoming less attractive, and I may also
consider adding Axle in to the mix.

Trevor

gcoan, 2026-08-21:
> I've been using givenergy cloud with a few custom entities to help control things. Works well, if you want more information just give me a shout

that sounds good. @Ryuisnod  

Can you share the details of your config, apps.yaml and custom sensors so we can add to the documentation and have a working 3 phase GivEnergy setup for others.

kennedy-nic, 2026-09-09:
Thanks @Ryuisnod – I've been wrestling with something similar on a 3-phase GivEnergy inverter and would really appreciate hearing more if  you're happy to share. Apologies if you've already covered some of this earlier in the thread and I've just missed it.

A few things that would help:

1. What inverter model are you on? I'm on the 3-phase HV (GIV-3HY-20), and some of what I've found seems quite specific to this model, so it'd be useful to know if yours is the same or different.

2. Are the custom entities you mention things you set up in Home Assistant yourself (e.g. via GivTCP, reading the inverter locally), or are they
   coming from the GivEnergy Cloud integration itself? I've found a few settings that exist and work fine when accessed one way, but are
   completely missing when accessed the other way – so knowing which side    yours are on would help a lot.

3. Is the issue you were solving mainly about the battery not holding its charge level and exporting solar properly (rather than just refilling
   itself), or was it more about charging/scheduling not working as expected?

No worries if you don't have time for a full write-up — even just a rough description of what you set up would help me work out if we're hitting the same underlying problem.

Ryuisnod, 2026-09-09:
hey,

yeah happy to help, i was planing to do a detailed write up when i had the time, this probably will become it 

1. GIV-3HY-11 is my model.
2. i completely gave up on GivTCP as it would work for a bit then stop working, think the modbus was locking up and i got fed up resetting everything to get it working again. I'm using the GE Cloud integration only, most of the custom sensors are for sensors that don’t do what Predbat wants. 

For example it wants to set discharge/charge rate  below 5% which the inverter won do so i have a helper sensor in predbat with an automation that sets the real one 5% if the helper is set to the 0% this wasn’t completely stopping Predbat from working but was a little 
annoying. code below for one but both are the same but with different input and outputs (don’t ask why both sensors weren’t done in watts or % as i'm sure i could have done it but this worked so i left it alone)

```
alias: "Battery: Adjust GE Inverter Charge Power Rate"
description: ""
triggers:
  - trigger: state
    entity_id:
      - input_number.predbat_battery_charge_watts
    to:
      - "12000"
      - "12000.0"
    id: set_to_100
  - trigger: state
    entity_id:
      - input_number.predbat_battery_charge_watts
    to:
      - "0"
      - "0.0"
    id: set_to_1
conditions: []
actions:
  - choose:
      - conditions:
          - condition: trigger
            id:
              - set_to_100
        sequence:
          - action: number.set_value
            metadata: {}
            target:
              entity_id: number.ge_inverter_td2418g007_charge_power_rate
            data:
              value: "100"
      - conditions:
          - condition: trigger
            id:
              - set_to_1
        sequence:
          - action: number.set_value
            metadata: {}
            target:
              entity_id: number.ge_inverter_td2418g007_charge_power_rate
            data:
              value: "1"
mode: single
```

also an automation to turn on/off "Enable Force Charge" when "Enable AC Charge"  changed state

i made a input select sensor to control the the mode since the inverter doesn’t seem to really have a 'mode' that works, basically Eco makes sure force discharge and charge are off. Timed charge and export flipped the switches required to to do either function, this is what i have in my apps.yaml

```
  inverter_mode: 'select.inverter_operating_mode'

  inverter_mode_options: 'Eco:Eco,Timed Charge:Timed Charge,Timed Export:Timed Export'
```

and this in configuration.yaml

```
- select:
[... 713 lines cut ...]
    -16: 2.2
    -14: 2.25
    -12: 2.3
    -10: 2.4
    -8: 2.5
    -6: 2.6
    -4: 2.7
    -2: 2.8
    0: 2.9
    2: 3.1
    4: 3.3
    6: 3.6
    8: 3.8
    10: 3.9
    12: 4.1
    14: 4.3
    16: 4.3
    18: 4.3
    20: 4.3
```
and is is my config

[predbat config.pdf](https://github.com/user-attachments/files/32019389/predbat.config.pdf)

any questions just let me know, hope this is related to your problems

kennedy-nic, 2026-09-09:
This is brilliant, thanks for the detail @Ryuisnod. Really useful, especially the custom mode-select entity trick, that's a cleaner way of solving the "no real inverter mode" problem – I think I'll explore that later this week.

One specific thing I'd love you to check, going off your apps.yaml naming pattern: does an entity called something like `number.ge_inverter_td2418g007_dc_discharge_1_lower_soc_limit` exist for you (mirroring your `ac_charge_1_upper_soc_limit`, but for discharge)?

I ask because I've found the GE Cloud API for my inverter (GIV-3HY-20) genuinely doesn't return that setting for slot 1 at all (confirmed by pulling the raw API response directly, and written up in #5017 if you're interested in the details). The equivalent exists fine for slots 3 onward. If yours does exist, that'd suggest it's specific to my inverter/API response rather than the 3-phase HV line generally; if it's missing for you too, that's useful to know as well and worth adding to that issue.

Ryuisnod, 2026-09-09:
"number.ge_inverter_td2418g007_dc_discharge_1_lower_soc_limit" I don't have. My entities also start at 3 for this section. You can set the lower discharge limit within predbat config section to set this. It will just change the mode when the limit is reached so this hasn't been a problem I've came across

gcoan, 2026-10-21:
> i completely gave up on GivTCP as it would work for a bit then stop working, think the modbus was locking up and i got fed up resetting everything to get it working again. I'm using the GE Cloud integration only, most of the custom sensors are for sensors that don’t do what Predbat wants.

@Ryuisnod @LeeFarm11 

Have you tried GivTCP 3.6?  There's been a lot of changes and fixes since 3.5 including new Modbus library and resolution of the 'loading too much data' issue that plagued the 3 phase and HV inverters.
From what I have seen these inverters should work much more reliably with GivTCP now

So would be worth revisiting and then seeing if we can get Predbat to work with it