springfall2008/batpred #1613: List index out of range error when fetching

Mikebert4, 2024-11-10:
Installed 8.5.10 this morning - I've been working on cost-per-kWh sensors on my installations so it was super nice to see you've been working on the same!

However, on both installs, I've been getting the following error:

`IndexError: list index out of range
2024-11-10 13:15:01.860628: Info: record_status Error: Exception raised list index out of range
2024-11-10 13:15:01.860801: Error: list index out of range
Traceback (most recent call last):
  File "/config/hass.py", line 184, in timer_tick
    item["callback"](None)
  File "/config/predbat.py", line 11226, in run_time_loop
    raise e
  File "/config/predbat.py", line 11221, in run_time_loop
    self.update_pred(scheduled=True)
  File "/config/predbat.py", line 9965, in update_pred
    self.fetch_sensor_data()
  File "/config/predbat.py", line 9028, in fetch_sensor_data
    soc_kwh_data[0],
IndexError: list index out of range`

This is on v8.5.10

**Environment details**
Two systems tested:

System 1:
- GivTCP Gen1 Hybrid
- Standard HAOS on Intel NUC

System 2
- GivTCP AIO
- Stabdard HAOS on Pi4
- 

Relevant logs:

`2024-11-10 13:15:00.606808: --------------- PredBat - update at 2024-11-10 13:15:00+00:00 with clock skew 0 minutes, minutes now 795
2024-11-10 13:15:00.607841: Saved current settings to /addon_configs/6adb4f0d_predbat/predbat_config.json
2024-11-10 13:15:00.607968: Using cached GITHub data for https://api.github.com/repos/springfall2008/batpred/releases age 42.1 minutes
2024-11-10 13:15:00.608053: Predbat /config/predbat.py version v8.5.10 currently running, latest version is v8.5.10 latest beta v8.5.10
2024-11-10 13:15:00.608120: Autoupdate: Currently up to date
2024-11-10 13:15:00.664957: Configuration: forecast_hours 36 num_cars 0 debug enable is False calculate_plan_every 15
2024-11-10 13:15:00.665559: Predbat mode is set to Control charge & discharge
2024-11-10 13:15:01.179155: iBoost energy today from sensor reads 0.0 kWh
2024-11-10 13:15:01.468972: Found 21601 load_today datapoints going back 15 days
2024-11-10 13:15:01.519380: select_event: update, select.predbat_update = v8.5.9 Fix bug in value per kwh attributes
2024-11-10 13:15:01.519473: Calling update service for v8.5.9 Fix bug in value per kwh attributes
2024-11-10 13:15:01.520150: Update Predbat to version v8.5.9 Fix bug in value per kwh attributes
2024-11-10 13:15:01.828570: Car charging hold True threshold 6.0
2024-11-10 13:15:01.828640: Current data so far today: load 14.74 kWh import 15.1 kWh export 0.0 kWh pv 0.5 kWh
2024-11-10 13:15:01.843765: Error: Exception raised list index out of range
2024-11-10 13:15:01.844014: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 11221, in run_time_loop
    self.update_pred(scheduled=True)
  File "/config/predbat.py", line 9965, in update_pred
    self.fetch_sensor_data()
  File "/config/predbat.py", line 9028, in fetch_sensor_data
    soc_kwh_data[0],
IndexError: list index out of range

2024-11-10 13:15:01.860628: Info: record_status Error: Exception raised list index out of range`

Mikebert4, 2024-11-10:
Also noticed this warning...
perhaps relevant?

`31258 | 2024-11-10 13:15:44.445020: Warn: No SoC data found for yesterday`

springfall2008, 2024-11-10:
> Also noticed this warning... perhaps relevant?
> 
> `31258 | 2024-11-10 13:15:44.445020: Warn: No SoC data found for yesterday`

Yes, I'll put a guard around this condition.

Has Predbat only just started running ?

springfall2008, 2024-11-10:
Fix merged to main to prevent the crash, the real question is why no data

Mikebert4, 2024-11-10:
> Has Predbat only just started running ?

Yup & thank you hugely for the fix!

In terms of why no data, that's confusing me a little - There's data in HA and apps.yaml seems to be set up correctly:

"sensor.{geserial}_soc_kwh" resolves to "sensor.solar_inverter_69_sa2244g163_soc_kwh", which in turn is an actual sensor, with data (see screenshot)
[image]

... and this data goes back beyond 'yesterday' ...

[image]

Though there are some spikes in the history that I'm putting down to GivTCP derps.

springfall2008, 2024-11-10:
Do you still get a warning about this in the logfile?

Mikebert4, 2024-11-10:
Every time the plan runs....

```
17783 | 2024-11-10 20:45:04.710633: Warn: No SoC data found for yesterday
17180 | 2024-11-10 20:30:04.827327: Warn: No SoC data found for yesterday
16571 | 2024-11-10 20:15:03.918687: Warn: No SoC data found for yesterday
15965 | 2024-11-10 20:00:04.129375: Warn: No SoC data found for yesterday
15357 | 2024-11-10 19:45:04.038734: Warn: No SoC data found for yesterday
14751 | 2024-11-10 19:30:04.513957: Warn: No SoC data found for yesterday
14143 | 2024-11-10 19:15:04.477581: Warn: No SoC data found for yesterday
13532 | 2024-11-10 19:00:05.037386: Warn: No SoC data found for yesterday
12919 | 2024-11-10 18:45:03.758787: Warn: No SoC data found for yesterday
12300 | 2024-11-10 18:30:04.481388: Warn: No SoC data found for yesterday
11688 | 2024-11-10 18:15:04.109449: Warn: No SoC data found for yesterday
11066 | 2024-11-10 18:02:56.725934: Warn: No SoC data found for yesterday
```

springfall2008, 2024-11-11:
Where is that message coming from, as it doesn't appear to be a Predbat message I can see in the code?

Mikebert4, 2024-11-11:
I've found the warning - it was removed in commit d6fded0 - (2 days ago, just prior to/as part of your releasing 8.5.10). 

Interesting, as I was getting the warning _after_ have upgraded to 8.5.10... which suggests that I wasn't upgrading properly!
I've un-installed and re-installed the addon and it looks like I'm on the correct release now, at least!

Thanks for you help - I'll get back to you if there's any further update!

And, just to add my voice to many - THANK YOU for building and maintaining this absolutely epic project.

closed 2024-11-11 by Mikebert4