springfall2008/batpred #1621: Charge and Discharge curves are not generated AC3

LeeFarm11, 2024-11-11:
**Describe the bug**
Charge and Discharge curves are not generated

**Expected behaviour**
Charge and Discharge curves should be generated in the log file after restart of Add-on

apps.yaml is looking for sensor.givtcp_{geserial}_battery_power
but I do not have battery_power.

I do have battery_charge_power and battery_discharge_power

**Predbat version**
1.2.1

**Environment details**
Inverter - GIV-3HY-11. 3 phase
Batteries - 19.94kWh
Stanadard HAOS in VMWare

**Screenshots**
[image]

**Log file**
```
2024-11-11 21:11:50.388132: Find charge curve with sensors sensor.givtcp_td2343g049_soc_kwh and number.givtcp_td2343g049_battery_charge_rate and predbat.status and sensor.givtcp_td2343g049_battery_power
2024-11-11 21:11:50.800935: Note: Cannot find battery charge curve (missing history), one of the required settings for predbat.status, soc_kw, battery_power and charge_rate do not have history, check apps.yaml
2024-11-11 21:11:50.802859: Find discharge curve with sensors sensor.givtcp_td2343g049_soc_kwh and number.givtcp_td2343g049_battery_discharge_rate and predbat.status and sensor.givtcp_td2343g049_battery_power
2024-11-11 21:11:51.204909: Note: Cannot find battery discharge curve (missing history), one of the required settings for predbat.status, soc_kw, battery_power and discharge_rate do not have history, check apps.yaml
```

springfall2008, 2024-11-12:
Hmm what about predbat.status?

gcoan, 2024-11-12:
> Side note - something went very wrong about 4am this morning. Restarted VM at about 8am and ALL my history (yes all of it) prior to 4am today has disappeared......
> 
> 

It sounds like you suffered from a HA database corruption.  If you look in the HA logs you will probably see something about a corrupt database error.  If HA thinks your database is corrupted then it moves the corrupt database to something like /config/ha_databasev2_corrupt_date_time.db and starts with a blank new database - you basically loose all your sensor history.

It is possible to move the corrupt database back to replace the blank HA database but without fixing the underlying issue, it can happen again.
There are instructions for how to fix a db corruption, to unload all of the data out of the corrupt database into a text file and reload it all again but its very much an advanced thing to do.

If you are fairly new with HA and can accept losing the history then I'm afraid you'll just have to live with it.

LeeFarm11, 2024-11-14:
Is this something for @britkat1980 ? Maybe sensor.givtcp_{geserial}_battery_power is not available on AC3 inverter so was removed when GivTCP recognises AC3 inverter? Maybe it can be reinstated with a calculation from somewhere? Or can predbat introduce a calculation depending on if it finds sensor.givtcp_{geserial}_battery_power or not?

gcoan, 2024-11-14:
Since we are find a few ways that the 3 phase inverter doesn't have the same controls as the rest of the GivEnergy range #1607 the suggestion from @springfall2008 to introduce a new inverter type into Predbat could get around these issues.

Would give the opportunity for the battery charge and discharge curve logic to take account of two different sensors rather than expect a single battery power sensor

Alternatively you could create a battery power template sensor from the two givtcp sensors you have, but it'd be a bit wasteful of HA database storage to duplicate all the data like that.

FYI the battery charge/discharge curve missing is not fatal to predbat, it just assumes your battery will smoothly charge and discharge at full rate to full/empty. Having the battery curves just makes the predbat plan a bit more accurate, that's all

chalfontchubby, 2026-08-29:
This has been quiet for a long time - gcoan noted the missing curve is cosmetic/non-fatal, and the discussion moved on to a possible new inverter type for the 3-phase AC3 (per #1607) which was never taken further. Is this still worth tracking, or has it become moot? Let us know and we'll keep it open, otherwise we'll close it in a week.

gcoan, 2026-08-29:
I think it's still an issue with 3 phase AC inverters as givtcp doesn't provide the battery_power sensor that the charge curve routine is looking for.   Creating a custom inverter type for 3 phase wouldn't improve the situation, it would just enable predbat to better handle the other inverter differences,  but the charge/discharge curve would still be unavailable.

Charge/discharge curves can be created for other inverter types successfully, as long as they can provide the required entities.

The workaround is to create a combined battery power template sensor and use that for creating the curves.

Or (more generic solution) enhance predbat's charge/discharge curve calculation to either accept a single battery power or battery charge and discharge sensors.

Depends what demand there is for this