springfall2008/batpred #1622: Battery pause error in v8.6.0

SwiftRR, 2024-11-12:
**Describe the bug**
Error message in Predbat addon log after upgrading from givTCP v2.4.9 to v3.0.4:
[image]

in v2.4.9, pause instructions were read successfully (on GE AC3.0, updated firmware), e.g.
[image]

v8.6.0 Features state: 'Determine automatically if GivTCP V3 is installed and use new REST API for Pause mode'

**Expected behaviour**
No error

**Predbat version**
8.6.0

**Environment details**
- Inverter and battery setup: GE AC3.0 daisychaned to two 8.2 kWh batteries
- Standard Predbat on Raspberry pi HAOS (latest version)

mpartington, 2024-11-12:
The graceful handing part is not really an error. This is the explanation from Mark

"The Critical log is just for me to see how prevalent a fw bug in the Gen1 is. That can be safely ignored, it’s not really an error, in as far as I made the lib handle the none standard function code response properly"

For the second point, have you hashed out the pause times in apps.yaml, as they are not available on AC3? Otherwise, maybe it's linked to the new REST control

mpartington, 2024-11-12:
Looks like Trefor has already fixed the REST error relating to the pause slot , in main. He doesn't hang around 🤣
[image]

mpartington, 2024-11-12:
Great stuff! I have the same inverter and it's a register issue, unfortunately there aren't enough spare time for all the functionality the later gens have.
Fingers crossed that's sorted now