springfall2008/batpred #1623: Changes to optimise GivEnergy firmware 3018

mpartington, 2024-11-13:
**Describe the bug**
Investigation by GivEnergy has identified that the peak voltage achieved during charge can reduce over time, resulting in reduced capacity and sudden SoC drops

New firmware 3018 for GivEnergy LV batteries is currently in beta.  This contains a new feature to allow batteries to continue charging even if SoC is reading 100% if the system to maintain a higher voltage and ensure batteries are full

The current Predbat logic has the potential to disrupt the new firmware function as it turns off enable charge schedule soon as the battery hits 100%

I have tried Trefors recommendation to set the reserve to 98% (from 100%) in apps.yaml, however this made no difference

**Expected behaviour**
It the battery is in a charge window and Predbat is not discharging, then enable charge schedule should be left switched on for GivEnergy LV inverters.

Personally on IOG tariff overnight, I would prefer it to be left on at all times with the default charge window (23.30-05:30), as it currently only gets switched on 30 mins before a charge session, so adds risk of a no charge scenario if there is a homeassistant or network failure. Although I guess this may not work for all tariffs

**Predbat version**

8.6.0

**Environment details**

- AC3 with 2x 5.2kWh batteries
- Standard HAOS installer

**Screenshots**
[image]

**Log file**
N/A

wilsondw, 2024-11-13:
Just to second this request. 3018 continues to charge after a reported SOC of 100% to find the real 100%, but Predbat currently assumes reported 100% is really 100% and switches off charging. Switching off charging interferes with the SOC correction in 3018 and prevents it from running to completion. The image below shows the behaviour with predbat in monitor only mode over a timed AC charge and shows that the continued top up that 3018 performs when it is given the chance.

[image]

mpartington, 2024-11-21:
Tested fix to this last night and confirmed all working - thank you.

closed 2024-11-21 by mpartington

nbullus, 2024-12-16:
What was the Predbat fix and what version was this rolled out at? Is the Predbat fix specific to this type of inverter or all GivEnergy inverters? I am asking as I'm having SOC capacity issues with my AIO. GivEnergy have indicated that a Beta firmware release will be forthcoming and I have been selected to trial it, so I want to make sure I can continue to use Predbat whilst this trial is going on.

mpartington, 2024-12-16:
It was over a month ago, but unsure of version. Essentially Predbat was turning off enable charge schedule' soon as it hit 100% soc. It now leaves it on for the full charge period, which allowed mini calibrations to occur post 100%

Unfortunately for me 3018 was the worst firmware for a long time, but seems to work great for others. I lost around 1kWh of capacity on it and have downgraded.

nbullus, 2024-12-16:
Thanks I'll try a few things out. I'm losing about 25% capacity daily now and it looks like a SOC issue. I'm not sure how long I have had it, but I've only really noticed after I switched to Octopus Go at the beginning of Nov. It may have been happening for much longer but not easy to spot on Agile as it charges more dynamically anyway and never leaves you short. I took a firmware upgrade a few weeks back, but I'm sure I was having SOC issues before this. GE are working on a Beta firmware upgrade but haven't said what exactly it will fix or what the problem is. I do know that when I did a calibration few days ago (which didn't fix the issue), it took nearly 2 hrs to discharge battery from 4% reserve to 0%. This tells me that 4% SOC is actually 25% SOC in real money. However, I'm putting it on a charge only tonight for the full 5 hours to see if the battery stops charging at 100% or carries on. I suspect the upper SOC limit is fine and its actually the lower limit its struggling to compute. I was just unsure if Trefor's fix was specific for the AC3 inverter or whether it would charge past 100% on my AIO.  I'll see what happens tonight.