springfall2008/batpred #1624: Standing charge still shown in Rates chart for Today p/kWh

SwiftRR, 2024-11-15:
**Describe the bug**
Contrary to 'What's changed' in v8.5.6 , standing charge still shows in Rates chart for Today p/kWh.
This gives a big skew round about midnight. 
This occurs in WebUI Rates chart and in Energy rates chart from withinTemplates/example_chart.yml 

[image]
[image]

**Expected behaviour**
Standing charge disappears!

**Predbat version**
v8.5.6 onwards

**Environment details**

- Inverter and battery setup
- Standard HAOS installer or Docker
- Anything else?

**Screenshots**
See above

**Log file**
N/A

springfall2008, 2024-11-16:
What version are you on now as there were fixes in 8.5.10?

springfall2008, 2024-11-16:
Can you share the YAML of our Apex charge, maybe there is something wrong with it?

SwiftRR, 2024-11-16:
The values between the WebUI and Apex charts all look the same (or close to the same) except for the values at midnight.

In my apps.yaml, I have the metric_standing_charge line commented out.

This is the chart code I have been using. Copied from chart.yml. 
As I have no gas, I have to delete the section:  '- entity: predbat.rates_gas' 
```
###############################################
# Energy rate chart
# NOTE: Tune the value*5 in the grid_power_best part to make the power peak line up with the energy rates
# so you can see charging/discharging aligned with slots
###############################################
type: custom:apexcharts-card
header:
  show: true
  title: Energy rates
  show_states: true
  colorize_states: true
graph_span: 72h
span:
  start: day
  offset: -24h
now:
  show: true
series:
  - entity: predbat.grid_power_best
    name: power
    stroke_width: 1
    type: area
    opacity: 0.2
    data_generator: >
      let res = []; for (const [key, value] of
      Object.entries(entity.attributes.results)) { res.push([new
      Date(key).getTime(), value*5]); } return res.sort((a, b) => { return a[0] -
      b[0]  })
  - entity: predbat.rates
    attribute: threshold
    name: Low threshold
    curve: stepline
    stroke_width: 2
  - entity: predbat.rates_export
    attribute: threshold
    name: High threshold
    curve: stepline
    stroke_width: 2
  - entity: predbat.rates
    stroke_width: 2
    curve: stepline
    name: import
    data_generator: >
      let res = []; for (const [key, value] of
      Object.entries(entity.attributes.results)) { res.push([new
      Date(key).getTime(), value]); } return res.sort((a, b) => { return a[0] -
      b[0]  })
    show:
      in_header: raw
  - entity: predbat.rates_export
    stroke_width: 1
    type: area
    opacity: 0.2
    curve: stepline
    name: export
    data_generator: >
      let res = []; for (const [key, value] of
      Object.entries(entity.attributes.results)) { res.push([new
      Date(key).getTime(), value]); } return res.sort((a, b) => { return a[0] -
      b[0]  })
    show:
      in_header: raw
  - entity: predbat.rates_gas
    stroke_width: 1
    type: area
    opacity: 0.2
    curve: stepline
    name: gas
    data_generator: >
      let res = []; for (const [key, value] of
      Object.entries(entity.attributes.results)) { res.push([new
      Date(key).getTime(), value]); } return res.sort((a, b) => { return a[0] -
      b[0]  })
    show:
      in_header: raw
  - entity: predbat.ppkwh_hour
    stroke_width: 2
    type: line
    curve: stepline
    name: Hourly p/kWh
    extend_to: false
  - entity: predbat.ppkwh_today
    stroke_width: 2
    type: line
    curve: stepline
    name: Today p/kWh
    extend_to: false
```

Rob

SwiftRR, 2024-11-16:
Is the discrepancy caused by battery value in attributes of Today p/kWh. See screenshot below
The value is close to the difference between the WebUI and Apex charts.
**Screenshot**
[image]

Rob

SwiftRR, 2024-11-23:
The spikes at midnight have now disappeared and WebUI and Apex charts look very similar.
So this issue seems to have been fixed or has worked its way through HA..

My only niggle now is that the Apex chart shows almost identical line colours for 'hourly' and 'today'. You can see this in my earlier screenshots. I guess that there is a way to tweak colours within Apex settings – something for me to investigate.

Rob

trevordowling, 2025-02-25:
Running version: Current version: 1.2.4 (Changelog) 

Hi, 

I'm also seeing high rate estiamtes at the crossover boundaries between different tariffs. 

[image]

Is this the same issue as reported here?

I wondered if this was linked to the rate settings in apps.yaml

snip from my apps.yaml are as follows: (Note: I'm not using any Octopus 'suto' data pulling, as I am with BGas)

                # Or set your actual rates across time for import and export
                  # If start/end is missing it's assumed to be a fixed rate
                  # Gaps are filled with zero rate
                  rates_import:
                    -  start: "00:00:01"
                       end: "05:00:00"
                       rate: 7.9
                    -  start: "05:00:01"
                       end: "00:00:00"
                       rate: 27.002
                  #
                  rates_export:
                    -  rate: 15.0

So I adjusted these to ensure no gap / overlap, but this made no difference.

I've also added here a chart that shows the time periods of the 'steps' where the calculation improves over time

[image]

gcoan, 2025-02-25:
> Running version: Current version: 1.2.4 (Changelog)
>
> Hi,
>
> I'm also seeing high rate estiamtes at the crossover boundaries between different tariffs.
>
> [image]
>
> Is this the same issue as reported [...]

Per the documentation, start and end time should align to 30 minute boundaries https://springfall2008.github.io/batpred/energy-rates/#rate-bands-to-manually-configure-energy-rates

So don't put 00:00:01, put 00:00:00 as the start time is the correct way to configure predbat

trevordowling, 2025-02-25:
Brilliant,  and thanks, I did have them compliant but I also had a 2 minute
clock skew, which possibly broke the adherence to 30 min boundaries... I'll
update (I've also removed the clock skew) and check this fixes it.

Thanks

Trevor

trevordowling, 2025-02-26:
Hi,

I let the system run overnight and I still see these high values on the boundaries of the rate changes

Settings from apps.yaml
  
    Rates
    
       rates_import:
        -  start: "00:00:00"
           end: "05:00:00"
           rate: 7.9
        -  start: "05:00:00"
           end: "00:00:00"
           rate: 27.002
    
    Skew settings
    
      inverter_clock_skew_start: 0
      inverter_clock_skew_end: 0
      inverter_clock_skew_discharge_start: 0
      inverter_clock_skew_discharge_end: 0
  
  
[image]

Any ideas?

Please shout if you need any additional data or tests run. (low priority from my side, as it doesn't impact operation)

Thanks

Trevor

gcoan, 2025-02-27:
Do you have a standing charge defined in apps.yaml and are you on the latest verison of predbat ?

I'm thinking its this that's corrupting the chart because at midnight, cost today and kwh today is reset to zero,  but standing charge applied and after a tiny but of energy consumed you'll get a spike of cost/kWh if standing charge is included in the calculation

trevordowling, 2025-02-27:
Hi,

Yes, I do have a standing charge defined
[image: image.png]

I'm running version
[image: image.png]
I've changed settings around the rates and standing changes to see if I
could confirm if the standing charge is the issue, but it looks like the
charts are only generated once, as I don't see any changes to the data that
appears in the past (data to the left of the NOW line).

[image: image.png]

Hope this helps

Trevor

gcoan, 2025-02-27:
just as a debug, could you comment the standing charge line out of apps.yaml and let the comparison run as normal through midnight to see if this resolves the problem please

and to confirm,you are on the latest predbat version (not the addon version, the predbat version) - it'll be displayed at the top of your HTML predbat plan

trevordowling, 2025-02-27:
Will do.

Version is v8.15.1 (apologies for grabbing the wrong version ID)

I'm also looking at significant register writes (hundreds a day, which is
not good), which needs attention. There are a significant number of writes
to set Eco mode every 5 minutes... I don't know if this is normal, but if
nothing else it is polluting my logs while I'm trying to track down why
forced export is not working correctly. I've been through the documentation
notes on items to set to minimise the flash writes, not sure if flash
writes and register writes are the same.

[image: image.png]

2025-02-27 06:22:15,368 - GivTCP - write       -  [INFO    ] - Setting
Eco mode was a success
2025-02-27 06:25:08,804 - GivTCP - write       -  [INFO    ] - Setting
Eco mode was a success
2025-02-27 06:29:52,669 - GivTCP - write       -  [INFO    ] - Setting
Eco mode was a success
2025-02-27 06:30:46,212 - GivTCP - write       -  [INFO    ] - Setting
Eco mode was a success
2025-02-27 06:32:20,035 - GivTCP - write       -  [INFO    ] - Setting
Eco mode was a success
2025-02-27 06:35:04,446 - GivTCP - write       -  [INFO    ] - Setting
Eco mode was a success
2025-02-27 06:40:07,537 - GivTCP - write       -  [INFO    ] - Setting
Eco mode was a success
2025-02-27 06:45:03,291 - GivTCP - write       -  [INFO    ] - Setting
Eco mode was a success
2025-02-27 06:50:07,511 - GivTCP - write       -  [INFO    ] - Setting
Eco mode was a success
2025-02-27 06:55:03,582 - GivTCP - write       -  [INFO    ] - Setting
Eco mode was a success
2025-02-27 07:00:04,160 - GivTCP - write       -  [INFO    ] - Setting
Eco mode was a success
2025-02-27 07:05:04,924 - GivTCP - write       -  [INFO    ] - Setting
Eco mode was a success
2025-02-27 07:10:04,555 - GivTCP - write       -  [INFO    ] - Setting
Eco mode was a success
2025-02-27 07:15:04,354 - GivTCP - write       -  [INFO    ] - Setting
Eco mode was a success
2025-02-27 07:20:05,064 - GivTCP - write       -  [INFO    ] - Setting
Eco mode was a success
2025-02-27 07:25:04,407 - GivTCP - write       -  [INFO    ] - Setting
Eco mode was a success

I have checked the Inverter logs and it doesn't look like these end up
being actual writes to the flash, which is good, but I'm still not
sure why the GivTCP logs are showing this many entries for this.
Possible, this is normal behaviour, but I need to look a bit more
before opening a new entry.

While I've got this to look at, I've suspended the system by setting the '
Read Only mode' but I assume the rate calculations will be fine and so
hopefully we can see if this 'fixes' things overnight.

Thanks

Trevor

gcoan, 2025-02-27:
> I'm also looking at significant register writes (hundreds a day, which is not good), which needs attention. There are a significant number of writes to set Eco mode every 5 minutes... I don't know [...]

No that's not right, it shouldn't be doing that, certainly not every 5 minutes.  Looks like its trying to make the change but the change isn't working.
That looks like its in the GivTCP log?
Do check that inverter_mode in apps.yaml is set correctly, and maybe restart givtcp and/or your inverter if its not responding to the command.
If necessary create a new issue.

trevordowling, 2025-02-27:
Hi,

Yep good to hear that this isn't correct behaviour. I'll continue to dig a
bit before formally raising the issue.

The three phase system has a few differences and even the GivEnergy app
doesn't work correctly in all areas.

Thanks

Trevor

trevordowling, 2025-02-27:
> ah, you have a 3 phase inverter. There are still some teething problems
> with that inverter and givtcp/predbat. There's other givtcp threads on
> getting it working, I think you need to be on the dev version of givtcp for
> maximum compatibility, but do look at the other threads

Will do, yes, I've seen a few of us on the 3 phase, many comments like
'there aren't many of us'

Everyone's been great and helpful. Many wanting help from us to make their
code support the 3 phase systems... I'm really inspired by the community
approach...

I'm probably too rusty to add new code, but very happy to test and debug as
needed.

Trevor

trevordowling, 2025-03-01:
@gcoan asked 

'just as a debug, could you comment the standing charge line out of apps.yaml and let the comparison run as normal through midnight to see if this resolves the problem please'

Line commented out in apps.yaml

      #metric_standing_charge: 0.487

Let is run a couple of days, sadly it still shows the peaks (but as before, low priority on this one)

[image]

gcoan, 2025-03-01:
Been looking at the charts you posted here and earlier https://github.com/springfall2008/batpred/issues/1624#issuecomment-2680973540

In the earlier ones the spikes were to 200p,  this most recent one its to about 75p?

My own chart which has never had standing charge include in the plan doesn't show big positive spikes as you have, but instead has a smaller negative one on the today figure, and some massive negative on the hourly figure
[image]

Are you charging at midnight?

I am on Cosy so predbat stops charging at midnight and goes into Eco/Demand mode.

Am wondering if the problem is down to one of the sensors not being updated as quickly as the other one, so the cost/energy figure gets some wacky numbers, and me discharging at that time and you (possibly) charging at that time explains why the sign is different

trevordowling, 2025-03-01:
> Been looking at the charts you posted here and earlier #1624 (comment)

Yes, my charge window is 00:00:00 - 05:00:00

chalfontchubby, 2026-08-20:
Thanks for the report. This issue has been open for over a year with no recent activity - we're volunteers working through a large backlog and need to filter what's still relevant. If this is still a problem for you, please let us know and we'll keep it open. Otherwise, we'll close it in a week (around 2026-08-28).

trevordowling, 2026-08-21:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

For now I'm happy if this is closed out, due to my current tariff I'm just
running a very basic setup, so this is not having an impact on my system
operation at this time.

Thanks

Trevor

closed 2026-08-21 by gcoan