springfall2008/batpred #1625: Not recognising that Inverter is calibrating AC3

LeeFarm11, 2024-11-15:
**Describe the bug**
According to documentation if the Battery is calibrating, predbat will go into predbat.status = Calibrating but it did not

**Expected behaviour**
I scheduled a battery calibration via GivEnergy Portal for 00:05 on 14th November. I expected predbat.status to change to Calibrating while the calibration was taking place. But it did not.

**Predbat version**
1.2.1

**Environment details**
Inverter - GIV-3HY-11. 3 phase
Batteries - 19.94kWh
Stanadard HAOS in VMWare

**Screenshots**
[image]
[image]

[image]

**Log file**
Sorry, log files only seem to go back about 24 hours. Are they stored somewhere else for longer?

gcoan, 2024-11-16:
This is probably a problem with GivTCP reporting that the 3 phase inverter is in calibration mode in a different way to how the other inverter types report a calibration.  Predbat looks for a specific status in GivTCP (can't remember what it is) and I suspect GivTCP is reporting something differently.

r/e logfile, there should be a /addon_configs/6adb4f0d_predbat/predbat.1.log which is the previous logfile.
I had previously put through a change to increase the number of log files to 9 but for some reason this isn't happening, but the prior file is available

springfall2008, 2024-11-16:
You would probably need to trigger another calibration and then read out the REST data to see what its set to, note that this information only comes via REST right right

LeeFarm11, 2024-11-16:
> r/e logfile, there should be a /addon_configs/6adb4f0d_predbat/predbat.1.log which is the previous logfile. I had previously put through a change to increase the number of log files to 9 but for some reason this isn't happening, but the prior file is available

Yes the predbat.1.log is the one I found which went back further, but only to about 24 hours ago.

LeeFarm11, 2024-11-16:
> You would probably need to trigger another calibration and then read out the REST data to see what its set to, note that this information only comes via REST right right

Will try to work out how to get that data before I next do a calibration.

LeeFarm11, 2024-11-17:
Hi. 
At 13:14 I scheduled a calibaration to start at 13:17.

Here are REST reads -
1) before scheduling calibaration. " "Battery_Calibration": "Off","
2) after scheduling calibaration but before calibration started. " "Battery_Calibration": "Off","
3) after calibration started. ""Battery_Calibration": "Start","
4) later in calibaration process. While charging. "       "Battery_Calibration": "Charge Only","

I don't know if I spotted all the different possibilities. Is there enough information here or how can I record REST responses over a period of time?

[1.txt](https://github.com/user-attachments/files/17790606/1.txt)
[2.txt](https://github.com/user-attachments/files/17790605/2.txt)
[3.txt](https://github.com/user-attachments/files/17790603/3.txt)
[4.txt](https://github.com/user-attachments/files/17791126/4.txt)

mpartington, 2024-12-29:
It doesn't detect calibration for me running GivTCP V3 either. I grabbed the REST data for you while it was in progress
'''
{ "Battery_Details": { "Battery_Stack_1": { "BE2152G380": { "Battery_Capacity": 93.46, "Battery_Cell_10_Voltage": 3.47, "Battery_Cell_11_Voltage": 3.515, "Battery_Cell_12_Voltage": 3.48, "Battery_Cell_13_Voltage": 3.478, "Battery_Cell_14_Voltage": 3.535, "Battery_Cell_15_Voltage": 3.515, "Battery_Cell_16_Voltage": 3.494, "Battery_Cell_1_Temperature": 20.3, "Battery_Cell_1_Voltage": 3.562, "Battery_Cell_2_Temperature": 20.5, "Battery_Cell_2_Voltage": 3.465, "Battery_Cell_3_Temperature": 20.2, "Battery_Cell_3_Voltage": 3.549, "Battery_Cell_4_Temperature": 20.5, "Battery_Cell_4_Voltage": 3.48, "Battery_Cell_5_Voltage": 3.507, "Battery_Cell_6_Voltage": 3.43, "Battery_Cell_7_Voltage": 3.499, "Battery_Cell_8_Voltage": 3.532, "Battery_Cell_9_Voltage": 3.61, "Battery_Cells": 16, "Battery_Cycles": 810, "Battery_Design_Capacity": 102.0, "Battery_Firmware_Version": 3017, "Battery_Remaining_Capacity": 93.46, "Battery_SOC": 100, "Battery_Serial_Number": "BE2152G380", "Battery_Temperature": 17.7, "Battery_USB_present": 8, "Battery_Voltage": 56.11 }, "BE2209G120": { "Battery_Capacity": 95.8, "Battery_Cell_10_Voltage": 3.536, "Battery_Cell_11_Voltage": 3.483, "Battery_Cell_12_Voltage": 3.581, "Battery_Cell_13_Voltage": 3.478, "Battery_Cell_14_Voltage": 3.506, "Battery_Cell_15_Voltage": 3.594, "Battery_Cell_16_Voltage": 3.607, "Battery_Cell_1_Temperature": 19.6, "Battery_Cell_1_Voltage": 3.395, "Battery_Cell_2_Temperature": 19.8, "Battery_Cell_2_Voltage": 3.412, "Battery_Cell_3_Temperature": 19.4, "Battery_Cell_3_Voltage": 3.564, "Battery_Cell_4_Temperature": 20.0, "Battery_Cell_4_Voltage": 3.461, "Battery_Cell_5_Voltage": 3.399, "Battery_Cell_6_Voltage": 3.557, "Battery_Cell_7_Voltage": 3.536, "Battery_Cell_8_Voltage": 3.527, "Battery_Cell_9_Voltage": 3.511, "Battery_Cells": 16, "Battery_Cycles": 820, "Battery_Design_Capacity": 102.0, "Battery_Firmware_Version": 3017, "Battery_Remaining_Capacity": 95.8, "Battery_SOC": 100, "Battery_Serial_Number": "BE2209G120", "Battery_Temperature": 17.2, "Battery_USB_present": 8, "Battery_Voltage": 56.143 }, "BMS_Temperature": 20.0, "BMS_Voltage": 56.45 } }, "CE2146G269": { "Battery_Capacity_kWh": 10.44, "Battery_Type": "Lithium", "Export_Limit": 6000, "Invertor_Firmware": "D0.205-A0.205", "Invertor_Max_Bat_Rate": 3000, "Invertor_Max_Inv_Rate": 3000, "Invertor_Serial_Number": "CE2146G269", "Invertor_Temperature": 27.0, "Invertor_Time": "2024-12-25T05:25:28+00:00", "Invertor_Type": "Gen1 Ac", "Meter_Type": "Em115", "Modbus_Version": "1.40" }, "Control": { "Active_Power_Rate": 100, "Battery_Calibration": "Charge Only", "Battery_Charge_Rate": 3000, "Battery_Charge_Rate_AC": 100, "Battery_Discharge_Rate": 3000, "Battery_Discharge_Rate_AC": 100, "Battery_Power_Cutoff": 4, "Battery_Power_Reserve": 4, "Battery_pause_mode": "Disabled", "Eco_Mode": "enable", "Enable_Charge_Schedule": "enable", "Enable_Charge_Target": "enable", "Enable_Discharge_Schedule": "disable", "Force_Charge": "Normal", "Force_Charge_Num": 0, "Force_Export": "Normal", "Force_Export_Num": 0, "Mode": "Eco", "Reboot_Addon": "disable", "Reboot_Invertor": "disable", "Sync_Time": "disable", "Target_SOC": 100, "Temp_Pause_Charge": "Normal", "Temp_Pause_Charge_Num": 0, "Temp_Pause_Discharge": "Normal", "Temp_Pause_Discharge_Num": 0 }, "Energy": { "Rates": { "Battery_Value": 1.399, "Battery_ppkwh": 0.134, "Current_Rate": 0.155, "Current_Rate_Type": "Night", "Day_Cost": 0.0, "Day_Energy_Total_kWh": 0, "Day_Energy_kWh": 0.0, "Day_Rate": 0.395, "Day_Start_Energy_kWh": 15451.6, "Export_Rate": 0.04, "Import_ppkwh_Today": 0.155, "Night_Cost": 2.1544999999999437, "Night_Energy_Total_kWh": 0, "Night_Energy_kWh": 13.899999999999636, "Night_Rate": 0.155, "Night_Start_Energy_kWh": 15451.6 }, "Today": { "AC_Charge_Energy_Today_kWh": 10.1, "Battery_Charge_Energy_Today_kWh": 9.8, "Battery_Discharge_Energy_Today_kWh": 0.1, "Battery_Throughput_Today_kWh": 9.9, "Export_Energy_Today_kWh": 0.0, "Import_Energy_Today_kWh": 13.9, "Invertor_Energy_Today_kWh": 0.1, "Load_Energy_Today_kWh": 3.9, "PV_Energy_Today_kWh": 0.0, "Self_Consumption_Energy_Today_kWh": 0 }, "Total": { "AC_Charge_Energy_Total_kWh": 7452.7, "Battery_Charge_Energy_Total_kWh": 668.3, "Battery_Discharge_Energy_Total_kWh": 6272.9, "Battery_Throughput_Total_kWh": 13494.8, "Export_Energy_Total_kWh": 3918.6, "Import_Energy_Total_kWh": 15465.5, "Invertor_Energy_Total_kWh": 6023.9, "Load_Energy_Total_kWh": 18540.3, "PV_Energy_Total_kWh": 8422.2, "Self_Consumption_Energy_Total_kWh": 4503.6 } }, "Meter_Details": {}, "Power": { "Flows": { "Battery_to_Grid": 0, "Battery_to_House": 0, "Grid_to_Battery": 558, "Grid_to_House": 292, "Solar_to_Battery": 0, "Solar_to_Grid": 0, "Solar_to_House": 2 }, "Power": { "AC_Charge_Power": 544, "Battery_Current": null, "Battery_Power": -558, "Battery_Voltage": 56.45, "Charge_Completion_Time": "2024-12-25T05:25:31.517660+00:00", "Charge_Power": 558, "Charge_Time_Remaining": 0, "Combined_Generation_Power": null, "Discharge_Power": 0, "Discharge_Time_Remaining": 0, "EPS_Power": 0, "Export_Power": 0, "Grid_Current": 3.86, "Grid_Frequency": 50.04, "Grid_Power": -850, "Grid_Voltage": 242.7, "Import_Power": 850, "Inverter_Output_Frequency": 50.04, "Invertor_Power": -544, "Load_Power": 308, "PV_Current_String_1": 0.0, "PV_Current_String_2": 0.0, "PV_Power": 2, "PV_Power_String_1": 2, "PV_Power_String_2": 0, "PV_Voltage_String_1": 245.3, "PV_V [... cut ...] battery": 20.0, "temp_charger": 26.3, "temp_inverter_heatsink": 27.0, "threephase_abc": 0, "threephase_balance_1": 0, "threephase_balance_2": 0, "threephase_balance_3": 0, "threephase_balance_mode": 0, "usb_device_inserted": 0, "user_code": 7, "v_ac1": 242.7, "v_battery": 56.45, "v_eps_backup": 245.3, "v_highbrigh_bus": 316.5, "v_n_bus": 0.0, "v_p_bus": 385.2, "v_pv1": 245.3, "v_pv2": 0.0, "v_pv_start": 150.0, "variable_address": 49152, "variable_value": 0, "work_time_total": 22698 } } } '''
'''

LeeFarm11, 2025-01-15:
Hi @gcoan or @springfall2008 is there any update on this please?
Thanks very much.

mpartington, 2025-01-17:
Seems fixed for me. Predbat detected my user initiated calibration

chalfontchubby, 2026-08-28:
No further activity since our notice, so closing this as stale. If this is still a problem for you, feel free to reopen with a fresh reproduction case and we'll take another look.

closed 2026-08-28 by chalfontchubby

gcoan, 2026-08-28:
I think I fixed this, battery calibration wasn't being detected with GivTCP v3+, I added logic for this a while ago