springfall2008/batpred #1626: Intelligent Octopus rate triggers when car is charging but dispatch not set

chris-y, 2024-11-15:
**Describe the bug**
Predbat activates the cheaper rate during the day as soon as the car starts charging, and triggers the battery to charge up in this slot - even though it is not the scheduled dispatch slot, and the car was only charging momentarily before the smart charge kicked in

**Expected behaviour**
Predbat should not use the car charging as a trigger to set the slot as cheap rate, unless it is specified in the IOG dispatched or dispatch is on.

**Predbat version**

Latest

**Environment details**

- Solis inverter
- Add-on version

**Screenshots**
Didn't get a screenshot sorry

**Log file**
225962024-11-15 18:46:08.528351: Warn: Historical day 4 has 10 minutes of gap in the data, filled from 19.81 kWh to make new average 19.95 kWh (percent 99%)225952024-11-15 18:46:08.511991: Warn: Historical day 3 has 150 minutes of gap in the data, filled from 24.68 kWh to make new average 27.55 kWh (percent 90%)225282024-11-15 18:45:49.975563: Watched event: sensor.xx20xxx_charge_state = {'entity_id': 'sensor.xx20xxx_charge_state', 'state': 'waiting_for_a_planned_charge', 'attributes': {'options': ['not_in_charge', 'waiting_for_a_planned_charge', 'charge_ended', 'waiting_for_current_charge', 'energy_flap_opened', 'charge_in_progress', 'charge_error', 'unavailable'], 'device_class': 'enum', 'friendly_name': 'XX20XXX Charge state'}, 'last_changed': '2024-11-15T18:45:49.969218+00:00', 'last_reported': '2024-11-15T18:45:49.969218+00:00', 'last_updated': '2024-11-15T18:45:49.969218+00:00', 'context': {'id': '01JCRJAMRHK4M27BF99C55C3N0', 'parent_id': None, 'user_id': None}} will trigger re-plan224332024-11-15 18:45:26.589288: Warn: Inverter 0: Using inverter type MINE - not all features are available224272024-11-15 18:45:26.278369: Warn: Historical day 7 has 65 minutes of gap in the data, filled from 19.73 kWh to make new average 20.66 kWh (percent 95%)224262024-11-15 18:45:26.242734: Warn: Historical day 6 has 175 minutes of gap in the data, filled from 18.73 kWh to make new average 21.32 kWh (percent 88%)224252024-11-15 18:45:26.224759: Warn: Historical day 5 has 70 minutes of gap in the data, filled from 17.73 kWh to make new average 18.63 kWh (percent 95%)224242024-11-15 18:45:26.217579: Warn: Historical day 4 has 10 minutes of gap in the data, filled from 19.81 kWh to make new average 19.95 kWh (percent 99%)224232024-11-15 18:45:26.201303: Warn: Historical day 3 has 150 minutes of gap in the data, filled from 24.68 kWh to make new average 27.55 kWh (percent 90%)223082024-11-15 18:45:09.382735: Warn: Inverter 0: Using inverter type MINE - not all features are available221962024-11-15 18:44:38.836414: Warn: Inverter 0: Using inverter type MINE - not all features are available221902024-11-15 18:44:38.54419
 21534	2024-11-15 18:39:09.210274: Warn: Historical day 3 has 150 minutes of gap in the data, filled from 24.68 kWh to make new average 27.54 kWh (percent 90%)
21466	2024-11-15 18:38:49.640848: Watched event: sensor.xx20xxx_charge_state = {'entity_id': 'sensor.xx20xxx_charge_state', 'state': 'charge_in_progress', 'attributes': {'options': ['not_in_charge', 'waiting_for_a_planned_charge', 'charge_ended', 'waiting_for_current_charge', 'energy_flap_opened', 'charge_in_progress', 'charge_error', 'unavailable'], 'device_class': 'enum', 'friendly_name': 'XX20XXX Charge state'}, 'last_changed': '2024-11-15T18:38:49.634925+00:00', 'last_reported': '2024-11-15T18:38:49.634925+00:00', 'last_updated': '2024-11-15T18:38:49.634925+00:00', 'context': {'id': '01JCRHXT92FC46WAK6VM0E21W1', 'parent_id': None, 'user_id': None}} will trigger re-plan
21427	2024-11-15 18:38:41.113189: Warn: Inverter 0: Using inverter type MINE - not all features are available
21314	2024-11-15 18:38:08.947502:

mpartington, 2024-11-15:
I think I had this issue, and Trefor recommend not using car charging now unless you have issues with IO.

I assume you've set this up in apps.yaml, so best to hash it out. If it sees the car charging, it sets the slot as.a 7p and the battery charge kicks off.

chris-y, 2024-11-15:
Yes, I've commented it out now. I have car_charging_planned too - is that needed? There's a comment which says it isn't for Intelligent. 
I'll see what happens next time I plug in.

mpartington, 2024-11-15:
I don't think it is needed, but doesn't do any harm

springfall2008, 2024-11-16:
Yeh car_charging_now is probably the issue, best avoided as it tells predbat that if the car is charging its cheap rate

chris-y, 2024-11-17:
Thanks, I'm sure that will fix it. 

I have a follow-up question. When the car is scheduled to charge on the middle of the regular cheap slot in the night, predbat is suspending charging (house battery already at 100%), and the car charges from battery - even though I have allow car charge from battery switched off. 

Car was scheduled between 4.30 and 5.00. At 4.30 predbat set the inverter to do a 0A charge (ie, don't discharge) from 5.00 until 5.30. I would expect that behaviour when the car was charging, not for when it had finished?

chris-y, 2024-11-19:
Schedule tonight hasn't exhibited the same, so I'm not sure what was happening yesterday. 
Commenting out car_charging_now has fixed the original issue too.  Thanks!

mpartington, 2024-11-20:
I've a follow up question. My IOG has been terribly temperamental and if not detecting and therefore planning a charge.

I am therefore just setting it to scheduled over night. Last night Predbat clearly didn't know the car was charging and discharged the battery.

There doesn't seem an easy way to switch between IOG controlled charge and manual scheduled charge within Predbat, to ensure the charge is held.

Am I missing something, or is the only way to either comment in or out the car charging now setting? If so, maybe this would be better exposed as a switch in HA

gcoan, 2024-11-20:
> There doesn't seem an easy way to switch between IOG controlled charge and manual scheduled charge within Predbat, to ensure the charge is held.

Have you tried turning switch.predbat_octopus_intelligent_charging to Off to swap between IOG and Predbat-led charging?

https://springfall2008.github.io/batpred/car-charging/

You have to tell Predbat one way or the other when you are charging your car, otherwise (depending on where your CT clamps are fitted to your house wiring) the inverter is likely to see this as house load and the battery discharge into your car

mpartington, 2024-11-20:
This is the issue, I don't really need an automation as the charge times
are just stored in the charger schedule. I'd rather local control, than
firing the commands through the API. I really need to enable car charging
now, but this brings with it the issue of assuming the slot is a cheap rate
during the first few minutes before Octopus set the IOG charge schedule.and
stop the charge.

The only way I can see is to switch car charging now on/off as needed, but
it's a bit clunky via apps.yaml.

Maybe the easiest way is to set up a template sensor that relies on both
the car charging and a selectable switch to feed a response to predbat

chris-y, 2024-11-20:
Related question/problem. Last night my car decided to continue to charge until past the off-peak period, even though it wasn't originally scheduled to do so. I checked the "completed dispatches" in the Octopus integration and these new time periods were present.  However they don't appear to have appeared in planned dispatches, and it looks like predbat wasn't aware of them, as my battery was being drained throughout. 
[image]

It looks like the intelligent dispatching binary sensor was off during the time period too, so I doubt there's anything that can be done about this.

Maybe it's something odd/broken with the Renault integration.

Very strange. I guess I need to check my bill to ensure it was actually charged at the cheaper rate!

mpartington, 2024-11-20:
It's the Renault integration I'm having trouble with as well. The mobilize app seems to keep loosing contact with the octopus app. I normally can't get it to recognise the mobilize app has started a smart charge and the car is plugged in to set the slots. 

I'm trying to find anyone at Octopus who knows enough to help, I think it's a back end issue between the apps

chris-y, 2024-11-20:
That communication seems fine here - it's just that it added some extra slots for some reason and the Octopus API didn't show them (or the Octopus integration didn't pick them up, but I suspect it's the API). I might raise this on the Octopus integration GitHub.

mpartington, 2024-11-20:
Does seem a different issue then. I've been given a slot at 19.30, dispatching sensor has triggered, but mobilize app isn't charging the car.

chris-y, 2024-11-20:
Yeah, it's opposite to what I have. In my case, the Mobilize app is charging the car, it has just added extra slots to the end (I'm not sure why, as it ended up above the target I had set!) - which the integration isn't picking up. Presumably Octopus did know about them or they wouldn't have been in my completed dispatches.

https://github.com/BottlecapDave/HomeAssistant-OctopusEnergy/issues/1088

chris-y, 2024-11-21:
Although actually overnight last night I think I may have had the same problem - it had scheduled a charge but didn't charge.  I changed my leave by time this morning and it immediately scheduled a slot and started charging.

chris-y, 2024-11-22:
I started a smart charge and then disabled smart charging. It's confused the Octopus integration as the slots are still showing and it thinks smart charging is enabled still. I think this is a bug/missing feature with the Renault integration. 

I'm now seeing this in the plan: (I've forced idle on the slots to ensure the battery doesn't kick in) 
[image]

As the integration thinks the car is plugged in (it is), when it gets to these slots will it still think it is charging? Where does it get that information from? As obviously car_charging_now isn't used as per above.

chris-y, 2024-11-22:
Dispatching sensor still shows the slots coming up - I'll find out tomorrow whether it switched itself on during them or not.

Did you put your Octopus account number into Mobilize? Looks like they aren't properly linked.

mpartington, 2024-11-22:
Yeah, it's all individually set up ok. Octopus smart tariffs specialist has been going over my data and has promised to get back to me tomorrow. I doubt it will be an easy fix

This was the other night, when it decided to work
[image]

chris-y, 2024-11-22:
Weird. What I did notice when I set mine up is that it only works on "phase 2" Zoes from October 2020.  I don't know if that's relevant, because, mine's an earlier one and it seems fine!

Maybe it's related to the BMS fix, which mine has had, and they are covering themselves as any with the old software won't work?

mpartington, 2024-11-23:
Mine was registered Nov 20, but not sure of the build date. It also has the BMS update.

I think I've found that I need to re-enter the octopus account every time I use it. That seems to force a sync of the 2 apps, although it didn't stop at the end of a slot earlier.

Octopus has taken my data and I'm being told the developers are analysing it, so hopefully improvements will be made that benefit everyone.

gcoan, 2024-12-13:
> Yeh car_charging_now is probably the issue, best avoided as it tells predbat that if the car is charging its cheap rate

Added extra guidance to the documentation about some cars start charging when plugged in and so you should not set car_charging_now unless you have problems with intelligent slot detection.

closed 2024-12-13 by gcoan