springfall2008/batpred #1632: Overnight charging not going close to 100% on SolarEdge from V8.5.3 onwards.

PieBeard, 2024-11-16:
**Describe the bug**
Since updating beyond V8.5.3 my SolarEdge Battery hasn’t charged to 100% even though the plans state that it should reach this as that is the set target.  Charging power is maintained at the battery’s capacity of 5kW.  An interim work around at the moment has been to manually set forced charge through predbat for the two 30min slots before 05:30 (OIG) or if a dispatch is scheduled for later i.e. 06:00 then that hour before that. All other predbat settings have remained the same throughout.

**Expected behaviour**
Charging to 100% as happened on 8.5.3 and before to be automatic.

**Predbat version**

8.5.3 through to 8.6.0

**Environment details**

SolarEdge 6kW Inverter with SolarEdge 10kWh Battery and a Zappi charger
- Standard HAOS installed and running Standalone PredBat Addon
- Anything else?

**Screenshots**

[image]
Shows the what happens after upgrading form 8.5.3

[image]
Example of after upgrade above 8.5.3 in more detail

[image]
Downgraded to find where it stopped working properly, this is plan for v8.5.3

[image]
And this is the resulting charge, all ok.

[image]
This is the plan for V8.5.11

[image]
And this shows the result the next day.

Here is the plan for the next night 
[image]
[image]
Plan for the Tuesday night

[image]
And resultant charging of the battery, not following plan at all.

**Log file**
This is the log from close to end of charge plan, unfortunately it was copied into google docs on iPad so isn’t the best layout.
[Predbat log.docx](https://github.com/user-attachments/files/17787936/Predbat.log.docx)

springfall2008, 2024-11-17:
```log
2024-11-13 04:55:00.262090: --------------- PredBat - update at 2024-11-13 04:55:00+00:00 with clock skew 0 minutes, minutes now 295
2024-11-13 04:55:00.262416: Saved current settings to /addon_configs/6adb4f0d_predbat/predbat_config.json
2024-11-13 04:55:00.271502: Using cached GITHub data for https://api.github.com/repos/springfall2008/batpred/releases age 120.0 minutes
2024-11-13 04:55:00.271560: Predbat /config/predbat.py version v8.5.11 currently running, latest version is v8.6.0 latest beta v8.6.0
2024-11-13 04:55:00.271587: Autoupdate: There is an update pending v8.6.0 Initial GivTCP v3.0 REST support, bug fixes - auto update is off
2024-11-13 04:55:00.288529: Configuration: forecast_hours 48 num_cars 1 debug enable is False calculate_plan_every 10
2024-11-13 04:55:00.288757: Cars 1 charging from battery False planned [True], charging_now [False] smart [False], max_price [0.0], plan_time ['05:30:00'], battery size [87.0], limit [87.0], rate [7.4], exclusive [False]
2024-11-13 04:55:00.288797: Predbat mode is set to Control charge & discharge
2024-11-13 04:55:04.392589: Found 11521 load_today datapoints going back 8 days
2024-11-13 04:55:07.162404: Current data so far today: load 44.13 kWh import 45.7 kWh export 0.01 kWh pv 0.0 kWh
2024-11-13 04:55:07.167462: Fetching carbon intensity data from sensor.carbon_intensity_uk
2024-11-13 04:55:07.235429: Warn: octopus_charge_limit is set to a bad value unavailable in apps.yaml, must be a number
2024-11-13 04:55:07.236428: Car 0 using Octopus Intelligent, charging planned - charging limit 87.0, ready time unavailable - battery size 87.0
2024-11-13 04:55:07.236468: Cars: SoC kWh: [0.0] Charge limit [87.0] plan time ['unavailable'] battery size [87.0]
2024-11-13 04:55:07.243130: Standing charge is set to 62.118629999999996 p
2024-11-13 04:55:07.250409: Octopus Intelligent slot at 11-11 00:00:00-11-11 00:30:00 assumed price 7.0 amount 0.9 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-13 04:55:07.250486: Octopus Intelligent slot at 11-11 02:00:00-11-11 02:30:00 assumed price 7.0 amount 3.43 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-13 04:55:07.250552: Octopus Intelligent slot at 11-11 01:30:00-11-11 02:00:00 assumed price 7.0 amount 3.18 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-13 04:55:07.250628: Octopus Intelligent slot at 11-11 01:00:00-11-11 01:30:00 assumed price 7.0 amount 0.11999999999999998 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-13 04:55:07.250721: Octopus Intelligent slot at 11-11 03:00:00-11-11 03:30:00 assumed price 7.0 amount 3.54 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-13 04:55:07.250784: Octopus Intelligent slot at 11-11 02:30:00-11-11 03:00:00 assumed price 7.0 amount 3.6 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-13 04:55:07.250844: Octopus Intelligent slot at 11-11 04:00:00-11-11 04:30:00 assumed price 7.0 amount 3.5 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-13 04:55:07.250903: Octopus Intelligent slot at 11-11 03:30:00-11-11 04:00:00 assumed price 7.0 amount 3.44 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-13 04:55:07.250963: Octopus Intelligent slot at 11-11 05:00:00-11-11 05:30:00 assumed price 7.0 amount 3.35 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-13 04:55:07.251022: Octopus Intelligent slot at 11-11 04:30:00-11-11 05:00:00 assumed price 7.0 amount 3.46 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-13 04:55:07.251082: Octopus Intelligent slot at 11-12 23:00:00-11-12 23:30:00 assumed price 7.0 amount 3.46 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-13 04:55:07.251143: Octopus Intelligent slot at 11-13 00:00:00-11-13 00:30:00 assumed price 7.0 amount 3.42 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-13 04:55:07.251206: Octopus Intelligent slot at 11-12 23:30:00-11-13 00:00:00 assumed price 7.0 amount 0.1 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-13 04:55:07.251267: Octopus Intelligent slot at 11-13 01:00:00-11-13 01:30:00 assumed price 7.0 amount 3.49 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-13 04:55:07.251326: Octopus Intelligent slot at 11-13 00:30:00-11-13 01:00:00 assumed price 7.0 amount 3.12 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-13 04:55:07.251386: Octopus Intelligent slot at 11-13 02:00:00-11-13 02:30:00 assumed price 7.0 amount 3.5 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-13 04:55:07.251446: Octopus Intelligent slot at 11-13 01:30:00-11-13 02:00:00 assumed price 7.0 amount 3.54 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-13 04:55:07.251503: Octopus Intelligent slot at 11-13 03:00:00-11-13 03:30:00 assumed price 7.0 amount 3.54 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-13 04:55:07.251562: Octopus Intelligent slot at 11-13 02:30:00-11-13 03:00:00 assumed price 7.0 amount 3.51 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-13 04:55:07.251621: Octopus Intelligent slot at 11-13 04:00:00-11-13 04:30:00 assumed price 7.0 amount 3.3399999999999994 kWh location AT_HOME source None octopus_slot_low_rate True
[... cut ...]
2024-11-13 04:55:19.341518: Inverter 0 adjust target soc for charge to 83% (8.05kWh/9.7kWh 9.7kWh) based on going from 20% -> 83% total add is 6.09kWh and this battery needs to add 6.09kWh to get to 8.05kWh
2024-11-13 04:55:19.341545: Inverter 0 Current Target SOC is 83%, already at target
2024-11-13 04:55:19.341577: Current SOC 20% is less than Target SOC 83. Grid charging enabled with charge current set to 65.00
2024-11-13 04:55:19.399903: Completed run status Charging
2024-11-13 04:55:19.408809: Info: record_status Charging target 20%-83%

```

springfall2008, 2024-11-17:
Can you share your apps.yaml, specifically the service API settings?

PieBeard, 2024-11-17:
# ------------------------------------------------------------------
# This is an example configuration, please modify it
# ------------------------------------------------------------------
---
pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # XXX: This is a configuration template, delete this line once you edit your configuration
  #template: True

  # Timezone to work in
  timezone: Europe/London

  # If you are using Predbat outside of HA then set the HA URL and Key (long lived access token here)
  #ha_url: 'http://homeassistant.local:8123'
  #ha_key: 'xxx'

  # Currency, symbol for main currency second symbol for 1/100s e.g. $ c or £ p or e c
  currency_symbols:
   - '£'
   - 'p'

  # Number of threads to use in plan calculation
  # Can be auto for automatic, 0 for off or values 1-N for a fixed number
  threads: auto

  #
  # Sensors, currently more than one can be specified and they will be summed up automatically
  # however if you have two inverters only set one of them as they will both read the same.
  #
  load_today:
    - sensor.solar_house_consumption_daily
  import_today:
    - sensor.solaredge_multi_m1_imported
  export_today:
    - sensor.solaredge_multi_m1_exported
  pv_today:
    - sensor.solar_panel_production_kwh   # This is a custom sensor - see Readme on how to create or comment out
  #
  # Controls/status - must by 1 per inverter
  #
  num_inverters: 1
  inverter_type: "SE"

  # Can be enabled if you have a backup interface
  #inverter:
  #  has_reserve_soc: True

  battery_rate_max:
    - sensor.solaredge_multi_b1_max_charge_power
  charge_rate:
    - sensor.solaredge_multi_b1_max_charge_power
  discharge_rate:
    - sensor.solaredge_multi_b1_max_discharge_power
  battery_power:
    - sensor.solaredge_multi_b1_dc_power
  pv_power:
    - sensor.solar_panel_production_w # This is a custom sensor - see Readme on how to create or comment out
  load_power:
    - sensor.solar_house_consumption_w # This is a custom sensor - see Readme on how to create or comment out
  soc_percent:
    - sensor.solaredge_multi_b1_state_of_energy
  soc_max:
    - sensor.solaredge_multi_b1_maximum_energy
  # Can be enabled you have a backup interface
  #reserve:
[... 251 lines cut ...]
  # Battery scaling makes the battery smaller (e.g. 0.9) or bigger than its reported
  # If you have an 80% DoD battery that falsely reports it's kwh then set it to 0.8 to report the real figures
  battery_scaling: 1.0

  # Can be used to scale import and export data, used for workarounds
  import_export_scaling: 1.0

  # Export triggers:
  # For each trigger give a name, the minutes of export needed and the energy required in that time
  # Multiple triggers can be set at once so in total you could use too much energy if all run
  # Creates an entity called 'binary_sensor.predbat_export_trigger_<name>' which will be turned On when the condition is valid
  # connect this to your automation to start whatever you want to trigger
  export_triggers:
     - name: 'large'
       minutes: 60
       energy: 1.0
     - name: 'small'
       minutes: 15
       energy: 0.25

  # If you have a sensor that gives the energy consumed by your solar diverter then add it here
  # this will make the predictions more accurate. It should be an incrementing sensor, it can reset at midnight or not
  # It's assumed to be in Kwh but scaling can be applied if need be
  #iboost_energy_today: 'sensor.xxxxx'
  #iboost_energy_scaling: 1.0

PieBeard, 2024-11-26:
Tested again with the new updates and got the following plan but the state of charge did reach 100% earlier in the night but not by 05:30. Seemed like it didn’t call for charge service with enough time for it to reach full charge.
[image]
[image]
[image]

cyberkryten, 2024-11-26:
I'm seeing the same after updating from an old version to 8.7.2

It usually charges to 100% overnight, but last night on the charge/discharge cycle it didn't get above 89%

[image]

Only one night, but thought I'd mention so you know it isn't just your setup

PieBeard, 2024-11-27:
Worse performance yesterday against the plan.

Plan
[image]

Result
[image]

cyberkryten, 2024-11-29:
Battery only at 80% this morning.

I was out so didn't manage to collect much info, but I did notice that late last night the Plan said 'Charging' but the Inverter was left in the state "Maximise Self Consumption".  Logging from back then was gone by the time I'm up and about this morning.

```
2024-11-29 07:15:28.484276 INFO pred_bat: Inverter 0 SOC: 19.5kW 67% Current charge rate 11400W Current discharge rate 11400W Current power 7757.0W Current voltage 52.0V
2024-11-29 07:15:28.525280 INFO pred_bat: Setting charging SOC to 92 as per target
2024-11-29 07:20:05.484197 INFO pred_bat: Inverter 0 SOC: 19.79kW 68% Current charge rate 11400W Current discharge rate 11400W Current power 7749.0W Current voltage 52.0V
2024-11-29 07:20:06.487510 INFO pred_bat: Setting charging SOC to 92 as per target
2024-11-29 07:25:05.409261 INFO pred_bat: Inverter 0 SOC: 20.66kW 71% Current charge rate 11400W Current discharge rate 11400W Current power 7745.0W Current voltage 52.0V
2024-11-29 07:25:06.380073 INFO pred_bat: Setting charging SOC to 92 as per target
2024-11-29 07:25:27.929616 INFO pred_bat: Inverter 0 SOC: 20.66kW 71% Current charge rate 11400W Current discharge rate 11400W Current power 7740.0W Current voltage 52.0V
2024-11-29 07:25:27.969153 INFO pred_bat: Setting charging SOC to 92 as per target
2024-11-29 07:30:05.473436 INFO pred_bat: Inverter 0 SOC: 20.66kW 71% Current charge rate 11400W Current discharge rate 11400W Current power -960.0W Current voltage 52.0V
2024-11-29 07:30:06.451863 INFO pred_bat: Setting charging SOC to 92 as per target
2024-11-29 07:35:05.139966 INFO pred_bat: Inverter 0 SOC: 20.66kW 71% Current charge rate 11400W Current discharge rate 11400W Current power -991.0W Current voltage 52.0V
2024-11-29 07:35:06.147020 INFO pred_bat: Setting charging SOC to 92 as per target
2024-11-29 07:35:28.196289 INFO pred_bat: Inverter 0 SOC: 20.66kW 71% Current charge rate 11400W Current discharge rate 11400W Current power -1046.0W Current voltage 52.0V
2024-11-29 07:35:29.544114 INFO pred_bat: Setting charging SOC to 88 as per target
2024-11-29 07:40:05.187405 INFO pred_bat: Inverter 0 SOC: 21.24kW 73% Current charge rate 11400W Current discharge rate 11400W Current power 7741.0W Current voltage 52.0V
2024-11-29 07:40:06.176498 INFO pred_bat: Setting charging SOC to 88 as per target
2024-11-29 07:45:05.420997 INFO pred_bat: Inverter 0 SOC: 21.82kW 75% Current charge rate 11400W Current discharge rate 11400W Current power 7734.0W Current voltage 52.0V
2024-11-29 07:45:06.404112 INFO pred_bat: Setting charging SOC to 88 as per target
2024-11-29 07:45:28.883011 INFO pred_bat: Inverter 0 SOC: 21.82kW 75% Current charge rate 11400W Current discharge rate 11400W Current power 7736.0W Current voltage 52.0V
2024-11-29 07:45:28.922292 INFO pred_bat: Setting charging SOC to 88 as per target
2024-11-29 07:50:05.415340 INFO pred_bat: Inverter 0 SOC: 22.41kW 77% Current charge rate 11400W Current discharge rate 11400W Current power 7740.0W Current voltage 52.0V
2024-11-29 07:50:06.407006 INFO pred_bat: Setting charging SOC to 88 as per target
2024-11-29 07:55:05.439420 INFO pred_bat: Inverter 0 SOC: 22.99kW 79% Current charge rate 11400W Current discharge rate 11400W Current power 7794.0W Current voltage 52.0V
2024-11-29 07:55:06.415283 INFO pred_bat: Setting charging SOC to 88 as per target
2024-11-29 07:55:27.562413 INFO pred_bat: Inverter 0 SOC: 22.99kW 79% Current charge rate 11400W Current discharge rate 11400W Current power 7796.0W Current voltage 52.0V
2024-11-29 07:55:27.600294 INFO pred_bat: Setting charging SOC to 88 as per target
2024-11-29 08:00:05.563520 INFO pred_bat: Inverter 0 SOC: 23.57kW 81% Current charge rate 11400W Current discharge rate 11400W Current power 7820.0W Current voltage 52.0V
2024-11-29 08:00:27.848024 INFO pred_bat: Setting charging SOC to 0 as we are not within the charge window and inverter doesn't support target soc
2024-11-29 08:05:05.145799 INFO pred_bat: Inverter 0 SOC: 23.57kW 81% Current charge rate 11400W Current discharge rate 11400W Current power -654.0W Current voltage 52.0V
2024-11-29 08:05:10.186735 INFO pred_bat: Setting charging SOC to 0 as we are not within the charge window and inverter doesn't support target soc
2024-11-29 08:05:35.363334 INFO pred_bat: Inverter 0 SOC: 23.57kW 81% Current charge rate 11400W Current discharge rate 11400W Current power -896.0W Current voltage 52.0V
2024-11-29 08:05:35.403023 INFO pred_bat: Setting charging SOC to 0 as we are not within the charge window and inverter doesn't support target soc
```

Limits are set in config:
```
  # Inverter max AC limit (one per inverter)                   
  # If you have a second inverter for PV only please add the two values together
  inverter_limit:                                                               
    - 8000                                                                      
  # Export limit is a software limit set on your inverter that prevents exporting above a given level
  # When enabled Predbat will model this limit                                                       
  export_limit:                                                                                      
    - 3680

  battery_rate_max:                                                      
    - number.solaredgemodbus_i1_storage_charge_limit                  
  charge_rate:                                                        
[... cut ...]
  discharge_rate:                                                                                 
    - number.solaredgemodbus_i1_storage_discharge_limit   
```

I wonder if there's been a change in what is used to calculate the charging speed?  My inverter reports the storage_charge/discharge_limit values being 11400 (always has done) - just a quirk of the SE Inverters.

However, it is an 8kW inverter, that could account for the fact it is no longer reaching 100% (maybe).

If there's anywhere that extra Debug output would be useful, let me know and I'll add it and report back.

closed 2026-08-28 by chalfontchubby