springfall2008/batpred #1635: sensors for today and tomorrow predicted items (kwh and cost)

eddysteurs, 2024-11-17:
**Is your feature request related to a problem? Please describe.**
with current sensors not possible to build a dashboard with today & tomorrow figures, only actual follow actual day hours

**Describe the solution you'd like**
would it be possible to have output sensors with the predicted kwh and cost items for today and tomorrow but on actual day (0 - 24h) (now it are moving hours depending on forecast span , for me 48 hrs)

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

springfall2008, 2024-11-17:
How is the chart you want different from this one?

[image]

gcoan, 2024-11-17:
@eddysteurs 

Have a look at the attributes of predbat.best_metric which is Predbat's future cost predictions

The attribute 'results' gives the future cost predictions from now to the end of the predicted plan so if you wanted to get tomorrow's predicted cost you could create a template sensor that calculates results['today+2'00:00] - predbat.best_metric

the today+2 is to get the value of the predicted cost at midnight tomorrow night

Bit of fiddling with date manipulation but all the data is there for you to extract and use as you wish

eddysteurs, 2024-11-17:
I will have a try at this tomorrow, thanks for the tip
Eddy Steurs

eddysteurs, 2024-11-18:
Gcoin

thanks a lot , got it working, 1 example on import from grid for tomorrow
        predbat_import_fcst_tomorrow:
          friendly_name: "predbat import forecast tomorrow"
          unit_of_measurement: 'kWh'
          value_template: >
                {% set data = state_attr('predbat.import_energy', 'results') %}
                {% set target_time = (now() + timedelta(days=2)).date() ~ 'T00:00:00+0100' %}
                {% set target_today = (now() + timedelta(days=1)).date() ~ 'T00:00:00+0100' %}
                {% set value =( data[target_time] - data[target_today] ) | round(2) %}
                 {{value}}

once you have 1 the rest is peace of cake 👍

closed 2024-11-18 by eddysteurs

gcoan, 2024-11-18:
Neat job @eddysteurs I'll add it to the documentation as an example for others

If you add a state_class of total then you'll collect long term statistics on this template so can  produce history graphs etc

eddysteurs, 2024-11-18:
some change as it seems 00:00 is not always present, so search for next
        predbat_import_fcst_today:
          friendly_name: "predbat import forecast today"
          unit_of_measurement: 'kWh'
          value_template: >
                {% set data = state_attr('predbat.import_energy', 'results') %}
                {% set target_today = (now() + timedelta(days=1)).date() ~ 'T00:05:00+0100' %}
                  {% set keys = data.keys() | list | sort %}
                  {% set next_item = keys | select('>', target_today) | list | first %}
                  {% set value = data[next_item] if next_item else '9999' %}
                  {{value}}

        predbat_import_fcst_tomorrow:
          friendly_name: "predbat import forecast tomorrow"
          unit_of_measurement: 'kWh'
          value_template: >
                {% set data = state_attr('predbat.import_energy', 'results') %}
                {% set target_time = (now() + timedelta(days=2)).date() ~ 'T00:00:00+0100' %}
                  {% set keys = data.keys() | list | sort %}
                  {% set next_item = keys | select('>', target_time) | list | first %}
                  {% set valuetom = data[next_item] if next_item else '9999' %}
                {% set value =( valuetom|float(0) - states('sensor.predbat_import_fcst_today')|float(0) ) | round(2) %}
                {{value}}