springfall2008/batpred #1638: HA error - sensor.predbat_sx4_0_charge_limit no longer has a state class

jefft4, 2024-11-20:
**Describe the bug**
Since several versions back, HA reports an error "sensor.predbat_sx4_0_charge_limit no longer has a state class". Deleting the history does not resolve the error, as the sensor is in use and creates further history.
Sensor entity has no ID, so is not manageable through the HA UI to change it.

**Expected behaviour**
No error, presumably by PB setting the state class attribute of the sensor.

8.6.1 (problem since 8.5.8, possibly earlier)

**Environment details**

- Inverter and battery setup
Solax G4 + battery
- Standard HAOS installer or Docker
Docker

**Screenshots**
[image]

chris-y, 2024-11-23:
I'm getting the same here for a number of entities, and also messages telling me the units have changed from '' to 'W' or back the other way.

These are currently showing as having no state class, although I've seen others for my custom inverter setup:
sensor.predbat_mine_0_charge_limit
sensor.predbat_mine_0_discharge_rate

cyberkryten, 2024-11-26:
Was on a much older version and I'm now also seeing this since I upgraded to 8.7.2 (SolarEdge Inverter and Batteries):

```
The entity no longer has a state class
We have generated statistics for 'predbat se 0 charge limit' (sensor.predbat_se_0_charge_limit) in the past, but it no longer has a state class, therefore, we cannot track long term statistics for it anymore.

Statistics cannot be generated until this entity has a supported state class.
If the state class was previously provided by an integration, this might be a bug. Please report an issue.
If you previously set the state class yourself, please correct it. [The different state classes and when to use which can be found in the developer documentation.](https://developers.home-assistant.io/docs/core/entity/sensor/#long-term-statistics)
If the state class has permanently been removed, you may want to delete the long term statistics of it from your database.
Do you want to permanently delete the long term statistics of sensor.predbat_se_0_charge_limit from your database?
```

stiplady, 2024-12-06:
Another vote here for predbat_mine_0_charge_limit showing this warning. @springfall2008 do you have any idea what is causing this, is it likely to be config issue in our files, or an issue with predbat?

closed 2026-08-28 by chalfontchubby