springfall2008/batpred #1640: KH8 Fox ess inverter settings

MathewBurnett, 2024-11-20:
I'm trying to setup predbat for HA to talk to my KH8 inverter. Modbus in HA reports the following versions.

Version: Manager (FoxESS)1.22
Version: Master (FoxESS)1.30
Version: Slave (FoxESS)1.01

I am struggling to find the correct settings for certain items. At the moment i have the following

`charge_rate:
    - number.foxess_force_charge_power
  discharge_rate:
    - number.foxess_force_discharge_power
  battery_power:
    - sensor.foxess_invbatpower
  pv_power:
    - sensor.pv_power_foxess
  load_power:
    - sensor.foxess_load_power
  #soc_kw:
  #  - sensor.inverter_bms_kwh_remaining
  soc_percent:
    - sensor.foxess_battery_soc`

However the force charge/discharge writable registers seem to be in KW which doesn't match what predbat expects and the other options is a RO value for bms charge rate which is in Amps. Am i miss understanding? Does anyone have a working KH model config they can share with me?

HeavyPXL, 2024-11-21:
I don't have a KH8 but on my FoxESS H1 I have the charge_rate and discharge_rate settings commented out. The inverter just charges or discharges at max rate (5kW in my case) when PredBat requests it.

I have not noticed any particular problems with that configuration.

MathewBurnett, 2024-11-24:
> I don't have a KH8 but on my FoxESS H1 I have the charge_rate and discharge_rate settings commented out. The inverter just charges or discharges at max rate (5kW in my case) when PredBat requests it.
> 
> 
> 
> I have not noticed any particular problems with that configuration.

Thanks I comment these out and appear to be error free.

closed 2026-08-28 by chalfontchubby