springfall2008/batpred #1643: (Automatic?) Conversion between units (watts and kilowatts, but possibly others?)

davidarkemp, 2024-11-22:
**Is your feature request related to a problem? Please describe.**
Different parts of predbat require input at different scales (pv_power is in watts, for example)

**Describe the solution you'd like**
Where possible, I'd like predbat to use the unit of the sensor to determine how to scale the output. 
As a sidle effect of this, predbat could also issue warnings when the input isn't an expected unit.

**Describe alternatives you've considered**
At the moment, I'm using a template sensor to convert the value before input to predbat, but this requires extra storage in the db history.

**Additional context**
As a new user, understanding this all has been the major hurdle to getting something setup. At some point, it would be great if there was a UI for all of this that only allowed likely inputs, like the HA Energy Dashboard does.

springfall2008, 2024-11-24:
In some cases Predbat already converts between watts and kw, which case are you refering to?

davidarkemp, 2024-11-24:
Specifically: `pv_power` 
Until I used a template sensor that does the conversion, `predbat.pv_power` seemed to be incorrect

WyndStryke, 2025-01-18:
I had this too, turned out that it was just a display issue rather than something which affected the plan.  The apex chart and the config page use the unconverted (original) figure, whereas it actually converts when storing into history, so things should work OK regardless.   Personally I made a sensor template to multiply the kW figure by 1000 to convert it to Watts, so that it would show up in the charts etc OK.

https://github.com/springfall2008/batpred/issues/1870#issuecomment-2586420207

gcoan, 2025-01-18:
Trefor has confirmed that Predbat will convert the input sensors OK so it looks like this is just a display issue on the charts.  I will add something to the documentation to explain how to convert the units in the charts

WyndStryke, 2025-06-22:
@gcoan Can this one be closed by #2463 ?  (From my viewpoint I think it is fully covered).

2463 allows both reads and writes to convert between kW and W.  This should also simplify a lot of the inverter templates.