springfall2008/batpred #1644: Preference for solar over free electricity

onetoomany, 2024-11-22:
I've just had a powerup (22/11/2024 0730-0930).  However, prebat didn't fully charge the battery as it thought I'd get more solar during the course of the day.

Solar is uncertain (especially at this time of year), so I'd have thought it should prioritise guaranteed free electricity now over potentially free electricity later (I don't get paid for my solar, so my export rate is set at 0p).

Also, iBoost seems to be kicking in, even though my battery is not going to hit 100%.
[image]

Log file:
[predbat.log](https://github.com/user-attachments/files/17868381/predbat.log)

springfall2008, 2024-11-23:
Was the power up configured as zero rate? Did Predbat charge during the power up using the full time or not?

What are your IBoost settings?

onetoomany, 2024-11-27:
Powerup manually configured as zero.
Predbat did charge but not to 100%

iBoost settings:
- binary_sensor.iboost_on	off
- binary_sensor.predbat_iboost_active	FALSE
- input_number.predbat_iboost_gas_scale	1
- input_number.predbat_iboost_max_energy	8
- input_number.predbat_iboost_max_power	2500
- input_number.predbat_iboost_min_power	500
- input_number.predbat_iboost_min_soc	75
- input_number.predbat_iboost_rate_threshold	100
- input_number.predbat_iboost_rate_threshold_export	100
- input_number.predbat_iboost_today	0
- input_number.predbat_iboost_value_scaling	0.75
- predbat.iboost_best	0
- switch.predbat_iboost_charging	off
- switch.predbat_iboost_enable	on
- switch.predbat_iboost_energy_subtract	on
- switch.predbat_iboost_gas	off
- switch.predbat_iboost_gas_export	off
- switch.predbat_iboost_on_export	off
- switch.predbat_iboost_prevent_discharge	off
- switch.predbat_iboost_smart	off
- switch.predbat_iboost_solar	on

closed 2025-01-01 by springfall2008