springfall2008/batpred #1645: Export rates not getting to

howlintj1, 2024-11-22:
**Describe the bug**
A clear and concise description of what the bug is.

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Predbat version**

xxxx

**Environment details**

- Inverter and battery setup
- Standard HAOS installer or Docker
- Anything else?

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.
If you are not keeping the full logs then please enable this in appdaemon.yaml (see the installation instruction in the Predbat docs area for details on how to do this)

gcoan, 2024-11-22:
What is your export sensor from the octopus integration saying ?

sensor.octopus_energy_electricity_XXXX_export_current_rate

This is almost certainly not a Predbat bug, nor an Octopus Integration bug,  I'm willing to bet that your Octopus account has only changed the import rate and Octopus have forgotten to change the export rate.

Same thing happened to me when I moved from Flux to Agile.  And then about 6 months later Octopus spotted the error and rebilled me for the entire 6 months

howlintj1, 2024-11-23:
You are correct!

It’s at the old rates. I’ll contact octopus to sort out.

SwiftRR, 2024-11-23:
If you contact Octopus using DM on X, you may find that the tariff is changed really quickly.

Rob

howlintj1, 2024-11-23:
Thanks Contacted them on x. I've just set up a manual export rate for now
and it looks to be working perfectly.

Thanks for everyones help. I'm not sure I'd of done it otherwise!

gcoan, 2024-11-23:
> You are correct!
> 
> It’s at the old rates. I’ll contact octopus to sort out.

I thought it would be the case.  When you change import tariff they don't seem to remember the linked export rate

Until Octopus fix your account you can put a rates_export_override in apps.yaml with the correct export rate

closed 2024-11-23 by springfall2008