springfall2008/batpred #1648: IOG - Use full 30min slot f car is unplugged part way through a slot

mpartington, 2024-11-23:
**Describe the bug**
Not sure if this is a bug or feature request.  My understanding of IOG is if the car partially charges in a 30 min slot, then the whole of that 30 mins is billed off peak.

What I am finding is, soon as I unplug the car, Predbat reverts the slot reverts back to peak rate and therefore doesn't take advantage of this.  What seems to be happening is the octopus dispatch sensor switches off and Predbat reacts immediately to change the rate.

**Expected behaviour**
Assuming my understanding is correct... Predbat should lock in the remainder of the 30 min slot at the off peak rate and continue running off grid, charge batteries etc

v8.7.1

Example below shows 

**Screenshots**

[image]

[image]

[image]

**Log file**

[predbat.log](https://github.com/user-attachments/files/17880314/predbat.log)

springfall2008, 2024-11-23:
Have you tried turning off 'octopus_intelligent_ignore_unplugged'?

mpartington, 2024-11-23:
Double checked and that's never been switched on

springfall2008, 2024-11-23:
Actually that switch doesn't control the peak/off peak rate anyhow. More the question is what the Octopus Integration does, if it removes the slots from the dispatch sensor then it will go away in Predbat too.

mpartington, 2024-11-23:
It does appear to remove the slots from the dispatch sensor. I can't see one for that time.

I'm sure its been said you do get the whole slot, maybe others can confirm. I'll see if I can find an example, as I've one of these billed already.

If so, maybe more one for bottlecap dave?

mpartington, 2024-11-23:
Here is an example from my first ever slot that seems to confirm I won't be getting an unwelcome bill : -)

[image]

[image]

springfall2008, 2024-11-23:
The real question is how Predbat would know, if Bottlecap Dave removes the slot from the sensor as Octopus removes it? Keeping the slots as sticky won't work as the plan can change at Octopus end too.

closed 2024-11-23 by mpartington