springfall2008/batpred #1654: Sudden loss of data and control with GivEnergy hybrid gen 3

steve-daedilus, 2024-11-24:
**Describe the bug**
As of early hours of this morning, Predbat is no longer seeing data from, nor able to control, my GivEnergy system. Data values are missing, and errors throughout the stack in Home assistant. No configurations made, though the HA VM was migrated earlier that evening to new storage location. Additionally, the battery charging controls on the inverter were set to charge, and even manual override failed to do more than trickle charge from solar panels.

Attempted downgrade of minor HA core patch that had auto rolled out a few days ago. No change.

Multiple restarts of all aspects of the stack collectively did nothing.

Restarting inverter did not help. Did a full inverter and battery power down and startup. Then reset defaults for settings and then was able to manually force a charge on the inverter.

Still seeing no data for Predbat in HA.

**Expected behaviour**
Data populating HA and Predbat managing as has been previously.

**Predbat version**

8.7.2

**Environment details**

- GE Hybrid 5kW gen 3, 2x 5.2kw in series. Ethernet connected.
- Standard HAOS installer
- HA deployed in proxmox VM.
- HA all adds ins and services for control set to auto update.

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**
Logfile (Warnings) - All Warnings Errors
14334	2024-11-24 11:45:07.472754: Error: 'NoneType' object has no attribute 'get'
14333	2024-11-24 11:45:07.472376: Info: record_status Error: Exception raised 'NoneType' object has no attribute 'get'
14331	AttributeError: 'NoneType' object has no attribute 'get'
14322	2024-11-24 11:45:07.458440: Error: Traceback (most recent call last):
14321	2024-11-24 11:45:07.458329: Error: Exception raised 'NoneType' object has no attribute 'get'
14319	2024-11-24 11:45:07.458166: Warn: Inverter control auto restart trigger: REST read failure
14317	2024-11-24 11:45:07.442836: Warn: Inverter 0 unable to read REST data from http://homeassistant.local:6345/runAll - REST will be disabled
14310	2024-11-24 11:45:01.294081: Warn: Historical day 5 has 25 minutes of gap in the data, filled from 25.77 kWh to make new average 26.23 kWh (percent 98%)
14309	2024-11-24 11:45:01.293486: Warn: Historical day 2 has 10 minutes of gap in the data, filled from 26.1 kWh to make new average 26.28 kWh (percent 99%)
14276	2024-11-24 11:40:07.076632: Error: 'NoneType' object has no attribute 'get'
14275	2024-11-24 11:40:07.076446: Info: record_status Error: Exception raised 'NoneType' object has no attribute 'get'
14273	AttributeError: 'NoneType' object has no attribute 'get'
14264	2024-11-24 11:40:07.064335: Error: Traceback (most recent call last):
14263	2024-11-24 11:40:07.064011: Error: Exception raised 'NoneType' object has no attribute 'get'
14261	2024-11-24 11:40:07.063780: Warn: Inverter control auto restart trigger: REST read failure
14259	2024-11-24 11:40:07.050951: Warn: Inverter 0 unable to read REST data from http://homeassistant.local:6345/runAll - REST will be disabled
14252	2024-11-24 11:40:00.896744: Warn: Historical day 5 has 25 minutes of gap in the data, filled from 25.74 kWh to make new average 26.19 kWh (percent 98%)
14251	2024-11-24 11:40:00.895575: Warn: Historical day 2 has 10 minutes of gap in the data, filled from 26.15 kWh to make new average 26.33 kWh (percent 99%)
14218	2024-11-24 11:35:07.673440: Error: 'NoneType' object has no attribute 'get'
14217	2024-11-24 11:35:07.673125: Info: record_status Error: Exception raised 'NoneType' object has no attribute 'get'
14215	AttributeError: 'NoneType' object has no attribute 'get'
14206	2024-11-24 11:35:07.660391: Error: Traceback (most recent call last):
14205	2024-11-24 11:35:07.660161: Error: Exception raised 'NoneType' object has no attribute 'get'
14203	2024-11-24 11:35:07.659801: Warn: Inverter control auto restart trigger: REST read failure
14201	2024-11-24 11:35:07.647189: Warn: Inverter 0 unable to read REST data from http://homeassistant.local:6345/runAll - REST will be disabled
14194	2024-11-24 11:35:01.498765: Warn: Historical day 5 has 25 minutes of gap in the data, filled from 25.75 kWh to make new average 26.2 kWh (percent 98%)
14193	2024-11-24 11:35:01.497625: Warn: Historical day 2 has 10 minutes of gap in the data, filled from 26.2 kWh to make new average 26.39 kWh (percent 99%)
14160	2024-11-24 11:30:07.292070: Error: 'NoneType' object has no attribute 'get'
14159	2024-11-24 11:30:07.291715: Info: record_status Error: Exception raised 'NoneType' object has no attribute 'get'
14157	AttributeError: 'NoneType' object has no attribute 'get'
14148	2024-11-24 11:30:07.279264: Error: Traceback (most recent call last):
14147	2024-11-24 11:30:07.279002: Error: Exception raised 'NoneType' object has no attribute 'get'
14145	2024-11-24 11:30:07.278607: Warn: Inverter control auto restart trigger: REST read failure
14143	2024-11-24 11:30:07.265703: Warn: Inverter 0 unable to read REST data from http://homeassistant.local:6345/runAll - REST will be disabled
14136	2024-11-24 11:30:01.115958: Warn: Historical day 5 has 25 minutes of gap in the data, filled from 25.73 kWh to make new average 26.19 kWh (percent 98%)
14135	2024-11-24 11:30:01.115335: Warn: Historical day 2 has 10 minutes of gap in the data, filled from 26.19 kWh to make new average 26.37 kWh (percent 99%)
14102	2024-11-24 11:25:07.872743: Error: 'NoneType' object has no attribute 'get'
14101	2024-11-24 11:25:07.872693: Info: record_status Error: Exception raised 'NoneType' object has no attribute 'get'
14099	AttributeError: 'NoneType' object has no attribute 'get'
14090	2024-11-24 11:25:07.857658: Error: Traceback (most recent call last):
[... 119 lines cut ...]
13367	2024-11-24 10:33:12.763307: Warn: Web Socket closed, will try to reconnect in 5 seconds
13365	aiohttp.client_exceptions.W SServerHandshakeError: 502, message='Invalid response status', url=URL('http://supervisor/core/api/websocket')
13364	raise WSServerHandshake Error(
13358	2024-11-24 10:33:12.762822: Error: Traceback (most recent call last):
13357	2024-11-24 10:33:12.762304: Error: Web Socket exception in startup: 502, message='Invalid response status', url=URL('http://supervisor/core/api/websocket')
13355	2024-11-24 10:33:07.754114: Warn: Web Socket closed, will try to reconnect in 5 seconds
13353	aiohttp.client_exceptions.W SServerHandshakeError: 502, message='Invalid response status', url=URL('http://supervisor/core/api/websocket')
13352	raise WSServerHandshake Error(
13346	2024-11-24 10:33:07.753250: Error: Traceback (most recent call last):
13345	2024-11-24 10:33:07.752785: Error: Web Socket exception in startup: 502, message='Invalid response status', url=URL('http://supervisor/core/api/websocket')
13343	2024-11-24 10:33:02.741267: Warn: Web Socket closed, will try to reconnect in 5 seconds
13342	2024-11-24 10:30:07.597979: Error: 'NoneType' object has no attribute 'get'
13341	2024-11-24 10:30:07.597663: Info: record_status Error: Exception raised 'NoneType' object has no attribute 'get'
13339	AttributeError: 'NoneType' object has no attribute 'get'
13330	2024-11-24 10:30:07.585528: Error: Traceback (most recent call last):
13329	2024-11-24 10:30:07.584978: Error: Exception raised 'NoneType' object has no attribute 'get'
13327	2024-11-24 10:30:07.584934: Warn: Inverter control auto restart trigger: REST read failure
13325	2024-11-24 10:30:07.572664: Warn: Inverter 0 unable to read REST data from http://homeassistant.local:6345/runAll - REST will be disabled
13318	2024-11-24 10:30:01.426732: Warn: Historical day 5 has 25 minutes of gap in the data, filled from 23.15 kWh to make new average 23.56 kWh (percent 98%)
13317	2024-11-24 10:30:01.425750: Warn: Historical day 2 has 10 minutes of gap in the data, filled from 27.65 kWh to make new average 27.84 kWh (percent 99%)
13284	2024-11-24 10:25:24.230128: Error: 'NoneType' object has no attribute 'get'
13283	2024-11-24 10:25:24.229799: Info: record_status Error: Exception raised 'NoneType' object has no attribute 'get'
13281	AttributeError: 'NoneType' object has no attribute 'get'
13272	2024-11-24 10:25:24.217038: Error: Traceback (most recent call last):
13271	2024-11-24 10:25:24.216772

springfall2008, 2024-11-24:
Can you check settings, system, logs, predbat and find the actual stack dump?

gcoan, 2024-11-24:
Looking at the GivTCP log it looks like its a GivTCP or inverter issue, not Predbat

https://github.com/britkat1980/ha-addons/issues/33

steve-daedilus, 2024-11-24:
Thanks both so far. I'll collate what I can if it helps.

steve-daedilus, 2024-11-24:
Version information:

Core 2024.11.2
Supervisor 2024.11.4
Operating System 13.2
Frontend 20241106.2

Logs from Supervisor around when we started seeing issues:

2024-11-24 00:39:54.503 INFO (SyncWorker_2) [supervisor.docker.manager] Stopping addon_a6a2857d_givtcp application
2024-11-24 00:39:55.102 INFO (SyncWorker_2) [supervisor.docker.manager] Cleaning addon_a6a2857d_givtcp application
2024-11-24 00:39:55.348 WARNING (MainThread) [supervisor.docker.addon] Can not set nested option PALM_BATT_UTILISATION as Docker env
2024-11-24 00:39:56.599 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on britkat/giv_tcp-ma with version 2.4.9
2024-11-24 00:39:57.509 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from a6a2857d_givtcp
2024-11-24 00:39:59.492 INFO (SyncWorker_1) [supervisor.docker.manager] Stopping addon_core_mosquitto application
2024-11-24 00:40:05.464 INFO (SyncWorker_1) [supervisor.docker.manager] Cleaning addon_core_mosquitto application
2024-11-24 00:40:07.293 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on homeassistant/amd64-addon-mosquitto with version 6.4.1
2024-11-24 00:40:10.008 ERROR (MainThread) [supervisor.services.modules.mqtt] There is already a MQTT service in use from core_mosquitto
2024-11-24 00:40:52.178 INFO (SyncWorker_4) [supervisor.docker.manager] Stopping addon_6adb4f0d_predbat application
2024-11-24 00:40:52.315 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Cannot proxy websocket message of unsupported type: 257
2024-11-24 00:40:52.315 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API for 6adb4f0d_predbat closed
2024-11-24 00:40:56.498 INFO (SyncWorker_4) [supervisor.docker.manager] Cleaning addon_6adb4f0d_predbat application

These general reports of "PALM_BATT_UTILISATION as Docker env" and "There is already a MQTT service in use from core_mosquitto" then appear periodically.

I've done some reboot / restarts this afternoon in last half hour or so, and there's a lot of aiohttp.server errors, at which point another kick has happened.

Once back up, I've then conducted a ha configuration check (come back ok).  I'll go and retrieve detailed logs

steve-daedilus, 2024-11-24:
As per suggestion elsewhere, I have restored to previous backup state, and after a short while of this (and re-configuring MQTT user settings), this has started working now (GivTCP info and Predbat info populating again).

I'm going to monitor this for the time being, however this has confirmed it is not a predbat issue.

Thanks for the help!

closed 2024-11-24 by steve-daedilus