springfall2008/batpred #1659: Respository structure not compliant for HACS install

scudderfish, 2024-11-24:
**Describe the bug**
I attempted to follow the HACS install guide.  When I attempt to add the repository `https://github.com/springfall2008/batpred` to HACS it gives the error 
```
<Integration springfall2008/batpred> Repository structure for v8.7.2 is not compliant
```

**Expected behaviour**
Able to add the custom repo to HACS

**Predbat version**

Latest

**Environment details**

Docker install of HA version 2024.11.3

nipar4, 2024-11-24:
Predbat is not a HACs repo, f your running ha in a docker environment you cannot add an addon so you'll need to follow the standalone instructions to install a separate docker container

nipar4, 2024-11-24:
Try this link for the documentation
https://github.com/springfall2008/predbat_addon/blob/main/predbat/DOCS.md
@springfall2008 the predbat documentation still seems to refer to appdaemon and misses the docker version of predbat

closed 2024-11-24 by scudderfish