springfall2008/batpred #1661: Entities missing after HA 2024.11.03 update

pgit2345, 2024-11-24:
After upgrading from HA 2024.11.02 to 2024.11.03 most of the predbat entities were missing from the predbat control card. Plan html missing. System not starting e.g. not going into demand or other modes. A few people have been having this issue as described in Giv Forum, links below. 

Had to restart predbat addon and then restart HA itself to sort it. Sometimes twice. 

Then when upgrading to PredBat 8.7.0 and again to 8.7.1 continued to have issues with some or all entities missing. Again had to restart several times (both PredBat and HA) to get the entities back and the system back in demand operation. 

It appears to be a web socket issues connecting to supervisor. Code snippet below on url.

https://community.givenergy.cloud/d/3696-first-night-live-on-predbat/2741

plus also discussed here 

https://community.givenergy.cloud/d/5277-issues-with-home-assistant-connecting-to-ge-hybrid-inverter-gen-3/2

**Expected behaviour**
succesful operation and all entities present after either a HA or Predbat upgrade. 

**Predbat version**
8.7.0 and 8.7.1 

**Environment details**

- Inverter and battery setup Hybrid Gen3, Gen1 5.2 Batt
- Standard HAOS installer or Docker: RPi HAOS 
- Anything else?

**Screenshots**
error snippet on above url, 

**Log file**
error snippet on above url.

pgit2345, 2024-11-24:
[image]
[image]
Couple of screenshots.
Just upgraded to 8.7.2, same issue. Tried a rebuild of PredBat addon, plus copied and pasted the controls card from the add-ons directory again in case anything had changed (I had updated it for all the discharge to export renamed, but did that manually). 
It is now working, probably took 15 minutes for all the entities to appear again in the card. Could see the last few (at the end of the card) arriving one by one just after I saved the card/dashboard. 
Not sure if it's just really slow to start up or if it really does need the restarts.

SwiftRR, 2024-11-30:
I find that it takes some time (5-10 min) for predbat to fully restart following an HA restart.
Then all of a sudden, the 'null' disappears as I get a plan and all the predbat entities become available.

So I would just leave predbat longer is this situation.
Take a look at the predbat log. You should see a smooth restart until you see all the .py files appear.
After that, it takes some time until the first instruction appears. Then predbat should have restarted.

Rob

springfall2008, 2025-01-01:
Did you ever sort this out?

pgit2345, 2025-01-01:
Yeah, just seems to be one of those things where you have to be patient.

closed 2025-01-01 by gcoan