springfall2008/batpred #1663: Predbat shows jumps in actual cost figure despite also showing no increase in consumption

cyberkryten, 2024-11-26:
**Describe the bug**
At various points during the day, the actual cost recorded by Predbat increases significantly.  It has been doing this for ages, and I've been unable to figure out why - only just updated everything to full versions and got back to it, hence raising now.

**Expected behaviour**
At these times, there is no increase in the energy usage on any metric or recorded by Predbat - the actual cost reported just jumps up for no apparent reason.  Debug doesn't show the calculation (that I can see) so I cannot tell what Predbat thinks is costing this large amount extra.

I'm on IOG, so rates do change throughout the day, but that's 26p peak and 7p off-peak, all that changes are some future slots become 7p, so I can't see it is related to that.

**Predbat version**
8.7.2
HAOS, HACS etc all updated to latest versions yesterday

**Environment details**
SolarEdge 8kW Inverter, 24 panels, Three 9.7kWh Solaredge batteries
Appdaemon install

**Screenshots**
Here you can see a plot of the Predbat values showing the actual reported cost above the readings from Smart Meter/Solar showing energy usage.  When the Predbat costs line jumps vertically up, there is no increase in any of the consumption graphs.

[image]

**Log file**

Yesterday afternoon - change from £2.91 to £5.41 despite energy import remaining at 43.32kWh (and exporting!)
```
2024-11-25 15:50:05.079517 INFO pred_bat: Today's energy import 41.59 kWh export 19.36 kWh total 22.24 kWh cost 295.64 p import 487.4 p export -253.99 p carbon 0.0 kg
2024-11-25 15:55:05.100751 INFO pred_bat: Today's energy import 42.35 kWh export 19.36 kWh total 22.99 kWh cost 300.95 p import 492.71 p export -253.99 p carbon 0.0 kg
2024-11-25 16:00:05.170348 INFO pred_bat: Today's energy import 43.09 kWh export 19.36 kWh total 23.74 kWh cost 306.16 p import 497.92 p export -253.99 p carbon 0.0 kg
2024-11-25 16:05:04.824493 INFO pred_bat: Today's energy import 43.32 kWh export 19.55 kWh total 23.77 kWh cost 306.92 p import 503.83 p export -259.13 p carbon 0.0 kg
2024-11-25 16:10:05.145151 INFO pred_bat: Today's energy import 43.32 kWh export 19.84 kWh total 23.48 kWh cost 299.35 p import 503.83 p export -266.7 p carbon 0.0 kg
2024-11-25 16:15:05.021988 INFO pred_bat: Today's energy import 43.32 kWh export 20.14 kWh total 23.18 kWh cost 291.64 p import 503.83 p export -274.41 p carbon 0.0 kg
2024-11-25 16:20:18.989122 INFO pred_bat: Today's energy import 43.32 kWh export 20.42 kWh total 22.9 kWh cost 541.83 p import 761.4 p export -281.8 p carbon 0.0 kg
2024-11-25 16:25:09.537761 INFO pred_bat: Today's energy import 43.32 kWh export 20.69 kWh total 22.63 kWh cost 534.64 p import 761.4 p export -288.99 p carbon 0.0 kg
2024-11-25 16:30:18.501248 INFO pred_bat: Today's energy import 43.32 kWh export 20.98 kWh total 22.34 kWh cost 527.15 p import 761.4 p export -296.48 p carbon 0.0 kg
2024-11-25 16:35:13.350500 INFO pred_bat: Today's energy import 43.32 kWh export 21.27 kWh total 22.05 kWh cost 519.41 p import 761.4 p export -304.21 p carbon 0.0 kg
```

Today:

Change from £1.82 to £3.52 - again energy import value remains constant.
Since midnight, my Smart Meter (read directly using Hildebrand Glow) shows import of 17.09kWh at off-peak (£1.19) and 0.09kWh at peak (£0.02) with a standing charge of 62p so the actual cost is £1.83 so the original £1.82 reported at 09:00 was the correct figure showing actual electricity cost and I've absolutely no idea where the additional £1.70 came from?

```
2024-11-26 08:50:04.716560 INFO pred_bat: Today's energy import 16.99 kWh export 0.04 kWh total 16.95 kWh cost 182.69 p import 120.9 p export -0.44 p carbon 0.0 kg
2024-11-26 08:55:05.260553 INFO pred_bat: Today's energy import 17.0 kWh export 0.05 kWh total 16.95 kWh cost 182.75 p import 121.03 p export -0.5 p carbon 0.0 kg
2024-11-26 09:00:05.057403 INFO pred_bat: Today's energy import 17.0 kWh export 0.05 kWh total 16.95 kWh cost 182.79 p import 121.11 p export -0.54 p carbon 0.0 kg
2024-11-26 09:05:05.018871 INFO pred_bat: Today's energy import 17.01 kWh export 0.05 kWh total 16.95 kWh cost 352.69 p import 291.02 p export -0.55 p carbon 0.0 kg
2024-11-26 09:10:04.790311 INFO pred_bat: Today's energy import 17.01 kWh export 0.05 kWh total 16.96 kWh cost 352.77 p import 291.12 p export -0.58 p carbon 0.0 kg
2024-11-26 09:15:04.738289 INFO pred_bat: Today's energy import 17.01 kWh export 0.06 kWh total 16.95 kWh cost 352.73 p import 291.18 p export -0.67 p carbon 0.0 kg
2024-11-26 09:20:04.767144 INFO pred_bat: Today's energy import 17.01 kWh export 0.06 kWh total 16.95 kWh cost 352.69 p import 291.18 p export -0.71 p carbon 0.0 kg
```

I assume I'm missing something obvious or have something misconfigured somewhere but despite looking at this many times over the last few months it seems mostly random and I cannot figure out why the daily costs are so far out.

cyberkryten, 2024-11-26:
Here's the entire appdaemon.log file from 09:00 to 09:15 this morning as I assume there answer will be in here somewhere but I've looked through and I can't figure out how the cost figure was reached

[PredBad Log 26Nov24.txt](https://github.com/user-attachments/files/17917488/PredBad.Log.26Nov24.txt)

cyberkryten, 2024-11-26:
Maybe it isn't quite as random as I thought, same time as yesterday (16:20) and the import cost jumped, yesterday £5.03 to £7.61 and today £5.33 to £8.08

```
2024-11-26 16:00:04.901361 INFO pred_bat: Today's energy import 36.66 kWh export 8.7 kWh total 27.95 kWh cost 475.02 p import 526.91 p export -114.12 p carbon 0.0 kg
2024-11-26 16:05:04.764680 INFO pred_bat: Today's energy import 36.9 kWh export 8.9 kWh total 28.0 kWh cost 476.13 p import 533.28 p export -119.37 p carbon 0.0 kg
2024-11-26 16:10:05.078413 INFO pred_bat: Today's energy import 36.9 kWh export 9.19 kWh total 27.71 kWh cost 468.66 p import 533.28 p export -126.85 p carbon 0.0 kg
2024-11-26 16:15:05.401415 INFO pred_bat: Today's energy import 36.9 kWh export 9.47 kWh total 27.43 kWh cost 461.17 p import 533.28 p export -134.34 p carbon 0.0 kg
2024-11-26 16:20:05.110012 INFO pred_bat: Today's energy import 36.9 kWh export 9.76 kWh total 27.15 kWh cost 729.13 p import 808.64 p export -141.73 p carbon 0.0 kg
2024-11-26 16:25:05.073130 INFO pred_bat: Today's energy import 36.9 kWh export 10.03 kWh total 26.87 kWh cost 721.92 p import 808.64 p export -148.94 p carbon 0.0 kg
```

Cost in p/kWh changes massively around these times too:

[image]

Trying to pick my way through the code, it seems there's three LOG outputs from the today_cost function:

```
2024-11-26 16:15:05.343675 INFO pred_bat: Battery level now 26.19 -1hr 19.79 midnight 18.04 battery value change hour 48.11 day 61.27 rate_forward 7.52
2024-11-26 16:15:05.344767 INFO pred_bat: Hour energy 6.92 import 7.69 export 0.77 car 0.0 load 1.25 cost 41.46 import 61.68 export -20.21 car 0.0 carbon 0.0 kG
2024-11-26 16:15:05.401415 INFO pred_bat: Today's energy import 36.9 kWh export 9.47 kWh total 27.43 kWh cost 461.17 p import 533.28 p export -134.34 p carbon 0.0 kg
2024-11-26 16:20:05.051045 INFO pred_bat: Battery level now 25.61 -1hr 20.66 midnight 18.04 battery value change hour 37.21 day 56.91 rate_forward 7.52
2024-11-26 16:20:05.052176 INFO pred_bat: Hour energy 5.81 import 6.87 export 1.05 car 0.0 load 1.38 cost 151.06 import 178.66 export -27.6 car 0.0 carbon 0.0 kG
2024-11-26 16:20:05.110012 INFO pred_bat: Today's energy import 36.9 kWh export 9.76 kWh total 27.15 kWh cost 729.13 p import 808.64 p export -141.73 p carbon 0.0 kg
2024-11-26 16:25:05.013939 INFO pred_bat: Battery level now 25.32 -1hr 21.24 midnight 18.04 battery value change hour 30.67 day 54.73 rate_forward 7.52
2024-11-26 16:25:05.015029 INFO pred_bat: Hour energy 4.73 import 6.06 export 1.33 car 0.0 load 1.55 cost 122.75 import 157.56 export -34.81 car 0.0 carbon 0.0 kG
2024-11-26 16:25:05.073130 INFO pred_bat: Today's energy import 36.9 kWh export 10.03 kWh total 26.87 kWh cost 721.92 p import 808.64 p export -148.94 p carbon 0.0 kg
```

I can't see anything really wrong in the code so I've added some more logging to grab the values as it cycles through each minute's data to calculate the values, will update when it next hits the issue.

cyberkryten, 2024-11-26:
Got another one, with the extra logging enabled which is the following to output the minute, import, load, rate and cost for every individual minute in the previous hour:

```
        for minute_back in range(60):                                                 
            minute = self.minutes_now - minute_back
            energy_import = self.get_from_incrementing(import_today, minute_back)
            load_energy = self.get_from_incrementing(load_today, minute_back)
                        
            self.log(   
                "MJI minute {} energy_import {} load_energy {} rate_import {} hour_cost_import {} ".format(
                    self.dp2(minute),
                    self.dp2(energy_import),
                    self.dp2(load_energy),
                    self.dp2(self.rate_import[minute]),
                    self.dp2(hour_cost_import),
                )
            )     
                  
            if car_today:
                energy_car = self.get_from_incrementing(car_today, minute_back)
            else:               
                energy_car = 0           
                          
            if export_today:
                energy_export = self.get_from_incrementing(export_today, minute_back)
            else:       
                energy_export = 0
```

So, during the run at 21:30 Predbat recorded the rate for all minutes from 1231 (20:31) to 1290 (21:31) as 7 pence:

```
2024-11-26 21:25:05.049247 INFO pred_bat: Hour energy 6.21 import 6.21 export 0.0 car 0.0 load 0.59 cost 43.5 import 43.5 export -0.0 car 0.0 carbon 0.0 kG
2024-11-26 21:25:05.109176 INFO pred_bat: Today's energy import 56.73 kWh export 18.47 kWh total 38.26 kWh cost 642.85 p import 949.52 p export -368.9 p carbon 0.0 kg
2024-11-26 21:30:05.037759 INFO pred_bat: Battery level now 17.17 -1hr 13.39 midnight 18.04 battery value change hour 28.41 day -6.54 rate_forward 7.52
2024-11-26 21:30:05.038741 INFO pred_bat: MJI minute 1290 energy_import 0.15 load_energy 0.0 rate_import 7.0 hour_cost_import 0 
2024-11-26 21:30:05.039596 INFO pred_bat: MJI minute 1289 energy_import 0.15 load_energy 0.03 rate_import 7.0 hour_cost_import 1.08 
2024-11-26 21:30:05.040448 INFO pred_bat: MJI minute 1288 energy_import 0.15 load_energy 0.0 rate_import 7.0 hour_cost_import 2.14 
2024-11-26 21:30:05.041289 INFO pred_bat: MJI minute 1287 energy_import 0.16 load_energy 0.0 rate_import 7.0 hour_cost_import 3.22 
2024-11-26 21:30:05.042358 INFO pred_bat: MJI minute 1286 energy_import 0.16 load_energy 0.01 rate_import 7.0 hour_cost_import 4.34 
2024-11-26 21:30:05.043280 INFO pred_bat: MJI minute 1285 energy_import 0.16 load_energy 0.0 rate_import 7.0 hour_cost_import 5.45 
2024-11-26 21:30:05.044124 INFO pred_bat: MJI minute 1284 energy_import 0.16 load_energy 0.0 rate_import 7.0 hour_cost_import 6.55 
2024-11-26 21:30:05.044974 INFO pred_bat: MJI minute 1283 energy_import 0.16 load_energy 0.03 rate_import 7.0 hour_cost_import 7.65 
2024-11-26 21:30:05.045817 INFO pred_bat: MJI minute 1282 energy_import 0.16 load_energy 0.0 rate_import 7.0 hour_cost_import 8.74 
2024-11-26 21:30:05.046659 INFO pred_bat: MJI minute 1281 energy_import 0.16 load_energy 0.0 rate_import 7.0 hour_cost_import 9.87 
2024-11-26 21:30:05.047480 INFO pred_bat: MJI minute 1280 energy_import 0.16 load_energy 0.0 rate_import 7.0 hour_cost_import 10.97 
2024-11-26 21:30:05.048348 INFO pred_bat: MJI minute 1279 energy_import 0.16 load_energy 0.03 rate_import 7.0 hour_cost_import 12.07 
2024-11-26 21:30:05.049173 INFO pred_bat: MJI minute 1278 energy_import 0.16 load_energy 0.0 rate_import 7.0 hour_cost_import 13.16 
2024-11-26 21:30:05.049994 INFO pred_bat: MJI minute 1277 energy_import 0.15 load_energy 0.0 rate_import 7.0 hour_cost_import 14.26 
2024-11-26 21:30:05.050821 INFO pred_bat: MJI minute 1276 energy_import 0.16 load_energy 0.0 rate_import 7.0 hour_cost_import 15.34 
2024-11-26 21:30:05.051630 INFO pred_bat: MJI minute 1275 energy_import 0.15 load_energy 0.02 rate_import 7.0 hour_cost_import 16.46 
2024-11-26 21:30:05.052448 INFO pred_bat: MJI minute 1274 energy_import 0.16 load_energy 0.0 rate_import 7.0 hour_cost_import 17.53 
2024-11-26 21:30:05.053278 INFO pred_bat: MJI minute 1273 energy_import 0.16 load_energy 0.0 rate_import 7.0 hour_cost_import 18.63 
2024-11-26 21:30:05.054097 INFO pred_bat: MJI minute 1272 energy_import 0.16 load_energy 0.0 rate_import 7.0 hour_cost_import 19.73 
2024-11-26 21:30:05.054916 INFO pred_bat: MJI minute 1271 energy_import 0.15 load_energy 0.03 rate_import 7.0 hour_cost_import 20.83 
2024-11-26 21:30:05.055742 INFO pred_bat: MJI minute 1270 energy_import 0.16 load_energy 0.0 rate_import 7.0 hour_cost_import 21.91 
2024-11-26 21:30:05.056559 INFO pred_bat: MJI minute 1269 energy_import 0.16 load_energy 0.0 rate_import 7.0 hour_cost_import 23.01 
2024-11-26 21:30:05.057692 INFO pred_bat: MJI minute 1268 energy_import 0.16 load_energy 0.0 rate_import 7.0 hour_cost_import 24.13 
2024-11-26 21:30:05.058532 INFO pred_bat: MJI minute 1267 energy_import 0.16 load_energy 0.03 rate_import 7.0 hour_cost_import 25.21 
2024-11-26 21:30:05.059368 INFO pred_bat: MJI minute 1266 energy_import 0.16 load_energy 0.0 rate_import 7.0 hour_cost_import 26.34 
2024-11-26 21:30:05.060249 INFO pred_bat: MJI minute 1265 energy_import 0.15 load_energy 0.0 rate_import 7.0 hour_cost_import 27.45 
2024-11-26 21:30:05.061102 INFO pred_bat: MJI minute 1264 energy_import 0.16 load_energy 0.0 rate_import 7.0 hour_cost_import 28.53 
[... cut ...]
2024-11-26 22:05:05.108510 INFO pred_bat: Hour energy 9.31 import 9.31 export 0.0 car 0.0 load 0.35 cost 136.8 import 136.8 export 0.0 car 0.0 carbon 0.0 kG
2024-11-26 22:05:05.166893 INFO pred_bat: Today's energy import 62.9 kWh export 18.47 kWh total 44.42 kWh cost 804.57 p import 1111.24 p export -368.9 p carbon 0.0 kg
```

[image]

If anyone can point me where to start looking to find how those rates are getting changed, it'll save me a load of digging as there's a lot more .py files with rate_import in than with cost_import!

cyberkryten, 2024-11-27:
Happened again, and I can see that the 9am scan today dropped out a previous 'smart-charge' slot from IOG which ran from 6am to 8am:

```
2024-11-27 08:55:04.629302 INFO pred_bat: Car 0 using Octopus Intelligent, not charging is planned
2024-11-27 08:55:04.640776 INFO pred_bat: Octopus Intelligent slot at 11-25 23:30:00-11-26 00:00:00 assumed price 7.0 amount 2.65 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-27 08:55:04.642237 INFO pred_bat: Octopus Intelligent slot at 11-25 23:00:00-11-25 23:30:00 assumed price 7.0 amount 0.09 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-27 08:55:04.643518 INFO pred_bat: Octopus Intelligent slot at 11-26 19:00:00-11-26 19:30:00 assumed price 7.0 amount 1.02 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-27 08:55:04.644707 INFO pred_bat: Octopus Intelligent slot at 11-26 20:00:00-11-26 20:30:00 assumed price 7.0 amount 2.21 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-27 08:55:04.645525 INFO pred_bat: Octopus Intelligent slot at 11-26 19:30:00-11-26 20:00:00 assumed price 7.0 amount 3.26 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-27 08:55:04.646811 INFO pred_bat: Octopus Intelligent slot at 11-26 23:00:00-11-26 23:30:00 assumed price 7.0 amount 0.11 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-27 08:55:04.647967 INFO pred_bat: Octopus Intelligent slot at 11-26 22:30:00-11-26 23:00:00 assumed price 7.0 amount 0.05 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-27 08:55:04.649245 INFO pred_bat: Octopus Intelligent slot at 11-26 22:00:00-11-26 22:30:00 assumed price 7.0 amount 0.05 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-27 08:55:04.650519 INFO pred_bat: Octopus Intelligent slot at 11-26 23:30:00-11-27 00:00:00 assumed price 7.0 amount 1.05 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-27 08:55:04.651463 INFO pred_bat: Octopus Intelligent slot at 11-27 06:00:00-11-27 08:00:00 assumed price 7.0 amount 13.86 kWh location None source smart-charge octopus_slot_low_rate True
2024-11-27 08:55:04.810227 INFO pred_bat: Car 0 on Octopus Intelligent, no active plan

2024-11-27 09:00:04.673595 INFO pred_bat: Car 0 using Octopus Intelligent, not charging is planned
2024-11-27 09:00:04.683990 INFO pred_bat: Octopus Intelligent slot at 11-25 23:30:00-11-26 00:00:00 assumed price 7.0 amount 2.65 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-27 09:00:04.684857 INFO pred_bat: Octopus Intelligent slot at 11-25 23:00:00-11-25 23:30:00 assumed price 7.0 amount 0.09 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-27 09:00:04.685689 INFO pred_bat: Octopus Intelligent slot at 11-26 19:00:00-11-26 19:30:00 assumed price 7.0 amount 1.02 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-27 09:00:04.686502 INFO pred_bat: Octopus Intelligent slot at 11-26 20:00:00-11-26 20:30:00 assumed price 7.0 amount 2.21 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-27 09:00:04.688012 INFO pred_bat: Octopus Intelligent slot at 11-26 19:30:00-11-26 20:00:00 assumed price 7.0 amount 3.26 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-27 09:00:04.689319 INFO pred_bat: Octopus Intelligent slot at 11-26 23:00:00-11-26 23:30:00 assumed price 7.0 amount 0.11 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-27 09:00:04.690565 INFO pred_bat: Octopus Intelligent slot at 11-26 22:30:00-11-26 23:00:00 assumed price 7.0 amount 0.05 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-27 09:00:04.691812 INFO pred_bat: Octopus Intelligent slot at 11-26 22:00:00-11-26 22:30:00 assumed price 7.0 amount 0.05 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-27 09:00:04.693210 INFO pred_bat: Octopus Intelligent slot at 11-26 23:30:00-11-27 00:00:00 assumed price 7.0 amount 1.05 kWh location AT_HOME source None octopus_slot_low_rate True
2024-11-27 09:00:04.854426 INFO pred_bat: Car 0 on Octopus Intelligent, no active plan
```

That makes sense from an Octopus point of view, there's no need to maintain previous slots once the time they were active has ended (though I'm not sure why the source 'None' ones persist for a day or two but the source 'smart-charge' do not).  

In any case, it doesn't seem unrealistic for that to be happening as they've got to stop being advertised at some point.

So, the issue is that Predbat is retrospectively recalculating previous usage, which is incorrect because those transient low-rate slots are not longer being returned by the Octopus integration.

Nordtech, 2024-11-29:
I beleive I am seeing the same issue (or at least same symptoms). Came across this post when I was investigating. Noticed that the Cost Today chart was showing a higher Net Cost than the difference between Import and Export. 
Had a closer look and it seems that, although Net Cost is zero'd at midnight, 5 mins later there is a sudden jump with no corresponding jump in Import.

[image]

[image]

[image]

cyberkryten, 2024-12-10:
I can't figure out a way to fix this, having spent some time going through the code - anyone else got ideas?

From what I can see in the code, this constant recalculating is intentional, but in doing so it makes the assumption that during the final 23:55 Predbat run of the day, all historical low rate slots for the previous 24 hours will be present in the Octopus data, which they aren't, at least in my IOG account (I'd be surprised if my account was unique in this regard).

I've now also compared the data with the Octopus statement to ensure it's not something I'm doing wrong and that agrees with my figures within a few pence.

In the Debug above, you can see at 22:05 on the 26th that Predbat already has the Import cost at £11.11 but the figure on the Octopus statement is £5.07 for the entire day.

As an example of how far out the figures are, for the three days Nov 25/26/27 Octopus billed me at total of £13.56 for Import electricity, before the £12.46 of export on those days.  

So the actual cost was £1.10 for the three days whereas Predbat recorded my overall Net electricity costs as £23.62 which is £10 more than Octopus actually billed me before netting off the export!!

This doesn't seem to affect Predbat's ability to make decisions on charging slots - it gets that right and is brilliant at it, but it means the actual costs are hugely out, making the various costs and savings figures it reports inaccurate and misleading,

Nordtech, 2024-12-10:
Hi there, I removed a reply where I'd said that I thought I had the same issue. Turned out not to be the case and that Predbat had started adding in standing charges because I'd sorted out the right entities. 
Out of curiosity, I went and looked at a comparable period to see if there were any differences between PB and Octopus. Good news is that PB is spot on. Whilst it doesn't get to the cause of your issue, I think we can be assured that PB is correct and the issue is specific to your configuration. Maybe re-check all the entities are correct? Not much help I know, but wanted to support as it looks a vexing issue.

[image]

gcoan, 2024-12-10:
> From what I can see in the code, this constant recalculating is intentional, but in doing so it makes the assumption that during the final 23:55 Predbat run of the day, all historical low rate slots [...]

Your logic is sound, if Predbat recalculates the 'so far today' figures every 5 minute run then it relies on the underlying data being static for the day.  If Octopus remove the IOG slot information after the slot has occurred then this calculation will be out.

I guess Trefor does the calculation this way so that you don't get rounding errors on part kWh consumption as the inverter only reports import/export to 1 decimal place.

An alternative approach is to not use the Predbat cost calculation and instead use the current accumulative cost from the Octopus integration https://bottlecapdave.github.io/HomeAssistant-OctopusEnergy/#cost-tracker-sensors.  This has the advantage of being able to be displayed  in the HA Energy dashboard as well https://bottlecapdave.github.io/HomeAssistant-OctopusEnergy/setup/energy_dashboard/

I'm not on IOG so I don't have the problem you have, but this works for me, and by using the Energy dashboard it means you can easily look at costs by day/week/month etc

[image]

cyberkryten, 2024-12-10:
Thanks for the responses.  I'm pretty sure all the entities are correct, Predbat sees the data and gets it correct 'at the time' so I don't think that's the issue.

The problem's cause is that it does a recalculation later in the day, assuming the old low-rate slots will still be in the Octopus data when they are not.  I'm sure there are good reasons to do this - one is as you say to avoid rounding or double counting any usage and the other I can see is to allow for periods of downtime, resets/restarts of HA and lack of 'state' between runs.

Octopus have definitely changed how slots are scheduled as now they don't appear to be aligned to 00/05 minutes and are more variable, so it's possible they sped up the historic slot removal of the transient slots.  It isn't consistent as I do see slots in there from 1-2 days ago when some of the earlier today ones are removed.  Also, their persistence may depend on how often the car is plugged in/unplugged and the charge method (I have a Tesla, but have IOG linked to my Zappi charger).
So, it is entirely possible that when written this worked fine and/or it works for most people who don't have a similar setup and plug/unplug their car often.

I'd be interested if anyone on IOG could grab the output late at night to see if they have any "location None source smart-charge" entires remaining in their Previous Dispatches.

I do use the energy dashboard to manually calculate costs  and also have separate Grafana datasets built which show the real values which is how I noticed this is so far out.  However there's no way to replicate the 'Predbat Savings' figures (which whilst notional, do help me self-justify the cost of the system and time spend messing about with HA!)

cyberkryten, 2024-12-11:
I think it only happens when IOG has scheduled additional smart slots outside of the 23:30-05:30 time.  If the car is left plugged in for a couple of days and unused then there's no extra slots and therefore there are none present to get removed and cause this issue.

The Dispatch info is in the logs, yes.  Probably depends on how you're running Predbat but for my AppDaemon install this shows all of those smart slots - how far back it goes depends on your logging settings and how much debug you're generating.  This can be done anytime really as the data will be recorded as far back as the logging goes.

```
grep "Octopus Intelligent slot.*source smart-charge" /homeassistant/appdaemon/appdaemon.log*
```

Or you can see the data in HA by going to your Octopus Integration->Charger and looking at the Intelligent Dispatching entity and expanding out it's Attributes which then shows the previous charges and check they are all present - this would need to be done when you know there's been an additional slot and the time it was, then you can look and see if it is still there late evening:

[image]

cyberkryten, 2024-12-16:
This is debug output from the Octopus integration - early hours dispatches are removed 12 hours later (as per Octopus spec):

```
2024-12-16 14:27:59.453 DEBUG (MainThread) [custom_components.octopus_energy.api_client] async_get_intelligent_dispatches: {'data': {'plannedDispatches': [{'start': '2024-12-16T13:24:00+00:00', 'end': '2024-12-16T16:00:00+00:00', 'delta': '-17.13', 'meta': {'source': 'smart-charge', 'location': None}}, {'start': '2024-12-16T23:00:00+00:00', 'end': '2024-12-17T08:00:00+00:00', 'delta': '-59.31', 'meta': {'source': 'smart-charge', 'location': None}}], 'completedDispatches': [{'start': '2024-12-16T01:30:00+00:00', 'end': '2024-12-16T02:00:00+00:00', 'delta': '-2.23', 'meta': {'source': None, 'location': 'AT_HOME'}}]}}
2024-12-16 14:30:59.459 DEBUG (MainThread) [custom_components.octopus_energy.api_client] async_get_intelligent_dispatches: {'data': {'plannedDispatches': [{'start': '2024-12-16T13:24:00+00:00', 'end': '2024-12-16T16:00:00+00:00', 'delta': '-17.13', 'meta': {'source': 'smart-charge', 'location': None}}, {'start': '2024-12-16T23:00:00+00:00', 'end': '2024-12-17T08:00:00+00:00', 'delta': '-59.31', 'meta': {'source': 'smart-charge', 'location': None}}], 'completedDispatches': [{'start': '2024-12-16T01:30:00+00:00', 'end': '2024-12-16T02:00:00+00:00', 'delta': '-2.23', 'meta': {'source': None, 'location': 'AT_HOME'}}]}}
2024-12-16 14:34:59.435 DEBUG (MainThread) [custom_components.octopus_energy.api_client] async_get_intelligent_dispatches: {'data': {'plannedDispatches': [{'start': '2024-12-16T13:24:00+00:00', 'end': '2024-12-16T16:00:00+00:00', 'delta': '-17.13', 'meta': {'source': 'smart-charge', 'location': None}}, {'start': '2024-12-16T23:00:00+00:00', 'end': '2024-12-17T08:00:00+00:00', 'delta': '-59.31', 'meta': {'source': 'smart-charge', 'location': None}}], 'completedDispatches': []}}
2024-12-16 14:38:59.449 DEBUG (MainThread) [custom_components.octopus_energy.api_client] async_get_intelligent_dispatches: {'data': {'plannedDispatches': [{'start': '2024-12-16T13:24:00+00:00', 'end': '2024-12-16T16:00:00+00:00', 'delta': '-17.13', 'meta': {'source': 'smart-charge', 'location': None}}, {'start': '2024-12-16T23:00:00+00:00', 'end': '2024-12-17T08:00:00+00:00', 'delta': '-59.31', 'meta': {'source': 'smart-charge', 'location': None}}], 'completedDispatches': []}}

```

It also seems not all the Smart Charge slots ever make it to the 'completed dispatches' at all:

```
2024-12-16 16:14:03.410 DEBUG (MainThread) [custom_components.octopus_energy.api_client] async_get_intelligent_dispatches: {'data': {'plannedDispatches': [{'start': '2024-12-16T13:24:00+00:00', 'end': '2024-12-16T16:00:00+00:00', 'delta': '-17.13', 'meta': {'source': 'smart-charge', 'location': None}}, {'start': '2024-12-16T23:00:00+00:00', 'end': '2024-12-17T08:00:00+00:00', 'delta': '-59.31', 'meta': {'source': 'smart-charge', 'location': None}}], 'completedDispatches': []}}
2024-12-16 16:18:03.918 DEBUG (MainThread) [custom_components.octopus_energy.api_client] async_get_intelligent_dispatches: {'data': {'plannedDispatches': [{'start': '2024-12-16T23:00:00+00:00', 'end': '2024-12-17T08:00:00+00:00', 'delta': '-59.31', 'meta': {'source': 'smart-charge', 'location': None}}], 'completedDispatches': []}}

```

I can't see a way to resolve this in the existing Predbat code structure.  Given the Octopus specs, recalculating old slots can't work because they are removed in-day, so the only way I can really see to resolve it is store the current cost and increment it as each block of time passes.

gcoan, 2024-12-19:
Another side effect of the way predbat calculates cost today and forecast cost in the plan...

I had an octopus power up event today of two hours of free electricity. I managed this in predbat as an import rate override, setup via the predbat manual API

All worked fine. 

After the event was over I removed the now redundant import rate override and as predbat recalculates the entire cost today on each run, the today cost sensor and cost so far today and forecast in the plan all jumped up as if the free electricity event had never happened 

[image]

cyberkryten, 2024-12-20:
Yep, that's exactly what I'm seeing (but daily).

I was beginning to think it was just me!

cyberkryten, 2025-02-03:
This causes problems when comparing plan predictions against actual in-day performance because the cost figure is always wrong because of these jumps.  I'm currently trying to figure out settings for the combining of import/export and the increases in the cost base 'to now' mean it is pretty much impossible to figure out if a given plan worked or not :(

The only way I can think of resolving this, is to allow Predbat to read a sensor which stores the correct 'up to now' cost for today.  I have that as I already calculate it and display it instead of the Predbat figure, but the problem is that the Predbat incorrect figure is what feeds into the Plan and so what I see varies throughout the day.

Predbat's future prediction calculations of cost are, I think, correct but it is the 'now' starting point that is incorrect and changes throughout the day (currently 13p correct v 202p Predbat) so I have to manually review the Predbat data, figure out the correction and then apply this to the figures displayed in the Plan - if I could tell Predbat to read a 'cost so far today' sensor to use as the starting point for the Plan's Total column showing the cost I think that would solve my issues - anyone think of problems with doing that?

closed 2026-08-28 by chalfontchubby