springfall2008/batpred #1664: Variation between SOC and Predbat status SOC

stiplady, 2024-11-26:
**Describe the bug**
As discussed on the Facebook group I'm seeing a difference between the SOC of my batteries and the one shown on the predbat plan. Currently my battery is full and excess is going to the grid, it's been like this for a while and stable. My inverter shows that the battery is 100%, but the top of the Predbat dash page shows SOC 90%, similarly the plan also starts with the first row showing 90%.

Attached apps.yaml for my Solax inverter, also the log file

[predbat.log](https://github.com/user-attachments/files/17919406/predbat.log)
[apps.yaml.txt](https://github.com/user-attachments/files/17919443/apps.yaml.txt)

**Expected behaviour**
Predbat predictions should be correctly recognising a fully charged battery, the difference of 10% appears to cause the charging plan to run longer than it needs to.

**Predbat version**

8.7.2 (also seen in 8.6.1)

**Environment details**

- Solax G4 Hybrid Inverter, with 3 x 5.8kWh batteries
- Standard HAOS installer

**Screenshots**
[image]
[image]
[image]
[image]

**Log file**
I installed through HACS originally, but can not find any mention of appdaemon in the expected locations /addon_configs/a0d7b954_appdaemon or /config/appdaemon

springfall2008, 2024-12-26:
Do you have battery_scaling set, I think there is a bug where its applied twice?

springfall2008, 2024-12-26:
If so fix on 'main' for testing

closed 2025-01-01 by springfall2008